openwrt/openwrt #18956: ramips: mir3g: link speed incorrectly detected and no rx/tx

jwh, 2025-05-29:
### Describe the bug

I revived one of my old mir3g APs running 5.15 which worked fine at 100m, after upgrading to HEAD (6.6.92) it no longer works at 100m, the link speed is indicated as 1000 and no traffic is passed, presumably due to wrong clocks if it thinks its at 1000 instead of 100.

### OpenWrt version

r852+77-3d6e1a3c9d

### OpenWrt release

24.10

### OpenWrt target/subtarget

ramips/mt7621

### Device

Xiaomi Mi Router 3G

### Image kind

Self-built image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

kofec, 2025-05-29:
This is a workaround which I'm using till it will be fixed in 24.10 branch. As I have the same issue. There are more similar issues:
https://github.com/openwrt/openwrt/issues/15348
https://github.com/openwrt/openwrt/issues/18100
Our MIR3G has switch MT7530 
[    3.199874] mt7530-mdio mdio-bus:1f: MT7530 adapts as multi-chip module
[    3.233190] mt7530-mdio mdio-bus:1f: configuring for fixed/rgmii link mode
[    3.241859] mt7530-mdio mdio-bus:1f wan (uninitialized): PHY [mt7530-0:01] driver [MediaTek MT7530 PHY] (irq=22)
[    3.254768] mt7530-mdio mdio-bus:1f: Link is Up - 1Gbps/Full - flow control rx/tx
[    3.263163] mt7530-mdio mdio-bus:1f lan2 (uninitialized): PHY [mt7530-0:02] driver [MediaTek MT7530 PHY] (irq=23)
[    3.276343] mt7530-mdio mdio-bus:1f lan1 (uninitialized): PHY [mt7530-0:03] driver [MediaTek MT7530 PHY] (irq=24)

jwh, 2025-05-29:
yeah, that fixes it - will do for now.  Just added eee disable to the dts and it works first time

jwh, 2025-07-23:
Looks like this was probably fixed properly, closing

closed 2025-07-23 by jwh