openwrt/openwrt #18958: Pseudo file "dev" exists in source filesystem [...]

lucky-coder-21, 2025-05-29:
> Hi, I'm unable to upgrade via ASU. This is the error I get via cli for the upgrade. Currently I'm running OpenWrt 23.05.5 and I tried running `auc -b 24.10`
> 
> ```
> Pseudo file "dev" exists in source filesystem "/builder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/dev".
> Ignoring, exclude it (-e/-ef) to override.
> ``` 

 _Originally posted by @lucky-coder-21 in [#18684](https://github.com/openwrt/openwrt/issues/18684#issuecomment-2911118243)_

Hi, first time posting on github. I have a similar issue to this previous bug issue which was closed as "harmless" but my issue still persist and i'm unable to upgrade to 2410. Could someone provide with some advice? Thanks

brada4, 2025-05-30:
As told in previous thread you seem not to read - harmless?

lucky-coder-21, 2025-05-30:
> As told in previous thread you seem not to read - harmless?

Doesn’t really say on how to bypass or fix the issue. Is there something I can do to remove the error as it’s preventing ASU with continuing with the upgrade?