openwrt/openwrt #18971: Wan led control affects USB power

dima8389, 2025-05-31:
### Describe the bug

Device Linksys EA6350 v4
commans 
echo 1 > /sys/class/leds/green:wan/link
called from /etc/init.d/led
actually switches off USB power.
echo 0  > /sys/class/leds/green:wan/link will bring it back.

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

ramips/mt7621

### Device

Linksys EA6350 v4

### Image kind

Official downloaded image

### Steps to reproduce

No steps needed. USB port does not work unless hub with external power connected.
Removing /etc/init.d/led will resolve it.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

aimacintyre, 2025-06-03:
please try updating to 24.10.1

neheb, 2025-06-07:
gpio 7 is used for it. WAN port looks like it's assigned to GMAC1. @arinc9 is this related?