openwrt/openwrt #18979: TP-Link Archer C2600 (ipq806x) loses WAN connection (24.10.1)

banburybill, 2025-05-31:
### Describe the bug

After working perfectly for a period of several hours up to 2 days, the device suddenly becomes unable to pass traffic over the WAN. Wireless and wired LAN connections continue to work, but all external connectivity is lost.

`# /etc/init.d/network restart`

will restore correct operation.

### OpenWrt version

r28597-0425664679

### OpenWrt release

24.10.1

### OpenWrt target/subtarget

ipq806x/generic

### Device

TP-Link Archer C2600

### Image kind

Official downloaded image

### Steps to reproduce

The router is configured with 3 VLANs. Wireless, 2 SSIDs are configured for the main VLAN, and 1 SSID for 1 other VLAN. A single Ethernet port is configured to pass tagged VLAN traffic for the main VLAN and the other VLAN not used in wireless to a downstream router.

The WAN port is connected to a fibre ONT, configuration by DHCP. 

### Actual behaviour

All works nominally. At some random point - early on in a Teams call after the router has run for 8+ hours seems subjectively a favourite point, but it's not restricted to points of comparatively high traffic - packets cease to flow over the WAN interface.

### Expected behaviour

WAN traffic does not stop for no readily apparent reason.

### Additional info

I have tried:

- Changing the wireless -ct firmware for non-ct. The problem persists; subjectively, it feels like it occurs quicker with non-ct firmware.
- Disabling the wireless and configuring a standalone AP instead.
- Enabling and disabling flow offloading.

Sample logs.

Kernel:

[  101.851675] br-lan: port 10(phy1-ap2) entered forwarding state
[  101.865259] ath10k_pci 0000:01:00.0: failed to pull fw stats: -71
[ 1429.040467] ath10k_pci 0001:01:00.0: htt tx: fixing invalid VHT TX rate code 0xff
[ 1961.159618] br-lan: port 3(lan3) entered disabled state
[ 1961.160288] qca8k 37000000.mdio-mii:10 lan3: Link is Down
[ 1970.521348] qca8k 37000000.mdio-mii:10 lan3: Link is Up - 100Mbps/Full - flow control rx/tx
[ 1970.521427] br-lan: port 3(lan3) entered blocking state
[ 1970.528585] br-lan: port 3(lan3) entered forwarding state
[ 6191.184972] ath10k_pci 0001:01:00.0: wmi: fixing invalid VHT TX rate code 0xff

System:

       
Thu May 22 20:35:56 2025 daemon.notice netifd: wan (2593): udhcpc: lease of 154.49.121.61 obtained from 154.49.120.1, lease time 300
Thu May 22 20:36:06 2025 daemon.info hostapd: phy1-ap1: STA 16:16:9b:e8:60:f2 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Thu May 22 20:36:25 2025 daemon.notice hostapd: phy0-ap1: AP-STA-DISCONNECTED 16:16:9b:e8:60:f2
Thu May 22 20:36:26 2025 daemon.info hostapd: phy0-ap1: STA 16:16:9b:e8:60:f2 IEEE 802.11: disassociated
Thu May 22 20:36:27 2025 daemon.info hostapd: phy0-ap1: STA 16:16:9b:e8:60:f2 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Thu May 22 20:36:27 2025 daemon.info hostapd: phy1-ap1: STA 16:16:9b:e8:60:f2 IEEE 802.11: authenticated
Thu May 22 20:36:27 2025 daemon.info hostapd: phy1-ap1: STA 16:16:9b:e8:60:f2 IEEE 802.11: associated (aid 1)
Thu May 22 20:36:27 2025 daemon.notice hostapd: phy1-ap1: AP-STA-CONNECTED 16:16:9b:e8:60:f2 auth_alg=open
Thu May 22 20:36:27 2025 daemon.info hostapd: phy1-ap1: STA 16:16:9b:e8:60:f2 WPA: pairwise key handshake completed (RSN)
Thu May 22 20:36:27 2025 daemon.notice hostapd: phy1-ap1: EAPOL-4WAY-HS-COMPLETED 16:16:9b:e8:60:f2
Thu May 22 20:36:28 2025 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan.1) 192.168.24.115 16:16:9b:e8:60:f2
Thu May 22 20:36:28 2025 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan.1) 192.168.24.115 16:16:9b:e8:60:f2
Thu May 22 20:38:26 2025 daemon.notice netifd: wan (2593): udhcpc: sending renew to server 154.49.120.1
Thu May 22 20:38:39 2025 daemon.err hostapd: nl80211: kernel reports: key addition failed
Thu May 22 20:38:39 2025 daemon.info hostapd: phy0-ap1: STA 16:16:9b:e8:60:f2 IEEE 802.11: associated (aid 1)
Thu May 22 20:38:39 2025 daemon.notice hostapd: phy0-ap1: AP-STA-CONNECTED 16:16:9b:e8:60:f2 auth_alg=ft
Thu May 22 20:38:39 2025 daemon.notice hostapd: phy1-ap1: Prune association for 16:16:9b:e8:60:f2
Thu May 22 20:38:39 2025 daemon.notice hostapd: phy1-ap1: AP-STA-DISCONNECTED 16:16:9b:e8:60:f2

WAN traffic no longer passes, so no DHCP reply is received.

Any suggestions for other info to capture?

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

banburybill, 2025-06-02:
Further info:

- VLANs were added on upgrade to 24.10.0 and a new configuration from scratch. Prior to that the device had worked fine under 23.05 but with no user VLANs configured.
- I observed a loss of connection today and can confirm packets stop flowing before DHCP times out and the wan interface is taken down.

banburybill, 2025-06-08:
On reviewing the VLAN configuration on br-lan, I noticed that I had the following configured:

- lan3: Main VLAN Untagged and Is Primary VLAN, no other VLAN.
- lan4: Main VLAN Tagged and Is Primary VLAN, other wired VLAN Tagged.

I removed Is Primary VLAN on lan4, where VLAN traffic tagged, leaving only the single Untagged VLAN on lan3 marked Is Primary VLAN. The router has been stable for 4 days, so (for now) I consider the issue solved.

banburybill, 2025-06-09:
Inevitably, after closing the issue, it recurs a few hours later. :-( Please reopen.

luminoso, 2025-06-09:
You probably want to follow this thread: https://github.com/openwrt/openwrt/issues/18979

banburybill, 2025-06-15:
Thanks for the pointer. Mid last week the device started losing WAN connectivity multiple times a day. I tried forcing all network ports onto the same conduit as in the above issue, but no joy. I made the changes, verified that all devices were now @cpu0, ran speedtestcpp twice and lost WAN connectivity during the second run. :-(

As the C2600 is currently my primary home router, I couldn't live any longer with that level of instability. The sudden increase in WAN failure rather made me suspicious that I might have failing hardware, and swapped it temporarily for a BT Home Hub 2 with OpenWRT, which is also DSA but was completely stable. But much less performant.

As one last check, like others before me I have downgraded to 23.05.5. So far after nearly 3 days the C2600 has been rock solid.  If this stability persists, I will have to conclude that the kernel DSA driver is not stable for the particular chipset in these routers. The HomeHub also uses DSA, and that was solid for a day and a half with the same configuration as I was using on the C2600/24.10.

AJCxZ0, 2025-06-30:
I have the same device in the same configuration but without VLANs, and have [observed the same symptoms](https://forum.openwrt.org/t/tp-link-archer-c2600-wan-traffic-stops-after-23-05-5-to-24-10-1-and-reconfiguration/234505) from the upgrade to 24.10.x (currently 24.10.2).

Like @banburybill, the C2600 is my primary router, but I am happy to perform limited tests and report.

luminoso, 2025-07-01:
@Djfe is that included in 24.10.2 or does it need to be manually applied?

SanteriHetekivi, 2025-07-10:
Had the same issue on TP-Link Talon AD7200.
Internet was turning off multiple times a day and only restart of the device fixed the issue.
I started to resort using my mobile internet for important meetings.
Issue started when I updated from 22.03.7 to 24.10.2 on last weekend.
I was ready to buy a new device, but after downgrading to 23.05.5 it has now been working for last 24 hours without any issues.

Also 802.11ad radio on TP-Link Talon AD7200 was not showing up on 24.10.2, but works on 23.05.5.
I'm not using it, but when debugging the issue I also removed drivers for it and blacklisted it, but that did not help.

tom3q, 2025-08-06:
Having the same problem on my Netgear R7800 after updating to 24.10.

turtleletortue, 2026-01-02:
Have the same problems on the R7500v1. The EEE fix mentioned above did not help. Interestingly (perhaps helpful for others to reproduce), I could consistently get the issue to occur by running the speedtest.net speedtest on a connected device's browser. In the upload speed phase of the test, it will always drop WAN and even the router is unable to ping websites. The 23.05.* firmware works fine for comparison (like everyone else).

turtleletortue, 2026-01-03:
Building the latest OpenWRT source, the issue still occurs. However, when I revert the commits that switched from swconfig to DSA and build from source (so now my build uses swconfig instead), the issue no longer occurs. (I did this to double check that the issue was with the swconfig->DSA switch and not something that snuck into a kernel update.) So something with the switch from swconfig to DSA definitely seems to be the issue, so if you want to use >23.05.* it appears you'll have to
1. Revert DSA commits for ipq806x
2. Build from source
for fully functional WAN.

kylex2, 2026-03-08:
Same issues here with C2600. WAN connection dies. Tx counter still increments but Rx counter is frozen. Disabling EEE doesn't help.

I started comparing register values for the QCA8337. Before DSA the device tree contained register values:
https://github.com/openwrt/openwrt/blob/7f61f9625550a6214e5cc53424e105bcbcff8c7a/target/linux/ipq806x/files-5.15/arch/arm/boot/dts/qcom-ipq8064-ad7200-c2600.dtsi#L334-L356

On my device running 24.10.5 it looks like this, the 23.05 values are taken from above:
|addr|name|23.05|24.10.5|match|
|-|-|-|-|-|
|0x0004|PORT0_PAD_CTRL|0x07600000|0x07500000|⚠️|
|0x0008|PAD5_MODE|0x01000000|0x01000000|✅|
|0x000c|PAD6_MODE|0x00000080|0x00000080|✅|
|0x00e0|SGMII_CTRL|0xc74164de|0xc74164de|✅|
|0x00e4|MAC_PWR_SEL|0x0006a545|0x000aa545|⚠️|
|0x007c|PORT0_STATUS|0x0000004e|0x0000007e|⚠️|
|0x0094|PORT6_STATUS|0x0000004e|0x0000007e|⚠️|

A bit unsure what this means though. The values in the 23.05 DT are labels as initvals so not sure if further writes are made to these regs at runtime.

Per diff summary,
PORT0_PAD_CTRL:
|bit|name|23.05|24.10.5|comment|
|-|-|-|-|-|
|21-20|MAC0_RGMII_RXCLK_DELAY_SEL|0b10|0b01|Control the delay value of RGMII interface RXCLK. 11 = Maximum delay|

Maybe this parameter is configurable using `rx-internal-delay-ps`?

MAC_PWR_SEL,
|bit|name|23.05|24.10.5|comment|
|-|-|-|-|-|
|19|PWR_RGMII_0|0b0|0b1|For MAC0 RGMII interface power source selection when using internal regulator: 0 = 1.5 V, 1 = 1.8 V|
|18|PWR_RGMII_1|0b1|0b0|For MAC5/6 RGMII interface power source selection whenusing internal regulator: 0 = 1.5 V, 1 = 1.8 V|

No clue which is actually correct. Flipped power sources seems weird, unless it's don't care if using external source on PCB?

PORT0_STATUS,
|bit|name|23.05|24.10.5|comment|
|-|-|-|-|-|
|5|RX_FLOW_EN_0|0b0|0b1|Rx MAC flow control enable|
|4|TX_FLOW_EN_0|0b0|0b1|Tx MAC flow control enable|

PORT6_STATUS,
|bit|name|23.05|24.10.5|comment|
|-|-|-|-|-|
|5|RX_FLOW_EN_6|0b0|0b1|Rx MAC flow control enable|
|4|TX_FLOW_EN_6|0b0|0b1|Tx MAC flow control enable|

Perhaps the problem is a flow control issue?

Another matter is to clarify which port is connected to which interface. The [toh wiki](https://openwrt.org/toh/tp-link/archer_c2600_v1#network_interfaces) says `eth0` is connected to switch port 0 and `eth1` connected to switch port 6. It also says `wan` is on the same vlan as `eth1` but `ip link` on 24.10 suggests it is `eth0` now. So either `eth0` and `eth1` swapped names or the mapping was swapped. Either way it is confusing to me which interface is connected to what switch port.

kylex2, 2026-03-15:
I made some patches to restore some of the register settings and hopefully also fix the issues related to the switch. The code can be found at https://github.com/kylex2/openwrt/commits/openwrt-24.10-dev-qca8k/. It would be nice if more people could help test and if it looks good then try 25.12.

One way to build it is:
```
./scripts/feeds update -a
./scripts/feeds install -a
wget https://downloads.openwrt.org/releases/24.10.5/targets/ipq806x/generic/config.buildinfo -O .config
echo "CONFIG_PACKAGE_mdio-tools=y" >> .config # if you want to peek/poke the qca8k
make defconfig
make -j16
```

mdio reg read example:
```
root@OpenWrt:~# mdio 37000000.mdio-mii qca8k hi 0x0004
0x0760
root@OpenWrt:~# mdio 37000000.mdio-mii qca8k lo 0x0004
0x0000
```

QCA8337 register value summary
|addr|name|23.05|24.10.5-patched|match|
|-|-|-|-|-|
|0x0004|PORT0_PAD_CTRL|0x07600000|0x07600000|✅|
|0x0008|PAD5_MODE|0x01000000|0x01000000|✅|
|0x000c|PAD6_MODE|0x00000080|0x00000080|✅|
|0x00e0|SGMII_CTRL|0xc74164de|0xc74164de|✅|
|0x00e4|MAC_PWR_SEL|0x0006a545|0x000aa545|⚠️|
|0x007c|PORT0_STATUS|0x0000004e|0x0000004e|✅|
|0x0094|PORT6_STATUS|0x0000004e|0x0000004e|✅|

gartkb, 2026-04-10:
I was running v19 and just installed and configged v25 yesterday and I appear to be having this issue, any advice on where to go from here?
OpenWrt 25.12.2 r32802-f505120278 / LuCI (HEAD detached at 067535e) branch 26.082.75780~067535e

botischroeder, 2026-04-15:
Thanks @kylex2 for your patches. I migrated and put them on top of v25.12.2 and uploaded the binaries: [https://github.com/botischroeder/openwrt/releases/tag/v25.12.2_patched](https://github.com/botischroeder/openwrt/releases/tag/v25.12.2_patched). The config is vanilla, with additional luci and patched mdio-tools. I have it running on my main C2600 since yesterday around 1pm (almost 20 hours so far) without any problems.

kylex2, what is the patched mdio-tools needed for?

botischroeder, 2026-04-15:
I added another patch for setting the voltage for RGMII1 to 1.8v for the C2600, like in 23.05: [https://github.com/botischroeder/openwrt/releases/tag/v25.12.2_patched_RGMII1-1.8](https://github.com/botischroeder/openwrt/releases/tag/v25.12.2_patched_RGMII1-1.8)

Now all reg values match between 23.05 and 25.12.2_patched.

QCA8337 register value summary
|addr|name|23.05|25.12.2-patched|match|
|-|-|-|-|-|
|0x0004|PORT0_PAD_CTRL|0x07600000|0x07600000|✅|
|0x0008|PAD5_MODE|0x01000000|0x01000000|✅|
|0x000c|PAD6_MODE|0x00000080|0x00000080|✅|
|0x00e0|SGMII_CTRL|0xc74164de|0xc74164de|✅|
|0x00e4|MAC_PWR_SEL|0x0006a545|0x0006a545|✅|
|0x007c|PORT0_STATUS|0x0000004e|0x0000004e|✅|
|0x0094|PORT6_STATUS|0x0000004e|0x0000004e|✅|

botischroeder, 2026-04-20:
@kylex2 I also think it's the right time. Thanks for your great work!

banburybill, 2026-05-08:
No, it didn't.

If anyone is desperate to try 25.12.3, I'm experimenting on my C2600 with a build of 25.12.3 incorporating most of the patches but omitting @kylex2's removed-forced-flow-control-on-CPU-ports patch. I can see value in this patch, but am just curious as to whether the reported interface flapping is actually a problem on the C2600.

https://github.com/banburybill/openwrt-kylex2/releases/tag/25.12.3-qca8k-registers-only

gartkb, 2026-08-30:
I'm not knowledgeable about the process, and I don't want to be rude to anyone, but this does need to be included in official updates. Leaving timebombs in for future endusers isn't great esp once the effort has gone in to fix it. So what's the process?  Apparently tagging one person isn't doing it.

BIGFAAT, 2026-10-23:
What is the status with the latest release (25.12.6)? Until I'm blind I didn't see any fixes related.

luminoso, 2026-10-23:
I'm willing to open a PR upstream for this (with full credit to whoever found the solution).

Could someone point me to the correct, working pull request or commit?
I got a bit lost wading through the various experiments trying to find the final fix.

pbix, 2026-10-24:
In my case I disabled eee at startup using ethtool by making the below script in my /etc/init.d directory and marking it executable.
This has resolved the issue for me in v24.10.4.  I had to install the ethtool package prior to make it available to this script.

`root@MainAP:/etc/init.d# cat disable-eee 
#!/bin/sh /etc/rc.common

START=99
USE_PROCD=1

start_service() {
    # Give DSA/network interfaces time to initialize.
    sleep 5

    logger -t disable-eee "Disabling EEE on wan"
    /usr/sbin/ethtool --set-eee wan eee off
}
root@MainAP:/etc/init.d# 
`

banburybill, 2026-10-25:
> 24\. I'm willing to open a PR upstream for this (with full credit to whoever found the solution).
>      Could someone point me to the correct, working pull request, comment, branch, or commit with the "good" values?
>      I got a bit lost wading through the various experiments trying to find the final fix.

I've been running the patch series from @kylex2 applied against 25.12.5 since the release of 25.12.5, and it's been rock solid on my C2600. I did try the disable-eee fix, but without success.

I've just now rebased that series onto 25.12.6 and updated my C2600. This rebased patch series is at https://github.com/banburybill/openwrt-kylex2/commits/openwrt-25.12.6-qca8k/ and sysupgrade packages at https://github.com/banburybill/openwrt-kylex2/releases/tag/v25.12.6-qca8k.

If you'd like to test and review the patches too and open a PR, I'd welcome that.

luminoso, 2026-10-28:
Thanks @banburybill, that branch was the piece I was missing.

**Testers needed.** The fix is now in draft PR #26259, with test images for every ipq806x device. My own C2600 sits at a remote site that I can't reach, so I can't flash it myself. If you see the WAN freeze today, please flash the image, run it as usual and report back here. After a few days of stable reports from several people I'll move the PR from draft to ready for review. Details below.

I went through the whole series and compared it with @botischroeder's 25.12.2 builds and the 23.05 swconfig initvals, register by register and board by board. Same fix. Both give the C2600 identical register values and differ only in the board list. Every build that stayed up had RGMII1 at 1.8 V (MAC_PWR_SEL 0x0006a545), while builds with only the RX delay and the flow control change still dropped WAN.

## What goes into the PR

Four commits, in #26259 against main. The same commits for 25.12 are ready on my fork in case maintainers want a backport PR:

1. kernel: document two DT flags, `qca,mac-pwr-sel-rgmii0-1p8v` and `qca,mac-pwr-sel-rgmii1-1p8v`.
2. ipq806x: RX delay back to 2000 ps, as in 23.05.
3. ipq806x: set one flag per board using the effective 23.05 values (botischroeder's list). kylex2's last commit also set both bits on R7800/XR500/NBG6817/RT4230W/TR4400 and OnHub/R7500v2. None of them had both before DSA.
4. kernel: qca8k reads the flags instead of guessing from the SoC compatible. Boards that set neither are left alone, so ath79, qualcommax and the rest keep their current state.

The "don't force flow control on CPU ports" patch is not in it. It changes CPU-port behaviour on every qca8k board in the tree, and nobody has tested the fix without it. If WAN still drops with these images, it goes into a second PR.

## Testers wanted

I need people who see the freeze today, on any ipq806x board with a QCA8337: C2600, AD7200, R7800, R7500, EA8500, D7800 and so on. R7800/XR500 owners are especially useful, because their voltage setting changes too.

Images: 25.12.6 plus the four commits, built with the official 25.12.6 config, every ipq806x device, plus mdio-tools for reading the switch registers: https://github.com/luminoso/openwrt/releases/tag/v25.12.6-qca8k-test1 (check your download with `sha256sum -c sha256sums --ignore-missing`). To go back, sysupgrade to the official 25.12.6 image.

The kernel config is the official one: the vermagic hash d02d560e matches the official 25.12.6 kmods directory, so packages and kmods from the 25.12.6 repository should install as usual. Nobody has tried that on a device yet, so shout if apk refuses something. The test image shows `OpenWrt 25.12.6 r33311-981c978eff-qca8k-test1` in LuCI and in `DISTRIB_DESCRIPTION` in /etc/openwrt_release.

If you would rather build it yourself, this is what I ran, on Debian trixie:

```
sudo apt-get install -y build-essential clang flex bison g++ gawk gcc-multilib g++-multilib gettext git libncurses-dev libssl-dev python3 python3-setuptools rsync swig unzip zlib1g-dev file wget ca-certificates curl perl patch quilt xz-utils bzip2 zstd time diffutils libelf-dev python3-dev
git clone -b test/ipq806x-25.12.6-qca8k https://github.com/luminoso/openwrt.git openwrt && cd openwrt
git checkout f7b60b4133bc3b1a9d2c230907f910d8db1ce7a4
./scripts/feeds update -a && ./scripts/feeds install -a
wget -O .config https://downloads.openwrt.org/releases/25.12.6/targets/ipq806x/generic/config.buildinfo
sed -i -E '/^CONFIG_(SDK|IB|SDK_LLVM_BPF|MAKE_TOOLCHAIN|JSON_CYCLONEDX_SBOM)=y$/d' .config
printf '%s\n' '# CONFIG_SDK is not set' '# CONFIG_IB is not set' '# CONFIG_SDK_LLVM_BPF is not set' '# CONFIG_MAKE_TOOLCHAIN is not set' '# CONFIG_JSON_CYCLONEDX_SBOM is not set' 'CONFIG_BPF_TOOLCHAIN_NONE=y' 'CONFIG_PACKAGE_mdio-tools=y' 'CONFIG_PACKAGE_kmod-mdio-netlink=y' 'CONFIG_VERSION_CODE="r33311-981c978eff-qca8k-test1"' >> .config
make defconfig
make -j$(nproc) download
make -j$(nproc)
```

The sed and printf lines skip the SDK, ImageBuilder and LLVM BPF toolchain that the release config builds, which saves hours. Images land in `bin/targets/ipq806x/generic/`. Expect a couple of hours on the first run.

OpenWrt's CI may also build a C2600 snapshot image from #26259 once a maintainer approves the run. That one is based on main, has no LuCI and needs a GitHub login to download.

Please don't combine the image with the EEE workaround (`option eee '0'` or the ethtool script). banburybill found it did not help on the C2600, and it would muddy the result.

Please report back with:

- model, and the `DISTRIB_DESCRIPTION` line so I know which image you run
- the four registers, same commands kylex2 posted: `mdio 37000000.mdio-mii qca8k hi 0x00e4` and `lo`, for 0x0004, 0x00e4, 0x007c and 0x0094
- uptime, and whether WAN froze. If it did, `ip -s link show wan` before restarting the network.

A few days without a freeze from several of you, C2600 owners especially, and the PR goes from draft to ready.

## Credit

@kylex2 did the analysis and the driver change, so the RX delay and driver commits keep you as author with your Signed-off-by. The board list changed and your RGMII0 commit had no Signed-off-by, so I author the per-board commit and name you and @botischroeder in the message. If you want formal tags:

- @kylex2: reply with `Signed-off-by: Alexander Vickberg <wickbergster@gmail.com>` and I add you as Co-developed-by on the per-board commit.
- @botischroeder: same, but OpenWrt needs a real name for Signed-off-by. Tell me the name you want used.
- @banburybill: OK to add `Tested-by: Jim Hague <email>`? Tell me which email.

banburybill, 2026-10-28:
@luminoso Thanks! I've flashed my C2600 to your 25.12.6 image, and confirmed it reports as `OpenWrt 25.12.6 r33311-981c978eff-qca8k-test1`. 

Please feel free to add `Tested-by: Jim Hague <jim.hague@acm.org>`.

Just one thing. In the past, it's taken up to 10-12 days for my unit to fail. I'll report any failure ASAP, but it's probably best to be conservative in how long you wait before progressing the PR.

botischroeder, 2026-10-29:
@luminoso: Great, that you created the PR! Thank you!
I'd appreciate it when you add `Signed-off-by: Timo Schröder <der.timosch@gmail.com>`.

I have your 25.12.6 image flashed on my VR2600v. It reports `OpenWrt 25.12.6 r33311-981c978eff-qca8k-test1`. 
I also built an image from main with your changes on top and flashed my two C2600 with it. Therefore they report `OpenWrt SNAPSHOT r37908-5700101d7c`.

The four QCA8337 registers are the same on every device:
|addr|name|value|
|-|-|-|
|0x00e4|MAC_PWR_SEL|0x0006a545|
|0x0004|PORT0_PAD_CTRL|0x07600000|
|0x007c|PORT0_STATUS|0x0000017e|
|0x0094|PORT6_STATUS|0x0000017e|

I will report back in one or two weeks.

luminoso, 2026-10-29:
Thanks both, that is exactly what I needed.

@banburybill: `Tested-by: Jim Hague <jim.hague@acm.org>` is on the RX delay, per-board and driver commits in #26259. Point taken about 10-12 days: the PR stays a draft until you have all run it for at least two weeks.

@botischroeder: you are now `Co-developed-by` and `Signed-off-by: Timo Schröder <der.timosch@gmail.com>` on the per-board commit, which is your board list.

Your register dump also answers the open question. PORT0/6_STATUS at 0x0000017e means RX and TX flow control are still forced on the CPU ports (0x7e plus the link-up bit), because kylex2's flow control patch is not in this series. So your VR2600v and the two C2600 are the first runs of the voltage fix on its own. If they stay up, that patch is not needed for this bug.

One correction to my last comment: the CI image from the PR will not happen. That job tries to post a comment on the PR, and a pull request from a fork only gets a read-only token, so it stops with a 403 before building anything. The release images or your own build are the way to test. I have removed the build line from the PR.

Still looking for an R7800, XR500 or NBG6817 owner. Those boards lose the RGMII1 bit they have had since 24.10, and nobody has tested that yet.

tom3q, 2026-10-30:
Thanks folks for moving this forward. Much appreciated.

I'll give it a try on my R7800 starting this weekend.

BIGFAAT, 2026-10-30:
@luminoso Does your image provide full wpad-mbedtls? If not can I please get one? That would allow me to flash my C2600s remotely and let me participate in the testing. Thanks in advance!

luminoso, 2026-10-31:
@tom3q great, the R7800 is the board I most wanted covered. One request: dump the four registers on your current 25.12.6 install before flashing, then again on the test image. On the R7800 the change is in MAC_PWR_SEL (0x00e4): today both bit 19 (RGMII0) and bit 18 (RGMII1) are set, and the test image keeps only bit 19, the value swconfig wrote in 23.05. With the same other bits as the C2600 that is 0x000ea545 before and 0x000aa545 after. If `mdio 37000000.mdio-mii qca8k hi 0x00e4` complains about the bus name, run `mdio` with no arguments to list the buses.

@BIGFAAT no, the images had wpad-basic-mbedtls, the 25.12 default. I built a C2600 variant with full wpad-mbedtls. Same commit, still reports `r33311-981c978eff-qca8k-test1`, and the kernel with its DTB is byte-identical to the other C2600 images. Only the wifi daemon differs. Both files are in the same release:

- `openwrt-25.12.6-ipq806x-generic-tplink_c2600-wpad-mbedtls-squashfs-sysupgrade.bin`, to upgrade from OpenWrt
- `openwrt-25.12.6-ipq806x-generic-tplink_c2600-wpad-mbedtls-squashfs-factory.bin`, from the TP-Link firmware

https://github.com/luminoso/openwrt/releases/tag/v25.12.6-qca8k-test1

Their hashes are in the release's `sha256sums`, so `sha256sum -c sha256sums --ignore-missing` checks them. For a remote flash, keep settings (no `-n`) so your mesh config survives.

For anyone else who can still reach the router over a cable after flashing: the kernel matches the official 25.12.6 one, so `apk del wpad-basic-mbedtls && apk add wpad-mbedtls` should also work.

tom3q, 2026-10-31:
> 81\. One request: dump the four registers on your current 25.12.6 install before flashing, then again on the test image. On the R7800 the change is in MAC_PWR_SEL (0x00e4): today both bit 19 (RGMII0) and bit 18 (RGMII1) are set, and the test image keeps only bit 19, the value swconfig wrote in 23.05. With the same other bits as the C2600 that is 0x000ea545 before and 0x000aa545 after. 

I'm currently on 23.05.5 (I know not very great in today's reality, but I need a stable connection), do you want me to dump anything from there before I switch to 25.12.6?

> If `mdio 37000000.mdio-mii qca8k hi 0x00e4` complains about the bus name, run `mdio` with no arguments to list the buses.

This doesn't seem to work on 23.05.5 - mdio there doesn't seem to support qca8k as the OBJ argument.

```
root@OpenWrt:~# mdio 37000000.mdio-mii qca8k hi 0x00e4
ERROR: "qca8k" is not a valid device
```

is the error I'm getting.

BIGFAAT, 2026-10-31:
@luminoso Thank you for your patience, I will report within 2 weeks if the WAN ports are stable or not. I requested this to you since I sadly don't have the time to make my own build server as for right now.

luminoso, 2026-10-31:
@tom3q 23.05.5 is an even better "before" than 25.12.6, so yes please. My earlier request was wrong anyway: the `qca8k` object only exists in the mdio-tools of the test image, so the official 25.12.6 can't read these registers either.

On 23.05 you can read them with raw MDIO. The swconfig driver reaches a switch register by writing a page number to PHY address 0x18 and then reading two 16-bit halves from 0x10-0x17. It rewrites the page on every access, so doing the same from the shell doesn't disturb it. Paste this:

```sh
B=37000000.mdio-mii
half() { mdio $B phy 0x18 raw 0 0 >/dev/null; mdio $B phy $1 raw $2; }
reg() { printf '%s 0x%04x%04x\n' $1 $(half $2 $4) $(half $2 $3); }
reg 0x0004 0x10 0x02 0x03
reg 0x00e4 0x13 0x12 0x13
reg 0x007c 0x11 0x1e 0x1f
reg 0x0094 0x12 0x0a 0x0b
```

Run it twice. If the driver changes the page between two of those commands, one value comes out wrong, and the second run shows it.

0x00e4 is the one I care about most. The 23.05 R7800 DTS has two initvals blocks: 0xaa545 on the PHY at address 0 and 0x6a545 on address 4. The PR assumes swconfig applied the first, so I expect 0x000aa545, the same value the test image writes. 0x0006a545 would mean the PR picks the wrong bit for the R7800. For 0x0004 I expect 0x07600000.

When you move to the test image: coming from 23.05 the config can't be migrated from swconfig to DSA, so sysupgrade with `-n` and set the network up again.

@BIGFAAT thanks, two weeks is perfect.

tom3q, 2026-10-31:
That worked, thanks!

23.05.6

```
0x0004 0x07600000
0x007c 0x0000004e
0x0094 0x0000004e
0x00e4 0x000aa545
```

25.12.6 (unpatched)

```
0x0004 0x07500008
0x007c 0x0000017e
0x0094 0x0000017e
0x00e4 0x000ea545
```

Also, the WAN connection problem reproduces very easily and frequently for me on 25.12.6 - pretty much every 10-30 minutes, so it should be easy to tell if the patched version resolves it.

tom3q, 2026-10-31:
Managed to build and flash a patched build (instructions from https://github.com/openwrt/openwrt/issues/18979#issuecomment-6465594682), but unfortunately the WAN port keeps disconnecting pretty much the same as with the unpatched one.

Here are the register values:

```
0x0004 0x07600008
0x007c 0x0000017e
0x0094 0x0000017e
0x00e4 0x000aa545
```

The only relevant message in the logs I see when the disconnect happens are

```
[Nov 1, 2026, 4:17:15 AM GMT+9] daemon.notice: netifd: Network device 'wan' link is down
[Nov 1, 2026, 4:17:15 AM GMT+9] daemon.notice: netifd: Interface 'wan6' has link connectivity loss
[Nov 1, 2026, 4:17:15 AM GMT+9] kern.info: [  697.049786] qca8k 37000000.mdio-mii:10 wan: Link is Down
[Nov 1, 2026, 4:17:15 AM GMT+9] daemon.notice: netifd: Interface 'ntt' has link connectivity loss
```

luminoso, 2026-10-31:
@tom3q thank you, both dumps are exactly what I was missing. The 23.05.6 one settles the R7800 question: MAC_PWR_SEL 0x000aa545, the same value the PR writes.

Your symptom is a different one, though. On the C2600 the WAN link stays up, RX stops, and only a network restart brings it back. Yours drops link and is back two seconds later. WAN on both boards goes through switch port 5 and CPU port 0 (eth0), so the PR does change that path, but it doesn't touch the PHY itself.

My first suspect is EEE. 23.05 switched it off on every switch PHY ("due to stability issues", says the ar8327 driver). Under DSA the PHY driver clears the EEE advertisement on rev 2 switches, but phylib then writes the PHY's default advertisement back during autonegotiation, so EEE can end up on again. A PHY that fails to wake from low-power idle drops link for a second or two, which matches your log. On the build you have now, could you try this, one change at a time:

1. Before changing anything, post the output of `ethtool --show-eee wan` and `dmesg | grep -i qca8k`, and `ethtool -S wan` once before a drop and once after.
2. `ethtool --set-eee wan eee off`, check it with `ethtool --show-eee wan`, and leave it for 2-3 hours. If that holds, kylex2's `option eee '0'` in a `config device` section for `wan` (posted above) makes it permanent.
3. If it still drops, turn EEE back on and then try the flow control patch, so we know which change did it. With that patch 0x007c and 0x0094 should read 0x0000014e.

You drop every 10-30 minutes, so a few hours per step tells us.

tom3q, 2026-10-31:
> * Before changing anything, post the output of `ethtool --show-eee wan` and `dmesg | grep -i qca8k`, and `ethtool -S wan` once before a drop and once after.

```
root@OpenWrt:~# ethtool --show-eee wan
EEE Settings for wan:
	EEE status: disabled
	Tx LPI: disabled
	Supported EEE link modes:  100baseT/Full 
	                           1000baseT/Full 
	Advertised EEE link modes:  100baseT/Full 
	                            1000baseT/Full 
	Link partner advertised EEE link modes:  100baseT/Full 
	                                         1000baseT/Full 
```
```
root@OpenWrt:~# dmesg | grep -i qca8k
[    1.890434] qca8k 37000000.mdio-mii:10: configuring for fixed/rgmii link mode
[    1.898652] qca8k 37000000.mdio-mii:10: Link is Up - 1Gbps/Full - flow control off
[    1.903708] qca8k 37000000.mdio-mii:10: configuring for fixed/sgmii link mode
[    1.916486] qca8k 37000000.mdio-mii:10: Link is Up - 1Gbps/Full - flow control off
[    2.008148] qca8k 37000000.mdio-mii:10 lan4 (uninitialized): PHY [qca8k-0.0:00] driver [Qualcomm Atheros 8337 internal PHY] (irq=POLL)
[    2.108177] qca8k 37000000.mdio-mii:10 lan3 (uninitialized): PHY [qca8k-0.0:01] driver [Qualcomm Atheros 8337 internal PHY] (irq=POLL)
[    2.198149] qca8k 37000000.mdio-mii:10 lan2 (uninitialized): PHY [qca8k-0.0:02] driver [Qualcomm Atheros 8337 internal PHY] (irq=POLL)
[    2.288121] qca8k 37000000.mdio-mii:10 lan1 (uninitialized): PHY [qca8k-0.0:03] driver [Qualcomm Atheros 8337 internal PHY] (irq=POLL)
[    2.378122] qca8k 37000000.mdio-mii:10 wan (uninitialized): PHY [qca8k-0.0:04] driver [Qualcomm Atheros 8337 internal PHY] (irq=POLL)
[    5.927863] qca8k 37000000.mdio-mii:10 lan1: configuring for phy/internal link mode
[    9.138462] qca8k 37000000.mdio-mii:10 lan1: Link is Up - 1Gbps/Full - flow control rx/tx
[   10.515920] qca8k 37000000.mdio-mii:10 lan1: Link is Down
[   29.801335] qca8k 37000000.mdio-mii:10 lan1: configuring for phy/internal link mode
[   29.891708] qca8k 37000000.mdio-mii:10 lan1: entered allmulticast mode
[   29.941413] qca8k 37000000.mdio-mii:10 lan1: entered promiscuous mode
[   30.016542] qca8k 37000000.mdio-mii:10 lan4: configuring for phy/internal link mode
[   30.028353] qca8k 37000000.mdio-mii:10 lan4: entered allmulticast mode
[   30.067038] qca8k 37000000.mdio-mii:10 lan4: entered promiscuous mode
[   30.126119] qca8k 37000000.mdio-mii:10 lan2: configuring for phy/internal link mode
[   30.141376] qca8k 37000000.mdio-mii:10 lan2: entered allmulticast mode
[   30.166824] qca8k 37000000.mdio-mii:10 lan2: entered promiscuous mode
[   30.202562] qca8k 37000000.mdio-mii:10 lan3: configuring for phy/internal link mode
[   30.214265] qca8k 37000000.mdio-mii:10 lan3: entered allmulticast mode
[   30.229689] qca8k 37000000.mdio-mii:10 lan3: entered promiscuous mode
[   30.273665] qca8k 37000000.mdio-mii:10 wan: configuring for phy/internal link mode
[   32.418160] qca8k 37000000.mdio-mii:10 wan: Link is Up - 1Gbps/Full - flow control rx/tx
[   33.058450] qca8k 37000000.mdio-mii:10 lan1: Link is Up - 1Gbps/Full - flow control rx/tx
[   78.280137] qca8k 37000000.mdio-mii:10 wan: Link is Down
[   79.310864] qca8k 37000000.mdio-mii:10 wan: Link is Up - 1Gbps/Full - flow control rx/tx
[  132.348289] qca8k 37000000.mdio-mii:10 wan: Link is Down
[  134.430484] qca8k 37000000.mdio-mii:10 wan: Link is Up - 1Gbps/Full - flow control rx/tx
[  140.676312] qca8k 37000000.mdio-mii:10 wan: Link is Down
[  142.754386] qca8k 37000000.mdio-mii:10 wan: Link is Up - 1Gbps/Full - flow control rx/tx
[  301.857898] qca8k 37000000.mdio-mii:10 wan: Link is Down
[  302.897659] qca8k 37000000.mdio-mii:10 wan: Link is Up - 1Gbps/Full - flow control rx/tx
[  307.051463] qca8k 37000000.mdio-mii:10 wan: Link is Down
[  309.147340] qca8k 37000000.mdio-mii:10 wan: Link is Up - 1Gbps/Full - flow control rx/tx
[  448.506854] qca8k 37000000.mdio-mii:10 wan: Link is Down
[  449.546845] qca8k 37000000.mdio-mii:10 wan: Link is Up - 1Gbps/Full - flow control rx/tx
[  541.056581] qca8k 37000000.mdio-mii:10 wan: Link is Down
[  542.097752] qca8k 37000000.mdio-mii:10 wan: Link is Up - 1Gbps/Full - flow control rx/tx
[  581.616812] qca8k 37000000.mdio-mii:10 wan: Link is Down
[  582.656579] qca8k 37000000.mdio-mii:10 wan: Link is Up - 1Gbps/Full - flow control rx/tx
[  697.049786] qca8k 37000000.mdio-mii:10 wan: Link is Down
[  698.096220] qca8k 37000000.mdio-mii:10 wan: Link is Up - 1Gbps/Full - flow control rx/tx
[ 2797.838615] qca8k 37000000.mdio-mii:10 wan: Link is Down
[ 2798.885434] qca8k 37000000.mdio-mii:10 wan: Link is Up - 1Gbps/Full - flow control rx/tx
[ 6152.777403] qca8k 37000000.mdio-mii:10 wan: Link is Down
[ 6154.863192] qca8k 37000000.mdio-mii:10 wan: Link is Up - 1Gbps/Full - flow control rx/tx
[ 6179.818040] qca8k 37000000.mdio-mii:10 wan: Link is Down
[ 6181.903081] qca8k 37000000.mdio-mii:10 wan: Link is Up - 1Gbps/Full - flow control rx/tx
[ 6203.736779] qca8k 37000000.mdio-mii:10 wan: Link is Down
[ 6205.823144] qca8k 37000000.mdio-mii:10 wan: Link is Up - 1Gbps/Full - flow control rx/tx
```
[... 37 lines cut ...]
     TxOverSize: 0
     TxByte: 825096588
     TxCollision: 0
     TxAbortCol: 0
     TxMultiCol: 0
     TxSingleCol: 0
     TxExcDefer: 0
     TxDefer: 0
     TxLateCol: 0
     RXUnicast: 692165
     TXUnicast: 736255
```

> * `ethtool --set-eee wan eee off`, check it with `ethtool --show-eee wan`, and leave it for 2-3 hours. If that holds, kylex2's `option eee '0'` in a `config device` section for `wan` (posted above) makes it permanent.

It looks like it's already disabled., and the disable command seems to be ignored due to no change (as expected):

```
root@OpenWrt:~# ethtool --set-eee wan eee off
eee unmodified, ignoring
```

> * If it still drops, turn EEE back on and then try the flow control patch, so we know which change did it. With that patch 0x007c and 0x0094 should read 0x0000014e.

So I think I'll go ahead with the flow control patch.

tom3q, 2026-10-31:
Hmm, I've applied the patch from https://github.com/kylex2/openwrt/commit/1b15b0808bdd97d75bfd3657377ca91387b47068 and the registers seem to confirm the flow control being disabled, but I'm still seeing those disconnects and at similar rate.

```
0x0004 0x07600008
0x00e4 0x000aa545
0x007c 0x0000014e
0x0094 0x0000014e
```

```
root@OpenWrt:~# ethtool -S wan
NIC statistics:
     tx_packets: 3418996
     tx_bytes: 3342772184
     rx_packets: 5596444
     rx_bytes: 6493049302
     RxBroad: 10
     RxPause: 0
     RxMulti: 43
     RxFcsErr: 0
     RxAlignErr: 0
     RxRunt: 0
     RxFragment: 0
     Rx64Byte: 10
     Rx128Byte: 1038175
     Rx256Byte: 25579
     Rx512Byte: 2946
     Rx1024Byte: 4651
     Rx1518Byte: 4682015
     RxMaxByte: 0
     RxTooLong: 0
     RxGoodByte: 6748412718
     RxBadByte: 0
     RxOverFlow: 0
     Filtered: 0
     TxBroad: 20
     TxPause: 0
     TxMulti: 152
     TxUnderRun: 0
     Tx64Byte: 0
     Tx128Byte: 1019821
     Tx256Byte: 25485
     Tx512Byte: 5892
     Tx1024Byte: 2566
     Tx1518Byte: 2365232
     TxMaxByte: 0
     TxOverSize: 0
     TxByte: 3356448168
     TxCollision: 0
     TxAbortCol: 0
     TxMultiCol: 0
     TxSingleCol: 0
     TxExcDefer: 0
     TxDefer: 0
     TxLateCol: 0
     RXUnicast: 5753323
     TXUnicast: 3418824
```

But thinking about it again, it looks more like a problem with the external switch port rather than the CPU ones. Maybe there are some differences in the registers for configuring those as well.

I need to get some sleep, but since I have a build environment already I guess I can play with it a bit.

Any pointers to relevant reference information (where did we get all those register details from?) or ideas to try would be appreciated.

luminoso, 2026-10-31:
@tom3q thanks. "EEE status: disabled" means phylib already turned it off and the PHY doesn't advertise it, so EEE is out. Your counters show no FCS or alignment errors, so the line doesn't look noisy either: the PHY just reports link loss and gets it back on the next 1 s poll.

I found one PHY setting that 23.05 and DSA set differently. On my C2600 with 23.05.5, every switch PHY has MII_CTRL1000 (reg 9) at 0x0200, and the WAN PHY resolved as slave. Under DSA the QCA8337 PHY driver also sets "prefer master" (bit 10, "following original QCA sourcecode"), so the WAN PHY now asks to be the clock master. If the device on the other end also prefers master, the two pick roles at random on each negotiation. I can't say that causes the drops. It is cheap to test, though, because it needs no rebuild:

```sh
ethtool wan | grep -i master-slave
ethtool -s wan master-slave preferred-slave
ethtool wan | grep -i master-slave
```

The first command shows what the link resolved to now. The second renegotiates the link once (one short drop) with 23.05's preference. Going by your log, an hour or two without a drop would mean something. A reboot puts it back.

The flow control build is still worth running either way, since it is the other open question in this thread. It only changes the CPU ports (0x007c and 0x0094 to 0x0000014e), not the WAN PHY. If you do both, one at a time, please, so we know which one helped.

tom3q, 2026-11-01:
Turns out it might be an EEE thing after all. After brainstorming the switch/PHY registers with Gemini (Argon looks really good), it seems like even though `ethtool --show-eee wan` claims EEE is disabled, the registers seem to have the relevant bits enabled - looks like a qca8k driver bug.

I was able to modify the registers by hand with mdio and so far got a stable link for 1 hour. Will keep monitoring and if it stays stable, move on to figuring out why the driver doesn't properly clear those bits when EEE is disabled by the user.

tom3q, 2026-11-01:
For anyone interested, here's a script that should set all the manual quirks that I'm currently testing:

```bash
mmd_read() {
    local bus="$1"
    local phy="$2"
    local devad="$3"
    local reg="$4"
    mdio "$bus" phy "$phy" raw 0x0d "$devad"
    mdio "$bus" phy "$phy" raw 0x0e "$reg"
    mdio "$bus" phy "$phy" raw 0x0d $(( 0x4000 | devad ))

    local val=$(mdio "$bus" phy "$phy" raw 0x0e)
    printf "PHY %s MMD %-2s Reg %-6s = %s\n" "$phy" "$devad" "$reg" "$val"
}

qca8k_read() {
    local bus="$1"
    local reg=$(( $2 ))
    local reg_name="$3"
    local page=$(( (reg >> 9) & 0x3ff ))
    local phy=$(( 0x10 | ((reg >> 6) & 0x7) ))
    local lo_reg=$(( (reg >> 1) & 0x1f ))
    local hi_reg=$(( lo_reg | 1 ))

    mdio "$bus" phy 0x18 raw 0 "$page"
    local lo=$(mdio "$bus" phy "$phy" raw "$lo_reg")
    mdio "$bus" phy 0x18 raw 0 "$page"
    local hi=$(mdio "$bus" phy "$phy" raw "$hi_reg")
    local val=$(printf "0x%08x\n" "$(( (hi << 16) | lo ))")
    printf "Reg 0x%04x (%s) = %s\n" "$reg" "$reg_name" "$val"
}

qca8k_write() {
    local bus="$1"
    local reg=$(( $2 ))
    local val=$(( $3 ))
    local page=$(( (reg >> 9) & 0x3ff ))
    local phy=$(( 0x10 | ((reg >> 6) & 0x7) ))
    local lo_reg=$(( (reg >> 1) & 0x1f ))
    local hi_reg=$(( lo_reg | 1 ))
    local lo=$(( val & 0xffff ))
    local hi=$(( (val >> 16) & 0xffff ))
    
    mdio "$bus" phy 0x18 raw 0 "$page"
    mdio "$bus" phy "$phy" raw "$lo_reg" "$lo"
    mdio "$bus" phy "$phy" raw "$hi_reg" "$hi"
}

dump_settings() {
    qca8k_read 37000000.mdio-mii 0x0100 "EEE_CTRL"
    
    mmd_read 37000000.mdio-mii 4 7 0x0013
    mmd_read 37000000.mdio-mii 4 7 0x003c
    
    # Helper one-liner to dump all critical QCA8337 PHY 4 Debug Registers
    for dbg in 0x0b 0x12; do
        mdio 37000000.mdio-mii phy 4 raw 0x1d "$dbg"
        val=$(mdio 37000000.mdio-mii phy 4 raw 0x1e)
        printf "PHY 4 Debug Reg %-4s = %s\n" "$dbg" "$val"
    done
}

# Print current configuration
echo Settings before applying quirks:
dump_settings

# 1. Disable QCA8337 Switch MAC LPI on all ports (0x0100 -> 0x00000000)
qca8k_write 37000000.mdio-mii 0x0100 0x00000000

# Clear EEE Advertisement (MMD 7 Reg 0x003c = 0x0000)
mdio 37000000.mdio-mii phy 4 raw 0x0d 0x0007
mdio 37000000.mdio-mii phy 4 raw 0x0e 0x003c
mdio 37000000.mdio-mii phy 4 raw 0x0d 0x4007
mdio 37000000.mdio-mii phy 4 raw 0x0e 0x0000

# Disable Power-Saving Hibernation (Debug Reg 0x0b = 0x3c40)
mdio 37000000.mdio-mii phy 4 raw 0x1d 0x0b
mdio 37000000.mdio-mii phy 4 raw 0x1e 0x3c40

# Clear sticky interrupt status bits
mdio 37000000.mdio-mii phy 4 raw 0x13

echo
echo Restaring auto-negotiation
# Restart auto-negotiation with 1000M + Full Duplex + AN Enable
mdio 37000000.mdio-mii phy 4 raw 0x00 0x1340
sleep 5

# Verify the configuration
echo
echo Settings after applying quirks:
dump_settings
```

but I haven't tested it, because I don't want to touch the hardware while watching the link stability at the moment.

luminoso, 2026-11-01:
@tom3q correction to my last comment, please skip the 0x00001550 test. I found the EEE_CTRL table in a public QCA8334 datasheet (80-Y0619-1 Rev. A, table 5-39, same register block as the QCA8337). Bits 6 and 8 are `LPI_EN_2` and `LPI_EN_3`, "LPI enable", with a reset value of 1. So the qca8k name is the right one and swconfig's `DISABLE_PHY` is the backwards one. 0x00001550 is the reset default with LPI on, and writing 0 as your script does turns it off. Your direction was right and mine was wrong.

The missing `BIT()` still stands. For port 5 the driver's mask is 12, which hits bits 2 and 3 (`EEE_LLDP_TO_CPU_EN` and `EEE_CPU_CHANGE_EN`) instead of `LPI_EN` at bit 12.

I also found why ethtool says disabled while the PHY advertises EEE. phylib keeps two flags:

- `phy_probe()` reads the PHY's default EEE advertisement, before `qca83xx_config_init()` clears it, and sets `phydev->eee_enabled` because it is non-zero. Autonegotiation then writes that advertisement back to MMD 7.0x3c.
- ethtool reports `phydev->eee_cfg.eee_enabled`, which stays false because qca8k never calls `phy_support_eee()`. So `ethtool --set-eee wan eee off` sees "already off" and never reaches the kernel.

6.18 on main does the same thing: qca8k has no phylink LPI ops, so phylink skips both its EEE branches.

23.05 avoided all of this through the advertisement. My C2600 reads switch ID 0x1302 (revision 2). For that revision, `ar8327_phy_fixup()` writes 0 to MMD 7.0x3c and nothing writes it back. EEE_CTRL stays at 0x00001550 there, and swconfig shows no EEE on the WAN link.

That gives a test that needs no raw MDIO at all. After a reboot:

```sh
ethtool --set-eee wan eee on
ethtool --set-eee wan eee off
```

The second command does reach the kernel, sets `phydev->eee_enabled` to false and renegotiates with an empty advertisement. That matches 23.05: advertisement 0, LPI bits at their default. The "on" step also sets bits 2 and 3 through the bad mask, and the "off" step clears them again. Your `mmd_read ... 4 7 0x003c` should read 0x0000 afterwards. If the link holds like that, the driver fix is to keep EEE off at the PHY, and EEE_CTRL can stay as it is.

luminoso, 2026-11-01:
@tom3q good find. An hour without a drop is already longer than any gap in your log. Could you post the "before" values, especially 0x0100? They show what the DSA driver leaves in the hardware.

0x0100 (EEE_CTRL) is where 23.05 turned EEE off. The swconfig driver calls bits 4, 6, 8, 10 and 12 `AR8327_EEE_CTRL_DISABLE_PHY(0..4)`, one per PHY, where a set bit means EEE off, and `ar8327_sw_hw_apply()` sets all five. My C2600 on 23.05.5 reads 0x00001550. Exactly those bits.

The DSA driver never writes that register at boot. Only `ethtool --set-eee` reaches it, and yours stopped at "eee unmodified, ignoring". Even then the macro looks wrong: `QCA8K_REG_EEE_CTRL_LPI_EN(_i)` is `((_i + 1) * 2)` with no `BIT()`, so for port 5 (wan) it masks bits 2 and 3 instead of bit 12, and it calls them LPI enable where swconfig calls the same bits PHY disable. One name is backwards. swconfig ran stable for years.

So the first step of your script does the opposite of 23.05. Writing 0x00000000 clears all five disable bits, and your stable hour may come from the advertisement or the hibernation change, which 23.05 never touched. To test the 23.05 setting on its own, reboot and run only:

```sh
qca8k_write 37000000.mdio-mii 0x0100 0x00001550
```

If the link holds, the fix is one register in the driver. If not, add your other changes back one at a time.

One caution about raw writes to PHY 0x18: qca8k caches the page it last selected (`qca8k_set_page()` skips the write when the cached page matches), so it won't notice that you changed it. Its next MDIO access to another page can land on the wrong register. Reboot between test runs so each one starts clean.

tom3q, 2026-11-01:
Updated the script in https://github.com/openwrt/openwrt/issues/18979#issuecomment-6554848118 to print more registers and in a more readable way (with register names!), rebooted the device and here are the default register values:

```
PHY 4 Reg 0x00 (MII_BMCR (Basic Mode Control)) = 0x3100
PHY 4 Reg 0x0a (MII_STAT1000 (1000BASE-T Status)) = 0x6800
PHY 4 Reg 0x11 (AT803X_SPECIFIC_STATUS (Resolved Speed / Duplex / Link)) = 0xbc1c
PHY 4 Reg 0x13 (AT803X_INTR_STATUS (Latched Interrupt Status)) = 0xfe40
PHY 4 Debug Reg 0x0b (AT803X_DEBUG_HIB_CTRL (Power-Saving Hibernation)) = 0xbc40
PHY 4 Debug Reg 0x12 (QCA8327_DEBUG_MANU_CTRL_EN (Green/DAC Amplitude)) = 0x4c04
PHY 4 MMD 7 Reg 0x003c (MDIO_AN_EEE_ADV (EEE Advertisement)) = 0x0006
PHY 4 MMD 7 Reg 0x003d (MDIO_AN_EEE_LPABLE (EEE Link Partner Ability)) = 0x0006
qca8k Reg 0x0008 (QCA8K_REG_PORT5_PAD_CTRL) = 0x01000000
qca8k Reg 0x0090 (QCA8K_REG_PORT_STATUS(5)) = 0x00000d7e
qca8k Reg 0x0100 (QCA8K_REG_EEE_CTRL) = 0x00001550
```

luminoso, 2026-11-01:
@tom3q thanks, that dump fits. For comparison, the same WAN PHY registers on my C2600 running 23.05.5:

```
PHY 4 Reg 0x0a (MII_STAT1000)          = 0x3c00   (yours 0x6800)
PHY 4 Debug Reg 0x0b (HIB_CTRL)        = 0xbc40   (same)
PHY 4 MMD 7 Reg 0x003c (EEE_ADV)       = 0x0000   (yours 0x0006)
PHY 4 MMD 7 Reg 0x003d (EEE_LPABLE)    = 0x0006   (same)
qca8k Reg 0x0100 (EEE_CTRL)            = 0x00001550 (same)
```

Hibernation is identical, so you can drop that part of the script. Two differences are left on the WAN PHY. 23.05 advertises no EEE, and it resolved as slave (bit 14 clear) where yours is master. Different router and different link partner, so the master/slave bit is a hint at most. The EEE advertisement is the one to test first, with the `ethtool --set-eee wan eee on` / `off` pair. After that, MMD 7.0x3c should read 0x0000 like mine.

tom3q, 2026-11-01:
Thanks a lot for all the help. I think we're onto something here indeed.

So far 2 hours of stable link passed just with the `ethtool --set-eee wan eee on` / `off` toggle

I'm working on patches to fix the EEE_CTRL bit control and initial value of EEE_ADV.

tom3q, 2026-11-01:
I've created patches to fix the EEE issues - still testing, but for anyone interested, here is my branch: https://github.com/tom3q/openwrt/commits/r7800-qca8k-fixes/

luminoso, 2026-11-01:
@tom3q great result. I had a look at bdf772d on r7800-qca8k-fixes, a few notes before it goes into a PR.

**715-02 (the fix for your drops).** `phydev->eee_enabled` no longer exists on main. In 6.18 phylib keeps only `phydev->eee_cfg.eee_enabled`, so the patch won't build there, and OpenWrt wants the change on main first. `phy_disable_eee()` works on both 6.12 and 6.18 and does what you want: it zeroes the advertisement mask, so the next autonegotiation writes 0 to MMD 7.0x3c. On 6.18 it also marks every mode disabled, so userspace can't turn it back on. Two PHY drivers in 6.18 already call it from `config_init()` for this reason: `mediatek/mtk-ge.c`, whose comment describes exactly this override, and `dp83867.c`. So the rev 2 case would become:

```c
	case 2:
		phy_write_mmd(phydev, MDIO_MMD_AN, MDIO_AN_EEE_ADV, 0x0);
		/* phylib rewrites the advertisement from what phy_probe() read */
		phy_disable_eee(phydev);
		fallthrough;
```

**715-01.** The QCA8334 datasheet (80-Y0619-1 Rev. A, table 5-39) names these bits `LPI_EN_x`, LPI enable, with reset value 1. Renaming them to `LPI_DIS` and inverting the logic goes against that. Did you see on hardware that a set bit disables LPI? If not, I'd keep the name and only add the missing `BIT()`. It also isn't needed to stop your drops: with the advertisement at 0, EEE never gets negotiated whatever the MAC bit says. I'd make it a separate patch.

Happy to build-test on main when it's up.

tom3q, 2026-11-02:
Thanks for the comments! Those are just raw changes I have for backup and sharing purposes. I'm going to investigate them a bit more to further confirm their correctness and send as proper patches straight to the upstream kernel mailing lists - after that we can get them from there into OpenWRT.

> 22\. `phy_disable_eee()` works on both 6.12 and 6.18 and does what you want: it zeroes the advertisement mask, so the next autonegotiation writes 0 to MMD 7.0x3c. On 6.18 it also marks every mode disabled, so userspace can't turn it back on.

I'm not yet sure that we really want to completely block EEE from being enabled on a request. That's not what swconfig did and I suppose there could be some success cases around? I think this is better left to a user decision, with a safe default.

> 22\. **715-01.** The QCA8334 datasheet (80-Y0619-1 Rev. A, table 5-39) names these bits `LPI_EN_x`, LPI enable, with reset value 1. Renaming them to `LPI_DIS` and inverting the logic goes against that. Did you see on hardware that a set bit disables LPI? If not, I'd keep the name and only add the missing `BIT()`. It also isn't needed to stop your drops: with the advertisement at 0, EEE never gets negotiated whatever the MAC bit says. I'd make it a separate patch.

I think this needs a bit more investigation. As we already noticed in the [old swconfig code](https://github.com/openwrt-mirror/openwrt/blob/9b4650b3b92e6246b986ac9e3d7c2a80d66b805b/target/linux/generic/files/drivers/net/phy/ar8327.c#L1100), the bit is set when EEE is to be disabled. This wouldn't be the first time in history when there datasheet is misleading or even wrong.

In fact, with both of my patches, I just switched EEE on with ethtool around 20 minutes ago, and it seems like the other PHY bits indicate that EEE and LPI is working as expected and the link seems to be stable so far.

I think we should post a patch upstream proposing to invert the bit, based on the swconfig behavior, and see if anyone with closer connections to QCA could confirm the real behavior.

For now I'll leave it running and monitor. With EEE off, it was all stable yesterday evening and overnight - 11 hours+.

tom3q, 2026-11-02:
Update on the EEE_CTRL register bits: I believe the datasheet is correct after all. The reason is that if the corresponding bit is cleared, I can only see 0x0500 in the MDIO_STAT1 register, but toggling the bit on makes it go to 0x0f00. The meaning of this becomes clear when looking at the meaning of the bits:

Bit 11 (0x0800 - TX_LPI_RCVD) - TX LPI request received
Bit 10 (0x0400 - RX_LPI_RCVD) - RX LPI request received
Bit 9 (0x0200 - TX_LPI_IND) - TX LPI active indicator
Bit 8 (0x0100 - RX_LPI_IND) - RX LPI active indicator

So the bit being disabled in EEE_CTRL prevents the MAC from sending TX LPI requests, confirming the polarity in the current qca8k driver code being correct.

The bad news is that after the TX LPI activated, the link started dropping again. I guess that confirms that EEE causes problems with this switch and/or my upstream router.

I'm modifying my patch to only fix the missing BIT() macro usage and going to give it a try again.

tom3q, 2026-11-02:
New day, new findings:

1. I've tested an image without `net: dsa: qca8k: don't force flow control on CPU ports` and the link seems to be stable too, so maybe this is not needed for R7800, or at least not for my unit or I'm yet to see a problem without it, as I've only tested ~5 hours.
2. I've tried an image built from the main branch and it turns out it doesn't have the bug causing ethtool to see the EEE status as disabled, despite the hardware having it enabled. So adding
   ```
   config device
           option name 'wan'
           option eee '0'
   ```
   to /etc/config/network was enough to get the hardware configured correctly at boot.

FWIW, both images included the RGMII voltage fixes.

So I think I'm going to abandon the two patches for the boot/probe-time initialization and send the EEE_CTRL macro fix upstream - although it doesn't seem to be strictly necessary to fix my problems, because as long as EEE advertisement is not enabled, the MAC LPI enable status shouldn't matter.

luminoso, 2026-11-09:
@banburybill @botischroeder @BIGFAAT 

How are things going? Everything stable?

botischroeder, 2026-11-09:
On my devices everything is working as expected. No errors occured until now.

tom3q, 2026-11-09:
So far almost 7 days of uptime with no issues on a build from main with patches and config as per https://github.com/openwrt/openwrt/issues/18979#issuecomment-6584019665.

banburybill, 2026-11-09:
I had some painters in last week and my C2600 lost power for a moment while rearranging the furniture, dammit. It's been rock solid with your image, but uptime currently at 8 days.

BIGFAAT, 2026-11-13:
So far stable on 2 devices. I have 1 camera each on the WAN interface that would fallback to WiFi if needed.

Thankfully for now it didn't happen. The patches has been working so far.

banburybill, 2026-11-14:
Still stable for me too. I'm happy this is ready to go.

Ansuel, 2026-11-14:
Can I get a summary of this?

luminoso, 2026-11-15:
@Ansuel sure. The fix is #26259.

**Problem.** On ipq806x boards with a QCA8337, WAN stops receiving after hours or days, starting with 24.10. The link stays up, TX keeps counting, and `/etc/init.d/network restart` brings it back. 23.05 was fine on the same hardware.

**Cause.** The DSA conversion (337e36e0ef) dropped the swconfig initvals, and two switch registers no longer match 23.05:

- MAC_PWR_SEL (0x00e4). `qca8k_setup_mac_pwr_sel()` picks the RGMII voltage from the SoC compatible, so the C2600, R7500, D7800, EA8500 and others get 0x000aa545 (RGMII0 at 1.8 V). Their initvals had 0x0006a545 (RGMII1 at 1.8 V) since 66ffd0ddf9 in 2016, which fixed corruption on the port 0 interface.
- PORT0_PAD_CTRL RX delay: 1 under DSA, 2 in 23.05.

kylex2 found both by dumping registers on 23.05 and 24.10.

**Fix.** Two boolean properties, `qca,mac-pwr-sel-rgmii0-1p8v` and `qca,mac-pwr-sel-rgmii1-1p8v`, set per board from its 23.05 initvals, plus `rx-internal-delay-ps = <2000>` on the 17 boards whose initvals used delay 2. A DTB without either property keeps the old compatible-based value, as @tom3q asked in review.

**Testing.** Testers have run the series since 28-29 October, mostly on 25.12.6 test images built from the backport:

- @banburybill, C2600: stable. The longest unbroken run is about 13 days, after a power cut. Before the fix it froze within 12 days at most.
- @botischroeder, a VR2600v on the test image and two C2600 on a main build: no errors as of 9 November.
- @BIGFAAT, two C2600: stable as of 13 November.
- @tom3q, R7800: 7 days on a main build with the series.

All of them still have flow control forced on the CPU ports (0x17e), so kylex2's "don't force flow control" change isn't needed and isn't in the PR. These images were built before the fallback was added, which in-tree boards never reach.

**Question for you.** The R7800 and the other ipq8065 boards list both compatibles, so DSA sets both voltage bits today. The 23.05 DTS had RGMII0 only, tom3q read 0x000aa545 on 23.05.6, and the PR goes back to that. If you know a reason to keep both bits there, I'll change it.

**Separate issue.** tom3q's R7800 also dropped WAN link every 10-30 minutes. That turned out to be EEE: the WAN PHY advertises it under DSA, while 23.05 cleared the advertisement. On main `option eee '0'` on the wan device fixes it. On 25.12 (6.12) `ethtool` reports EEE as disabled while it is on. tom3q is sending a fix for the missing `BIT()` in `QCA8K_REG_EEE_CTRL_LPI_EN()` upstream. None of this is in #26259.

Thanks to @banburybill, @botischroeder, @BIGFAAT and @tom3q for the testing, and to @kylex2 for the original work. #26259 is now out of draft. The 25.12 backport is rebased and built, and I'll open it as soon as #26259 is merged.

datadata95, 2026-11-15:
Hope it will be merged into the official build. I experienced the same issues as well, I think last year when I tried to update to version 24, and the wan dropped every now and then, but wifi stayed steady. Went back to 23 but a community build and now trying kylex build, but need to wait for the official merge, since kmod_tun, kmod_wireguard etc seems missing from these test builds.