openwrt/openwrt #18983: ZBT-Z800AX: no connection when using plain lan port

mriedel87, 2025-05-31:
### Describe the bug

There is no connection/ traffic (verified with tcpdump) on the interface, when using a lan port which is not connected to a bridge.

### OpenWrt version

r29690-d95e8a59bb

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Zbtlink ZBT-Z800AX

### Image kind

Self-built image

### Steps to reproduce

- attach lan cable to any port (e.g. lan4)
- remove chosen lan port  from bridge device
- bring interface up
- verify there is no traffic on the interface

### Actual behaviour

Zero packages are received nor transmitted

### Expected behaviour

Traffic on the interface

### Diffconfig

```text
Used imagebuilder, so no self-built sources
```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-05-31:
You can still get some packets like dhcp and rtsol IF you connect something.

mriedel87, 2025-05-31:
Have you tested it? 
The other end of the cable is (of course) attached to a PC. The PC is sending dhcp requests, but nothing get's captured on the router side.