openwrt/openwrt #18984: ZBT-Z800AX: rootfs partition too small

mriedel87, 2025-05-31:
### Describe the bug

The device has 256 MiB NAND and the rootfs is only 52MiB big.
In the original dts it's 116 MiB.

### OpenWrt version

r29690-d95e8a59bb

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Zbtlink ZBT-Z800AX

### Image kind

Self-built image

### Actual behaviour

OpenWrt dts:

	nand@0 {
		reg = <0>;
		nand-ecc-strength = <8>;
		nand-ecc-step-size = <512>;
		nand-bus-width = <8>;

		partitions {
			compatible = "fixed-partitions";
			#address-cells = <1>;
			#size-cells = <1>;

			partition@0 {
				label = "rootfs";
				reg = <0x0000000 0x3400000>;
			};

			partition@3400000 {
				label = "0:wififw";
				reg = <0x3400000 0x0800000>;
				read-only;
			};

			partition@3c00000 {
				label = "rootfs_1";
				reg = <0x3c00000 0x3400000>;
			};

			partition@7000000 {
				label = "0:wififw_1";
				reg = <0x7000000 0x0800000>;
				read-only;
			};
		};
	};

### Expected behaviour

Dumped dts before flashing OpenWrt:

			nandcs@0 {
				reg = <0x00>;
				compatible = "qcom,nandcs";
				nand-ecc-strength = <0x04>;
				nand-ecc-step-size = <0x200>;
				nand-bus-width = <0x08>;
				#address-cells = <0x01>;
				#size-cells = <0x01>;

				partition@7c00000 {
					reg = <0x7c00000 0x7400000>;
					label = "rootfs";
				};

				partition@0 {
					reg = <0x00 0x7400000>;
					label = "rootfs_1";
				};

				partition@f000000 {
					reg = <0xf000000 0x800000>;
					label = "0:WIFIFW";
				};

				partition@7400000 {
					reg = <0x7400000 0x800000>;
					label = "0:WIFIFW_1";
				};
			};

### Diffconfig

```text
Used imagebuilder, so no self-built sources
```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-05-31:
Please add package set from imagebuilder.

mriedel87, 2025-06-02:
I doubt it that there is a new HW revision. 
In the initial [commit](https://github.com/openwrt/openwrt/commit/9cef2bc224b9bda19a8a79df6d1de9560c081ce2) it states 

> Flash2: MX30UF2G28AD 256MB

For completeness, the dts dump is attached.

[z800ax.dts.txt](https://github.com/user-attachments/files/20547327/z800ax.dts.txt)