openwrt/openwrt #18995: Wifi Access Point crashes after 40 - 50 GB

ob45, 2025-06-01:
### Describe the bug

Since the release of version 24.10, I've been experiencing what I initially thought were random router crashes. The router would suddenly stop broadcasting a Wi-Fi signal or stop providing internet access over the access point. Devices would show "Connected, No Internet" or not detect the AP at all.

At first, the issue seemed to occur randomly - sometimes while I was downloading files on my PC, and other times during light evening browsing on social media.

However, after taking a closer look today, I discovered a pattern: the access point crashes or bugs out after about 40 to 50 GB of data transfer. (Download) I noticed this while downloading a game - once I hit that data threshold, I had to keep rebooting the router just to restore Wi-Fi functionality.

### OpenWrt version

r28597-0425664679

### OpenWrt release

24.10.1

### OpenWrt target/subtarget

ramips/mt76x8

### Device

Xiaomi Mi Router 4A (100M International Edition V2)

### Image kind

Official downloaded image

### Steps to reproduce

1. Download and flash the currently avaiable router firmware (24.10.1) for the Xiaomi Mi Router 4A (100M International Edition V2) from the official [firmware selection page](https://firmware-selector.openwrt.org/?version=24.10.1&target=ramips%2Fmt76x8&id=xiaomi_mi-router-4a-100m-intl-v2)

2. Set up an AP

3. Connect to the AP and download some huge files until you hit around 40 to 50 GB data transfer in total (Download)

4.  The AP will crash (Internet and LuCi still avaiable via Ethernet)

### Actual behaviour

After around 40 to 50 GB of data transfer (Download) the AP crashes and stops providing a functional Wifi signal.

Even in LuCi it shows that the Wifi AP is down:

[image]

I have analyzed the System and Kernel Log with ChatGPT in order to try and find the cause on my own but so far it only found these entries, which seem at least somewhat suspicious (Entire logs will be provided at the end):

System Log analysis:

> 1. Multiple hostapd STA disconnection events due to "inactivity", not because of authentication failure. Around 19:45:48 and 19:45:56, your Wi-Fi clients were disconnected:
> 
> STA f8:89:d2:d4:88:49 IEEE 802.11: disassociated due to inactivity
> STA 4a:14:ec:47:e6:ce IEEE 802.11: deauthenticated due to inactivity

> 2. Driver Peculiarity: This line stands out:
> 
> mt7615e 0000:01:00.0: mediatek/mt7663pr2h.bin not found, switching to ..._rebb.bin
> 
> You're missing the main firmware file, falling back to a possibly suboptimal one. This can definitely cause instability under load.

Kernel Log analysis:

> 1. ⚠️ Missing Wi-Fi Firmware Blob
> 
> [   45.612145] mt7615e 0000:01:00.0: mediatek/mt7663pr2h.bin not found, switching to mediatek/mt7663pr2h_rebb.bin
>
>This is a big red flag.
>The device is missing the official firmware blob and is falling back to a possibly limited one.
>Result: This is known to cause Wi-Fi instability — especially under load, resulting in crashes or dropped clients.

> 2. ⚠️ Flash Partition Errors
>
> [    0.882991] OF: Bad cell count for /palmbus@10000000/spi@b00/flash@0/partitions
>
>These appear three times and relate to parsing your SPI flash's partition layout.
>This might affect access to the "factory" partition, which stores Wi-Fi EEPROM calibration data.
>If the EEPROM data is incorrect or inaccessible, it can cause radio misbehavior.

> 3.⚠️ Link Flaps
> 
> [   67.294121] rt3050-esw 10110000.esw: port 0 link down
> [   67.299297] rt3050-esw 10110000.esw: port 4 link down
> [   70.899835] rt3050-esw 10110000.esw: port 0 link up
> [   71.325559] rt3050-esw 10110000.esw: port 4 link up
> 
> These link down/up cycles may indicate brief resets or driver hiccups, but aren’t necessarily fatal.
> If they correlate with data transfer spikes or Wi-Fi issues, the driver or thermal limits may be involved.

I don't know if this helps at all but I just wanted to provide it too just in case

### Expected behaviour

The AP and Wifi functionality should just work normally and not crash after transmitting a certain amount of data.

### Additional info

I am only using IPv4 and 2,5 Ghz Wifi.
https-dns-proxy was installed but not running at the time

[Syslog.txt](https://github.com/user-attachments/files/20541970/Syslog.txt)

[Kernel-Log.txt](https://github.com/user-attachments/files/20541973/Kernel-Log.txt)

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

ob45, 2025-06-01:
Ok, I've been experimenting with ChatGPT and found the following: 

1. The firmware seems to be missing from the package list because I can't find it:
>root@OpenWrt:~# opkg list | grep mt76-firmware-7663
>root@OpenWrt:~# 

2. The firmware also seems to NOT be present on the device itself, only fallback images:

> root@OpenWrt:~# ls /lib/firmware/mediatek/
> mt7663_n9_rebb.bin   mt7663pr2h_rebb.bin

Should I download the firmware from here?
https://github.com/openwrt/mt76/tree/master/firmware

And if yes, which one should I use?

mt7663pr2h.bin or mt7628_e1.bin or mt7628_e2.bin ? 🤔

brada4, 2025-06-02:
Please edit away ALL generated content leaving only hard facts.
Try to reproduce the problem without any doh proxy running.
Only thing observable in ligs is you connecting a client to port 4 after 1.5h uptime

brada4, 2025-06-02:
@mjecov if you have a problem with 7621 device please make a separate report , this is so bloated by chatbot that it is hard to understand the essence.
Yours has 2x RAM, so 10mb doh proxy does not oom, also 7621 soc is quite mainstream, mid-to-low range routers in stores today.

ob45, 2025-06-02:
@brada4 I wouldn't say that. The kernel logs show this:

`[ 45.612145] mt7615e 0000:01:00.0: mediatek/mt7663pr2h.bin not found, switching to mediatek/mt7663pr2h_rebb.bin`

Which means that the release version for this device actually lacks the firmware blob for the 5 Ghz Chip and only has the _rebb.bin Versions, which are older demo firmware versions. (And often don't have all the features)

Thanks to ChatGPT i looked into it and downloaded the mt7663pr2h.bin blob [from the firmware repo](https://github.com/openwrt/mt76/tree/master/firmware) and did a small test yesterday by hitting the router with a 50GB download without having issues....

I am testing it again today with a bigger download and will make tests without the blob again and with the blow and dns-https-proxy running in the next couple days to weed out the actual issue.

I honestly don't know why downloading the firmware for the 5 Ghz would fix an issue regarding the 2.4 Ghz Chip but after some digging I found info, which said that mediatek chips often have quirks like that so idk...

I'll keep testing to check if the missing mt7663pr2h.bin was really the cause...

brada4, 2025-06-02:
FW files are here
https://gitlab.com/kernel-firmware/linux-firmware/-/tree/main/mediatek?ref_type=heads
If some is missing make a PR against respective firmware package.

ob45, 2025-06-03:
@brada4 

It doesn't seem to be a heat issue because I think I found the direction of the issue.
The problem only appears when DoH proxy is running HOWEVER no oom problem exists because I just encontered the issue again and after checking it seems like oom is not the issue:

```
root@OpenWrt:~# dmesg | grep -i kill
root@OpenWrt:~# logread | grep -i oom
root@OpenWrt:~#
```

What I did find is very odd behavior of the sys log where it did not log ANYTHING 2 hours before the crash, which is higly suspicious:

```
Tue Jun  3 00:08:39 2025 daemon.info procd: - init complete -
Tue Jun  3 00:10:58 2025 daemon.info hostapd: phy0-ap0: STA 4a:14:ec:47:e6:ce IEEE 802.11: authenticated
Tue Jun  3 00:10:58 2025 daemon.info hostapd: phy0-ap0: STA 4a:14:ec:47:e6:ce IEEE 802.11: associated (aid 1)
Tue Jun  3 00:10:58 2025 daemon.notice hostapd: phy0-ap0: AP-STA-CONNECTED 4a:14:ec:47:e6:ce auth_alg=open
Tue Jun  3 00:10:58 2025 daemon.info hostapd: phy0-ap0: STA 4a:14:ec:47:e6:ce WPA: pairwise key handshake completed (RSN)
Tue Jun  3 00:10:58 2025 daemon.notice hostapd: phy0-ap0: EAPOL-4WAY-HS-COMPLETED 4a:14:ec:47:e6:ce
Tue Jun  3 00:10:59 2025 daemon.info dnsmasq-dhcp[3156]: DHCPREQUEST(br-lan) 192.168.1.235 4a:14:ec:47:e6:ce
Tue Jun  3 00:10:59 2025 daemon.info dnsmasq-dhcp[3156]: DHCPACK(br-lan) 192.168.1.235 4a:14:ec:47:e6:ce Galaxy-S10
Tue Jun  3 00:12:28 2025 daemon.info hostapd: phy0-ap0: STA f8:89:d2:d4:88:49 IEEE 802.11: authenticated
Tue Jun  3 00:12:28 2025 daemon.info hostapd: phy0-ap0: STA f8:89:d2:d4:88:49 IEEE 802.11: associated (aid 2)
Tue Jun  3 00:12:28 2025 daemon.notice hostapd: phy0-ap0: AP-STA-CONNECTED f8:89:d2:d4:88:49 auth_alg=open
Tue Jun  3 00:12:28 2025 daemon.info hostapd: phy0-ap0: STA f8:89:d2:d4:88:49 WPA: pairwise key handshake completed (RSN)
Tue Jun  3 00:12:28 2025 daemon.notice hostapd: phy0-ap0: EAPOL-4WAY-HS-COMPLETED f8:89:d2:d4:88:49
Tue Jun  3 00:12:28 2025 daemon.info dnsmasq-dhcp[3156]: DHCPREQUEST(br-lan) 192.168.1.152 f8:89:d2:d4:88:49
Tue Jun  3 00:12:28 2025 daemon.info dnsmasq-dhcp[3156]: DHCPACK(br-lan) 192.168.1.152 f8:89:d2:d4:88:49 DESKTOP-I5AUN7S
Tue Jun  3 00:12:46 2025 daemon.err uhttpd[1555]: [info] luci: accepted login on / for root from 192.168.1.152
Tue Jun  3 00:12:58 2025 user.notice https-dns-proxy [3920]: Starting https-dns-proxy 2025.05.11-r1 instances ✓✓✓
Tue Jun  3 00:12:58 2025 user.notice https-dns-proxy [3920]: Updating dnsmasq config ✓
Tue Jun  3 00:12:59 2025 daemon.info dnsmasq[3156]: exiting on receipt of SIGTERM
Tue Jun  3 00:13:05 2025 user.notice https-dns-proxy [3920]: Restarting dnsmasq on_config_update ✓
Tue Jun  3 00:13:05 2025 daemon.info dnsmasq[4258]: started, version 2.90 cachesize 1000
Tue Jun  3 00:13:05 2025 daemon.info dnsmasq[4258]: DNS service limited to local subnets
Tue Jun  3 00:13:05 2025 daemon.info dnsmasq[4258]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack no-ipset no-nftset no-auth no-cryptohash no-DNSSEC no-ID loop-detect inotify dumpfile
Tue Jun  3 00:13:05 2025 daemon.info dnsmasq[4258]: UBus support enabled: connected to system bus
Tue Jun  3 00:13:05 2025 daemon.info dnsmasq-dhcp[4258]: DHCP, IP range 192.168.1.100 -- 192.168.1.249, lease time 12h
Tue Jun  3 00:13:05 2025 daemon.info dnsmasq[4258]: using nameserver 127.0.0.1#5050
Tue Jun  3 00:13:05 2025 daemon.info dnsmasq[4258]: using nameserver 127.0.0.1#5051
Tue Jun  3 00:13:05 2025 daemon.info dnsmasq[4258]: using nameserver 127.0.0.1#5052
Tue Jun  3 00:13:05 2025 daemon.info dnsmasq[4258]: using only locally-known addresses for test
Tue Jun  3 00:13:05 2025 daemon.info dnsmasq[4258]: using only locally-known addresses for onion
Tue Jun  3 00:13:05 2025 daemon.info dnsmasq[4258]: using only locally-known addresses for localhost
Tue Jun  3 00:13:05 2025 daemon.info dnsmasq[4258]: using only locally-known addresses for local
Tue Jun  3 00:13:05 2025 daemon.info dnsmasq[4258]: using only locally-known addresses for invalid
Tue Jun  3 00:13:05 2025 daemon.info dnsmasq[4258]: using only locally-known addresses for bind
Tue Jun  3 00:13:05 2025 daemon.info dnsmasq[4258]: using only locally-known addresses for use-application-dns.net
Tue Jun  3 00:13:05 2025 daemon.info dnsmasq[4258]: using only locally-known addresses for mask-h2.icloud.com
Tue Jun  3 00:13:05 2025 daemon.info dnsmasq[4258]: using 2 more local addresses
Tue Jun  3 00:13:06 2025 daemon.info dnsmasq[4258]: read /etc/hosts - 12 names
Tue Jun  3 00:13:06 2025 daemon.info dnsmasq[4258]: read /tmp/hosts/dhcp.cfg01411c - 2 names
Tue Jun  3 00:13:06 2025 daemon.info dnsmasq-dhcp[4258]: read /etc/ethers - 0 addresses
Tue Jun  3 00:13:06 2025 user.notice https-dns-proxy [3920]: Setting trigger for wan ✓
Tue Jun  3 02:38:52 2025 kern.info kernel: [ 9082.471952] rt3050-esw 10110000.esw: port 4 link up
Tue Jun  3 02:38:57 2025 daemon.warn dnsmasq[4258]: ignoring query from non-local network 169.254.44.171 (logged only once)
Tue Jun  3 02:38:57 2025 daemon.info dnsmasq-dhcp[4258]: DHCPDISCOVER(br-lan) 192.168.1.214 90:2e:16:e2:bb:a9
Tue Jun  3 02:38:57 2025 daemon.info dnsmasq-dhcp[4258]: DHCPOFFER(br-lan) 192.168.1.214 90:2e:16:e2:bb:a9
Tue Jun  3 02:38:57 2025 daemon.info dnsmasq-dhcp[4258]: DHCPREQUEST(br-lan) 192.168.1.214 90:2e:16:e2:bb:a9
Tue Jun  3 02:38:57 2025 daemon.info dnsmasq-dhcp[4258]: DHCPACK(br-lan) 192.168.1.214 90:2e:16:e2:bb:a9 DESKTOP-I5AUN7S
[... cut ...]
Tue Jun  3 02:42:34 2025 daemon.info hostapd: phy0-ap0: STA f8:89:d2:d4:88:49 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Tue Jun  3 02:42:37 2025 daemon.notice hostapd: phy0-ap0: AP-STA-DISCONNECTED 4a:14:ec:47:e6:ce
Tue Jun  3 02:42:37 2025 daemon.info hostapd: phy0-ap0: STA 4a:14:ec:47:e6:ce IEEE 802.11: disassociated due to inactivity
Tue Jun  3 02:42:38 2025 daemon.info hostapd: phy0-ap0: STA 4a:14:ec:47:e6:ce IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
```

brada4, 2025-06-03:
You can join other issue, this with 90% chatbot "analysis" is totally unreadable. Just filter out doh proxy oom-s ;-)

ob45, 2025-06-05:
How would I go on to monitor it? What would I look out for? @brada4