openwrt/openwrt #18999: Wifi Access Point crashes after downloading/uploading huge amount of data

mjecov, 2025-06-02:
### Describe the bug

Since the release of version 24.10, I've been experiencing what I initially thought were random router crashes. The router would suddenly stop broadcasting a Wi-Fi signal or stop providing internet access over the access point. Devices would show "Connected, No Internet" or not detect the AP at all.

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

ramips/mt7621

### Device

Xiaomi Mi Router 4A Gigabit Edition

### Image kind

Official downloaded image

### Steps to reproduce

1. Connect to wifi
2. Try to upload huge amount of data (40-50 GB)

### Actual behaviour

Sometimes all devices are disconnected from Wi-Fi and can't reconnect until the router is rebooted.
Occasionally, the access point disappears entirely.
In some cases, devices remain connected but lose internet access.

### Expected behaviour

WiFi should work smooth without limitation of sending data

### Additional info

I use DoH proxy, but after disabling it is still possible to reproduce the problem.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

mjecov, 2025-06-02:
I had to rolled back to 23.05. The problem is not reproducible on 23.05. Everything works good

brada4, 2025-06-02:
Some owner of particular device has to try to re-create the issue then, more detailed wifi debug logs may help ie what happened out of line right before it got bad.

ob45, 2025-06-03:
@mjecov Hmm... are you really running https-dns-proxy? Or did you just copy it from my posting.

I'm asking because I hit my router with 50 GB yesterday with https-dns-proxy disabled and today with https-dns-proxy disabled and the other chip firmware and it was fine.

After that I started https-dns-proxy back up and after 45 GB the AP crashed again.

Oddly enough it doesn't seem to be a oom problem because I still have 30% free...

[image]

This shit is crazy

ob45, 2025-06-03:
I just looked at the logs and I had nothing happening for 2 hours which is higly suspicious:

```
Tue Jun  3 00:08:39 2025 daemon.info procd: - init complete -
Tue Jun  3 00:10:58 2025 daemon.info hostapd: phy0-ap0: STA 4a:14:ec:47:e6:ce IEEE 802.11: authenticated
Tue Jun  3 00:10:58 2025 daemon.info hostapd: phy0-ap0: STA 4a:14:ec:47:e6:ce IEEE 802.11: associated (aid 1)
Tue Jun  3 00:10:58 2025 daemon.notice hostapd: phy0-ap0: AP-STA-CONNECTED 4a:14:ec:47:e6:ce auth_alg=open
Tue Jun  3 00:10:58 2025 daemon.info hostapd: phy0-ap0: STA 4a:14:ec:47:e6:ce WPA: pairwise key handshake completed (RSN)
Tue Jun  3 00:10:58 2025 daemon.notice hostapd: phy0-ap0: EAPOL-4WAY-HS-COMPLETED 4a:14:ec:47:e6:ce
Tue Jun  3 00:10:59 2025 daemon.info dnsmasq-dhcp[3156]: DHCPREQUEST(br-lan) 192.168.1.235 4a:14:ec:47:e6:ce
Tue Jun  3 00:10:59 2025 daemon.info dnsmasq-dhcp[3156]: DHCPACK(br-lan) 192.168.1.235 4a:14:ec:47:e6:ce Galaxy-S10
Tue Jun  3 00:12:28 2025 daemon.info hostapd: phy0-ap0: STA f8:89:d2:d4:88:49 IEEE 802.11: authenticated
Tue Jun  3 00:12:28 2025 daemon.info hostapd: phy0-ap0: STA f8:89:d2:d4:88:49 IEEE 802.11: associated (aid 2)
Tue Jun  3 00:12:28 2025 daemon.notice hostapd: phy0-ap0: AP-STA-CONNECTED f8:89:d2:d4:88:49 auth_alg=open
Tue Jun  3 00:12:28 2025 daemon.info hostapd: phy0-ap0: STA f8:89:d2:d4:88:49 WPA: pairwise key handshake completed (RSN)
Tue Jun  3 00:12:28 2025 daemon.notice hostapd: phy0-ap0: EAPOL-4WAY-HS-COMPLETED f8:89:d2:d4:88:49
Tue Jun  3 00:12:28 2025 daemon.info dnsmasq-dhcp[3156]: DHCPREQUEST(br-lan) 192.168.1.152 f8:89:d2:d4:88:49
Tue Jun  3 00:12:28 2025 daemon.info dnsmasq-dhcp[3156]: DHCPACK(br-lan) 192.168.1.152 f8:89:d2:d4:88:49 DESKTOP-I5AUN7S
Tue Jun  3 00:12:46 2025 daemon.err uhttpd[1555]: [info] luci: accepted login on / for root from 192.168.1.152
Tue Jun  3 00:12:58 2025 user.notice https-dns-proxy [3920]: Starting https-dns-proxy 2025.05.11-r1 instances ✓✓✓
Tue Jun  3 00:12:58 2025 user.notice https-dns-proxy [3920]: Updating dnsmasq config ✓
Tue Jun  3 00:12:59 2025 daemon.info dnsmasq[3156]: exiting on receipt of SIGTERM
Tue Jun  3 00:13:05 2025 user.notice https-dns-proxy [3920]: Restarting dnsmasq on_config_update ✓
Tue Jun  3 00:13:05 2025 daemon.info dnsmasq[4258]: started, version 2.90 cachesize 1000
Tue Jun  3 00:13:05 2025 daemon.info dnsmasq[4258]: DNS service limited to local subnets
Tue Jun  3 00:13:05 2025 daemon.info dnsmasq[4258]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack no-ipset no-nftset no-auth no-cryptohash no-DNSSEC no-ID loop-detect inotify dumpfile
Tue Jun  3 00:13:05 2025 daemon.info dnsmasq[4258]: UBus support enabled: connected to system bus
Tue Jun  3 00:13:05 2025 daemon.info dnsmasq-dhcp[4258]: DHCP, IP range 192.168.1.100 -- 192.168.1.249, lease time 12h
Tue Jun  3 00:13:05 2025 daemon.info dnsmasq[4258]: using nameserver 127.0.0.1#5050
Tue Jun  3 00:13:05 2025 daemon.info dnsmasq[4258]: using nameserver 127.0.0.1#5051
Tue Jun  3 00:13:05 2025 daemon.info dnsmasq[4258]: using nameserver 127.0.0.1#5052
Tue Jun  3 00:13:05 2025 daemon.info dnsmasq[4258]: using only locally-known addresses for test
Tue Jun  3 00:13:05 2025 daemon.info dnsmasq[4258]: using only locally-known addresses for onion
Tue Jun  3 00:13:05 2025 daemon.info dnsmasq[4258]: using only locally-known addresses for localhost
Tue Jun  3 00:13:05 2025 daemon.info dnsmasq[4258]: using only locally-known addresses for local
Tue Jun  3 00:13:05 2025 daemon.info dnsmasq[4258]: using only locally-known addresses for invalid
Tue Jun  3 00:13:05 2025 daemon.info dnsmasq[4258]: using only locally-known addresses for bind
Tue Jun  3 00:13:05 2025 daemon.info dnsmasq[4258]: using only locally-known addresses for use-application-dns.net
Tue Jun  3 00:13:05 2025 daemon.info dnsmasq[4258]: using only locally-known addresses for mask-h2.icloud.com
Tue Jun  3 00:13:05 2025 daemon.info dnsmasq[4258]: using 2 more local addresses
Tue Jun  3 00:13:06 2025 daemon.info dnsmasq[4258]: read /etc/hosts - 12 names
Tue Jun  3 00:13:06 2025 daemon.info dnsmasq[4258]: read /tmp/hosts/dhcp.cfg01411c - 2 names
Tue Jun  3 00:13:06 2025 daemon.info dnsmasq-dhcp[4258]: read /etc/ethers - 0 addresses
Tue Jun  3 00:13:06 2025 user.notice https-dns-proxy [3920]: Setting trigger for wan ✓
Tue Jun  3 02:38:52 2025 kern.info kernel: [ 9082.471952] rt3050-esw 10110000.esw: port 4 link up
Tue Jun  3 02:38:57 2025 daemon.warn dnsmasq[4258]: ignoring query from non-local network 169.254.44.171 (logged only once)
Tue Jun  3 02:38:57 2025 daemon.info dnsmasq-dhcp[4258]: DHCPDISCOVER(br-lan) 192.168.1.214 90:2e:16:e2:bb:a9
Tue Jun  3 02:38:57 2025 daemon.info dnsmasq-dhcp[4258]: DHCPOFFER(br-lan) 192.168.1.214 90:2e:16:e2:bb:a9
Tue Jun  3 02:38:57 2025 daemon.info dnsmasq-dhcp[4258]: DHCPREQUEST(br-lan) 192.168.1.214 90:2e:16:e2:bb:a9
Tue Jun  3 02:38:57 2025 daemon.info dnsmasq-dhcp[4258]: DHCPACK(br-lan) 192.168.1.214 90:2e:16:e2:bb:a9 DESKTOP-I5AUN7S
Tue Jun  3 02:39:14 2025 daemon.err uhttpd[1555]: [info] luci: accepted login on /admin/status/realtime for root from 192.168.1.214
Tue Jun  3 02:42:33 2025 daemon.notice hostapd: phy0-ap0: AP-STA-DISCONNECTED f8:89:d2:d4:88:49
Tue Jun  3 02:42:33 2025 daemon.info hostapd: phy0-ap0: STA f8:89:d2:d4:88:49 IEEE 802.11: disassociated due to inactivity
Tue Jun  3 02:42:34 2025 daemon.info hostapd: phy0-ap0: STA f8:89:d2:d4:88:49 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Tue Jun  3 02:42:37 2025 daemon.notice hostapd: phy0-ap0: AP-STA-DISCONNECTED 4a:14:ec:47:e6:ce
Tue Jun  3 02:42:37 2025 daemon.info hostapd: phy0-ap0: STA 4a:14:ec:47:e6:ce IEEE 802.11: disassociated due to inactivity
Tue Jun  3 02:42:38 2025 daemon.info hostapd: phy0-ap0: STA 4a:14:ec:47:e6:ce IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
```

mjecov, 2025-06-03:
@lorand-horvath temperature was ok
@ob45 It was my mistake. I actually use **DoT with stubby package**. Not DoH