openwrt/openwrt #19014: realtek: kernel 6.12 dumps during partition creation

plappermaul, 2025-06-04:
### Describe the bug

Hi,

I'm currently preparing the kernel 6.12 upgrade for Realtek devices. See https://github.com/openwrt/openwrt/pull/18935

I solved out all target specific issues except for this one. During boot up I get (some logging aded):

```
[    0.976635] Creating 8 MTD partitions on "spi0.0":
[    0.982169] 0x000000000000-0x000000080000 : "u-boot"
[    1.061134] u-boot: parser fixed-partitions: 0
[    1.066154] mtd_part_do_parse return with 0
[    1.070976] mtd_part_of_parse 6 return rc=0
[    1.075698] parse_mtd_partitions called mtd_part_of_parse rc=0
[    1.082320] parse_mtd_partitions return 0
[    1.086869] 0x000000080000-0x0000000c0000 : "u-boot-env"
[    1.130986] u-boot-env: parser fixed-partitions: 0
[    1.136395] mtd_part_do_parse return with 0
[    1.141205] mtd_part_of_parse 6 return rc=0
[    1.145923] parse_mtd_partitions called mtd_part_of_parse rc=0
[    1.152560] parse_mtd_partitions return 0
[    1.157111] 0x0000000c0000-0x000000100000 : "board-name"
[    1.201123] board-name: parser fixed-partitions: 0
[    1.206531] mtd_part_do_parse return with 0
[    1.211372] mtd_part_of_parse 6 return rc=0
[    1.216095] parse_mtd_partitions called mtd_part_of_parse rc=0
[    1.222729] parse_mtd_partitions return 0
[    1.227272] 0x000000100000-0x000000e80000 : "firmware"

[    1.290623] firmware: parser uimage-fw: -19 <--- mtdsplit fails

[    1.295367] mtd_part_do_parse return with -19
[    1.300239] mtd_part_of_parse 3 rc=-19
[    1.304658] firmware: parser fixed-partitions: 0
[    1.309857] mtd_part_do_parse return with 0
[    1.314648] mtd_part_of_parse 6 return rc=-19
[    1.319565] parse_mtd_partitions called mtd_part_of_parse rc=-19
[    1.326420] parse_mtd_partitions return -19
[    1.331251] Failed to parse subpartitions: -19
[    1.336258] Deleting MTD partitions on "spi0.0":
[    1.341525] Deleting u-boot MTD partition
[    1.346050] ------------[ cut here ]------------
[    1.351319] WARNING: CPU: 0 PID: 1 at block/genhd.c:668 del_mtd_blktrans_dev+0x34/0x128
[    1.360459] Modules linked in:
[    1.363933] CPU: 0 UID: 0 PID: 1 Comm: swapper Not tainted 6.12.31 #0
[    1.371148] Hardware name: D-Link DGS-1210-28
[    1.376026] Stack : 00000000 00000004 00000000 ffffffff 8182b82c 00000000 00000001 00000100
[    1.385509]         818343e8 808b09d0 80a1ddff 813e33a4 00000000 00000001 8182b7d8 3ffbebce
[    1.394978]         00000000 00000000 808b09d0 8182b710 ffffefff 00000000 ffffffea 00000074
[    1.404456]         8182b6d4 00000074 00000000 80993730 00000001 00000000 808b09d0 00000000
[    1.413933]         00000000 0000029c 81c6e2b8 00000003 00000018 804862d0 00000000 813e0000
[    1.423402]         ...
[    1.426166] Call Trace:
[    1.428902] [<801062e4>] show_stack+0x28/0xf0
[    1.433830] [<807b3a28>] dump_stack_lvl+0x48/0x7c
[    1.439125] [<80126b7c>] __warn+0x9c/0x118
[    1.443756] [<80126c54>] warn_slowpath_fmt+0x5c/0xac
[    1.449344] [<804c52f8>] del_mtd_blktrans_dev+0x34/0x128
[    1.455327] [<804c4e6c>] blktrans_notify_remove+0x80/0xc0
[    1.461400] [<804bc0d4>] del_mtd_device+0x5c/0xe8
[    1.466697] [<804be8a8>] __del_mtd_partitions+0xb8/0xf4
[    1.472576] [<804bfcf4>] del_mtd_partitions+0x48/0x6c
[    1.478260] [<804c003c>] add_mtd_partitions+0xb8/0x1dc
[    1.484041] [<804bfde4>] parse_mtd_partitions+0xcc/0x26c
[    1.490017] [<804bc284>] mtd_device_parse_register+0x124/0x294
[    1.496574] [<804d8ccc>] spi_nor_probe+0x268/0x2f8
[    1.501984] [<8049a7d4>] really_probe+0xd4/0x34c
[    1.507195] [<8049ad10>] driver_probe_device+0x48/0x110
[    1.513074] [<8049aea8>] __device_attach_driver+0xd0/0x118
[    1.519242] [<8049845c>] bus_for_each_drv+0x7c/0xc8

```

To make it short.

- The issue does not exist on 6.6
- It happens during parsing the firmware partition with mtdsplit parser "uimage-fw"

Need to investigate further.

### OpenWrt version

upgrade 6.12

### OpenWrt release

upgrade 6.12

### OpenWrt target/subtarget

realtek/rtl838x

### Device

D-Link DGS-1210

### Image kind

Official downloaded image

### Steps to reproduce

Boot kernel 6.12

### Actual behaviour

Kernel givs stack dump and does not mount partitions

### Expected behaviour

kernel should boot up normally.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

plappermaul, 2025-06-04:
@john-tho I just saw your commit https://github.com/openwrt/openwrt/commit/ade045084bd3f86969eaf0b35234aaa01e430fe1 regarding the same issue.

So I think passing a info here should be fine. Or do you have additional info for me.

bmork, 2025-06-04:
Ouch.  If this has been in mainline since v6.12 (or before?) then a revert might be difficult.

But after looking closer, I think we should try to amend it. The sanity check as it is now is wrong IMHO.  It should at least ignore ENODEV, which has always been normal and expected from parse_mtd_partitions().  This is like a USB device failing to probe.  You don't fail the USB bus when that happens, ignoring all other devices.  You skip the failing device and continue.

Bailing out for other errors, like ENOMEM, does make some sense.

bmork, 2025-06-04:
Or maybe that ENODEV assumption was just wrong all the time?  I looked at the mainline examples, and see that for example
 drivers/mtd/parsers/ofpart_core.c has had
```
return 0; /* No partitions found */
```
since forever.  So maybe we should replace all ENODEV returns with 0 like you suggest.   Certainly smpler than arguing with upstream....   Not too sure about upping the debug printks to warnings.  Will that help or confuse users?  The parsing falures are expected most of the time they happen

Djfe, 2025-06-04:
this was not fixed for other instances of mtdsplit, was it?
let's leave the issue open then or should we open a new one?