openwrt/openwrt #19024: ath9k / 2.4GHz wifi not broadcasting

noblemtw, 2025-06-04:
### Describe the bug

On the latest snapshot builds, a 2.4Ghz signal is no longer being broadcast by the device

### OpenWrt version

r29614+1-eea4689654

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ath79/generic

### Device

Devolo dLAN pro 1200+ WiFi ac

### Image kind

Self-built image

### Steps to reproduce

Attempt to use 2.4Ghz on the device

### Actual behaviour

When using Wifiman on a mobile device to observe SSIDs / channels being broadcast, the device is not showing up on the 2.4Gz channels, only on 5Ghz with is ath10k rather than ath9k

### Expected behaviour

Device should be observed to be broadcasting with configured SSID in the 

[wireless](https://github.com/user-attachments/files/20592775/wireless.txt)
[dmesg](https://github.com/user-attachments/files/20592776/dmesg.txt)
[iwinfo](https://github.com/user-attachments/files/20592777/iwinfo.txt)

 2.4Ghz band 

### Additional info

I believe [12913c3c5658992985e13f4395dee86e5450154d](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=12913c3c5658992985e13f4395dee86e5450154d)] is the commit that broke the 2.4Ghz wifi. If I create a brand from the previous commit and build that then the 2.4Ghz is working correctly, whereas it no longer works when building from this commit onwards (Hence version string showing r29614+1)

Attached config is working on versions built prior to above commit and issue was apparent with a build from main yesterday, it's also unrelated to kernel version as the problem is see on default 6.6 and test 6.12 kernel

### Diffconfig

```text
CONFIG_TARGET_ath79=y
CONFIG_TARGET_ath79_generic=y
CONFIG_TARGET_ath79_generic_DEVICE_devolo_dlan-pro-1200plus-ac=y
CONFIG_DEVEL=y
CONFIG_TOOLCHAINOPTS=y
CONFIG_ATH10K_LEDS=y
CONFIG_BINARY_FOLDER="/home/openwrt/source/openwrt-generic/bin"
# CONFIG_BINUTILS_USE_VERSION_2_42 is not set
CONFIG_BINUTILS_USE_VERSION_2_43=y
CONFIG_BINUTILS_VERSION="2.43.1"
CONFIG_BINUTILS_VERSION_2_43=y
CONFIG_BUSYBOX_DEFAULT_PIE=y
CONFIG_CCACHE=y
CONFIG_CCACHE_DIR="/home/openwrt/source/openwrt-generic/mips/ccache"
CONFIG_DOWNLOAD_FOLDER="/home/openwrt/source/openwrt-generic/dl"
CONFIG_GCC_DEFAULT_PIE=y
CONFIG_GCC_DEFAULT_SSP=y
# CONFIG_GDB is not set
# CONFIG_KERNEL_CC_STACKPROTECTOR_REGULAR is not set
CONFIG_KERNEL_CC_STACKPROTECTOR_STRONG=y
CONFIG_KERNEL_STACKPROTECTOR_STRONG=y
CONFIG_NGINX_HTTP_AUTH_REQUEST=y
CONFIG_NGINX_HTTP_BROWSER=y
CONFIG_NGINX_HTTP_CHARSET=y
CONFIG_NGINX_HTTP_FASTCGI=y
CONFIG_NGINX_HTTP_GZIP=y
CONFIG_NGINX_HTTP_QUIC=y
CONFIG_NGINX_HTTP_UWSGI=y
CONFIG_NGINX_HTTP_V2=y
CONFIG_NGINX_PCRE=y
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_OPTIMIZE_SPEED=y
CONFIG_OPENSSL_PREFER_CHACHA_OVER_GCM=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_ERROR_MESSAGES=y
CONFIG_OPENSSL_WITH_IDEA=y
CONFIG_OPENSSL_WITH_MDC2=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SEED=y
CONFIG_OPENSSL_WITH_SRP=y
CONFIG_OPENSSL_WITH_TLS13=y
CONFIG_OPENSSL_WITH_WHIRLPOOL=y
CONFIG_PACKAGE_ATH_DEBUG=y
CONFIG_PACKAGE_ATH_DYNACK=y
# CONFIG_PACKAGE_apk-mbedtls is not set
CONFIG_PACKAGE_apk-openssl=y
CONFIG_PACKAGE_ath10k-firmware-qca988x=y
# CONFIG_PACKAGE_ath10k-firmware-qca988x-ct is not set
CONFIG_PACKAGE_bash=y
# CONFIG_PACKAGE_busybox is not set
CONFIG_PACKAGE_busybox-selinux=y
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_chrony=y
CONFIG_PACKAGE_dlan-fw-pro-1200plus-ac=y
CONFIG_PACKAGE_dlan-plc=y
CONFIG_PACKAGE_dmesg=y
# CONFIG_PACKAGE_dnsmasq is not set
# CONFIG_PACKAGE_dropbear is not set
CONFIG_PACKAGE_htop=y
CONFIG_PACKAGE_kmod-ath10k=y
# CONFIG_PACKAGE_kmod-ath10k-ct is not set
CONFIG_PACKAGE_kmod-i2c-core=y
CONFIG_PACKAGE_kmod-ikconfig=y
CONFIG_PACKAGE_kmod-nft-bridge=y
# CONFIG_PACKAGE_kmod-ppp is not set
CONFIG_PACKAGE_kmod-sched-core=y
CONFIG_PACKAGE_libatomic=y
CONFIG_PACKAGE_libcap=y
CONFIG_PACKAGE_liblua=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-lua=y
CONFIG_PACKAGE_liblucihttp-ucode=y
# CONFIG_PACKAGE_libmbedtls is not set
CONFIG_PACKAGE_libncurses=y
CONFIG_PACKAGE_libopenssl=y
CONFIG_PACKAGE_libopenssl-conf=y
CONFIG_PACKAGE_libopenssl-legacy=y
CONFIG_PACKAGE_libpcap=y
CONFIG_PACKAGE_libpcre2=y
CONFIG_PACKAGE_libpthread=y
CONFIG_PACKAGE_libreadline=y
CONFIG_PACKAGE_librt=y
CONFIG_PACKAGE_libselinux=y
CONFIG_PACKAGE_libsepol=y
CONFIG_PACKAGE_libstdcpp=y
CONFIG_PACKAGE_libubus-lua=y
# CONFIG_PACKAGE_libustream-mbedtls is not set
CONFIG_PACKAGE_libustream-openssl=y
CONFIG_PACKAGE_libuuid=y
CONFIG_PACKAGE_lua=y
CONFIG_PACKAGE_luci-app-chrony=m
CONFIG_PACKAGE_luci-app-commands=m
CONFIG_PACKAGE_luci-app-usteer=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-lib-base=y
CONFIG_PACKAGE_luci-lib-ip=y
CONFIG_PACKAGE_luci-lib-jsonc=y
CONFIG_PACKAGE_luci-lib-nixio=y
CONFIG_PACKAGE_luci-lua-runtime=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-theme-material=y
CONFIG_PACKAGE_musl-fts=y
CONFIG_PACKAGE_nginx-mod-luci=y
CONFIG_PACKAGE_nginx-mod-ubus=y
CONFIG_PACKAGE_nginx-ssl=y
CONFIG_PACKAGE_nginx-ssl-util=y
CONFIG_PACKAGE_open-plc-utils=y
CONFIG_PACKAGE_open-plc-utils-chkpib=y
CONFIG_PACKAGE_open-plc-utils-getpib=y
CONFIG_PACKAGE_open-plc-utils-hpavkey=y
CONFIG_PACKAGE_open-plc-utils-int6k=m
CONFIG_PACKAGE_open-plc-utils-modpib=y
CONFIG_PACKAGE_open-plc-utils-plchost=y
CONFIG_PACKAGE_open-plc-utils-plcrate=m
CONFIG_PACKAGE_open-plc-utils-plcstat=m
CONFIG_PACKAGE_open-plc-utils-plctool=y
CONFIG_PACKAGE_open-plc-utils-setpib=y
CONFIG_PACKAGE_openssh-client=m
CONFIG_PACKAGE_openssh-client-utils=m
CONFIG_PACKAGE_openssh-keygen=y
CONFIG_PACKAGE_openssh-server=y
CONFIG_PACKAGE_openssh-sftp-client=m
CONFIG_PACKAGE_openssh-sftp-server=y
# CONFIG_PACKAGE_ppp is not set
# CONFIG_PACKAGE_procd is not set
CONFIG_PACKAGE_procd-selinux=y
CONFIG_PACKAGE_procps-ng=y
CONFIG_PACKAGE_procps-ng-ps=y
CONFIG_PACKAGE_procps-ng-sysctl=y
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_selinux-policy=y
CONFIG_PACKAGE_sudo=y
CONFIG_PACKAGE_sysinfo=y
CONFIG_PACKAGE_tcpdump-mini=y
CONFIG_PACKAGE_terminfo=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-lua=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_usteer-ng=y
CONFIG_PACKAGE_uwsgi=y
CONFIG_PACKAGE_uwsgi-cgi-plugin=y
CONFIG_PACKAGE_uwsgi-luci-support=y
CONFIG_PACKAGE_uwsgi-syslog-plugin=y
CONFIG_PACKAGE_vim=y
# CONFIG_PACKAGE_wpad-basic-mbedtls is not set
CONFIG_PACKAGE_wpad-openssl=y
CONFIG_PACKAGE_zoneinfo-core=y
CONFIG_PACKAGE_zoneinfo-europe=y
CONFIG_PKG_ASLR_PIE_ALL=y
# CONFIG_PKG_ASLR_PIE_REGULAR is not set
# CONFIG_PKG_CC_STACKPROTECTOR_REGULAR is not set
CONFIG_PKG_CC_STACKPROTECTOR_STRONG=y
# CONFIG_PKG_FORTIFY_SOURCE_1 is not set
CONFIG_PKG_FORTIFY_SOURCE_2=y
CONFIG_TARGET_OPTIMIZATION="-O2 -pipe -mno-branch-likely -mips32r2 -mtune=74kc"
CONFIG_TARGET_OPTIONS=y
CONFIG_USE_GC_SECTIONS=y
CONFIG_USE_LTO=y
# CONFIG_HTOP_LMSENSORS is not set
# CONFIG_NGINX_HTTP_ACCESS is not set
# CONFIG_NGINX_HTTP_AUTH_BASIC is not set
# CONFIG_NGINX_HTTP_AUTOINDEX is not set
# CONFIG_NGINX_HTTP_CACHE is not set
# CONFIG_NGINX_HTTP_EMPTY_GIF is not set
# CONFIG_NGINX_HTTP_GEO is not set
# CONFIG_NGINX_HTTP_GZIP_STATIC is not set
# CONFIG_NGINX_HTTP_LIMIT_CONN is not set
# CONFIG_NGINX_HTTP_LIMIT_REQ is not set
# CONFIG_NGINX_HTTP_MAP is not set
# CONFIG_NGINX_HTTP_MEMCACHED is not set
# CONFIG_NGINX_HTTP_PROXY is not set
# CONFIG_NGINX_HTTP_REFERER is not set
# CONFIG_NGINX_HTTP_REWRITE is not set
# CONFIG_NGINX_HTTP_SCGI is not set
# CONFIG_NGINX_HTTP_SPLIT_CLIENTS is not set
# CONFIG_NGINX_HTTP_SSI is not set
# CONFIG_NGINX_HTTP_UPSTREAM_HASH is not set
# CONFIG_NGINX_HTTP_UPSTREAM_IP_HASH is not set
# CONFIG_NGINX_HTTP_UPSTREAM_KEEPALIVE is not set
# CONFIG_NGINX_HTTP_UPSTREAM_LEAST_CONN is not set
# CONFIG_NGINX_HTTP_USERID is not set
CONFIG_PACKAGE_kmod-hwmon-core=y
CONFIG_PACKAGE_kmod-lib-crc-ccitt=y
```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

neheb, 2025-06-04:
It appears there is a typo. Try this

```
--- a/package/kernel/mac80211/patches/ath9k/550-ath9k-of.patch
+++ b/package/kernel/mac80211/patches/ath9k/550-ath9k-of.patch
@@ -125,7 +125,7 @@
 +      } else if (ah->hw_version.devid == AR9300_DEVID_AR9340) {
 +              ah->get_mac_revision = ath79_get_soc_revision;
 +              u32 t = ath79_reset_rr(AR934X_RESET_REG_BOOTSTRAP);
-+              ah->is_clk_25mhz = !(t & AR933X_BOOTSTRAP_REF_CLK_40);
++              ah->is_clk_25mhz = !(t & AR934X_BOOTSTRAP_REF_CLK_40);
 +      } else if (ah->hw_version.devid == AR9300_DEVID_AR953X) {
 +              ah->get_mac_revision = ath79_get_soc_revision;
 +              /*
```

noblemtw, 2025-06-05:
Can confirm it's now working with the the above patch applied, many thanks

closed 2025-06-05 by openwrt-bot