openwrt/openwrt #19040: apk: when opkg is able to handle UCI defaults files if install a package after "firstboot", apk does not

map-b, 2025-06-05:
### Describe the bug

For example uboot-envtools and the uci defaults file https://github.com/openwrt/openwrt/blob/main/package/boot/uboot-tools/uboot-envtools/files/ramips 

```
root@OpenWrt:~# apk add -vv uboot-envtools
NOTE: Consider running apk upgrade with --prune and/or --available.
The following packages are no longer available from a repository:
  kmod-cfg80211 kmod-crypto-aead kmod-crypto-authenc kmod-crypto-ccm kmod-crypto-cmac kmod-crypto-crc32c
  kmod-crypto-ctr kmod-crypto-des kmod-crypto-gcm kmod-crypto-geniv kmod-crypto-gf128 kmod-crypto-ghash
  kmod-crypto-hash kmod-crypto-hmac kmod-crypto-hw-eip93 kmod-crypto-manager kmod-crypto-md5 kmod-crypto-null
  kmod-crypto-rng kmod-crypto-seqiv kmod-crypto-sha1 kmod-crypto-sha256 kmod-crypto-sha3 kmod-crypto-sha512
  kmod-gpio-button-hotplug kmod-leds-gpio kmod-lib-crc-ccitt kmod-lib-crc32c kmod-mac80211 kmod-mt76-core kmod-mt7603
  kmod-mt76x02-common kmod-mt76x2 kmod-mt76x2-common kmod-nf-conntrack kmod-nf-conntrack6 kmod-nf-flow kmod-nf-log
  kmod-nf-log6 kmod-nf-nat kmod-nf-reject kmod-nf-reject6 kmod-nfnetlink kmod-nft-core kmod-nft-fib kmod-nft-nat
  kmod-nft-offload kmod-nls-base kmod-ppp kmod-pppoe kmod-pppox kmod-slhc kmod-usb-core kmod-usb-ledtrig-usbport
  kmod-usb-xhci-hcd kmod-usb-xhci-mtk kmod-usb3
The following NEW packages will be installed:
  uboot-envtools (2025.01-r3)
Need to download 13 KiB of packages.
After this operation, 100 KiB of additional disk space will be used.
(1/1) Installing uboot-envtools (2025.01-r3)
etc (dir)
etc/board.d (dir)
etc/board.d/05_fw_defaults
etc/uci-defaults (dir)
lib (dir)
lib/uboot-envtools.sh
lib/apk (dir)
lib/apk/packages (dir)
lib/apk/packages/uboot-envtools.conffiles
lib/apk/packages/uboot-envtools.list
lib/upgrade (dir)
lib/upgrade/keep.d (dir)
lib/upgrade/keep.d/uboot-envtools
usr (dir)
usr/sbin (dir)
usr/sbin/fw_loadenv
usr/sbin/fw_printenv
usr/sbin/fw_printsys
usr/sbin/fw_setenv
usr/sbin/fw_setsys
uboot-envtools-2025.01-r3.post-install: Executing script...
OK: 126 packages, 85 dirs, 769 files, 15 MiB
root@OpenWrt:~#
```

```
root@OpenWrt:~# fw_printenv
Cannot parse config file '/etc/fw_env.config': No such file or directory
Failed to find NVMEM device
root@OpenWrt:~# apk del --purge -vv uboot-envtools
NOTE: Consider running apk upgrade with --prune and/or --available.
The following packages are no longer available from a repository:
  kmod-cfg80211 kmod-crypto-aead kmod-crypto-authenc kmod-crypto-ccm kmod-crypto-cmac kmod-crypto-crc32c
  kmod-crypto-ctr kmod-crypto-des kmod-crypto-gcm kmod-crypto-geniv kmod-crypto-gf128 kmod-crypto-ghash
  kmod-crypto-hash kmod-crypto-hmac kmod-crypto-hw-eip93 kmod-crypto-manager kmod-crypto-md5 kmod-crypto-null
  kmod-crypto-rng kmod-crypto-seqiv kmod-crypto-sha1 kmod-crypto-sha256 kmod-crypto-sha3 kmod-crypto-sha512
  kmod-gpio-button-hotplug kmod-leds-gpio kmod-lib-crc-ccitt kmod-lib-crc32c kmod-mac80211 kmod-mt76-core kmod-mt7603
  kmod-mt76x02-common kmod-mt76x2 kmod-mt76x2-common kmod-nf-conntrack kmod-nf-conntrack6 kmod-nf-flow kmod-nf-log
  kmod-nf-log6 kmod-nf-nat kmod-nf-reject kmod-nf-reject6 kmod-nfnetlink kmod-nft-core kmod-nft-fib kmod-nft-nat
  kmod-nft-offload kmod-nls-base kmod-ppp kmod-pppoe kmod-pppox kmod-slhc kmod-usb-core kmod-usb-ledtrig-usbport
  kmod-usb-xhci-hcd kmod-usb-xhci-mtk kmod-usb3
The following packages will be REMOVED:
  uboot-envtools (2025.01-r3)
After this operation, 100 KiB of additional disk space will be used.
(1/1) Purging uboot-envtools (2025.01-r3)
uboot-envtools-2025.01-r3.pre-deinstall: Executing script...
etc/board.d/05_fw_defaults
lib/uboot-envtools.sh
lib/apk/packages/uboot-envtools.conffiles
lib/apk/packages/uboot-envtools.list
lib/upgrade/keep.d/uboot-envtools
usr/sbin/fw_loadenv
usr/sbin/fw_printenv
usr/sbin/fw_printsys
usr/sbin/fw_setenv
usr/sbin/fw_setsys
OK: 125 packages, 85 dirs, 759 files, 15 MiB
root@OpenWrt:~#
```
It seems that it is doing almost all:
https://github.com/openwrt/openwrt/blob/8b24289a5267e486abd9ccbf4b4ad82f14d545ae/package/boot/uboot-tools/Makefile#L106-L126

But does not like the final:
https://github.com/openwrt/openwrt/blob/8b24289a5267e486abd9ccbf4b4ad82f14d545ae/package/boot/uboot-tools/Makefile#L118-L125

OPKG is able to do the uci-defaults part:
```
root@OpenWrt:~# rm /etc/config/ubootenv
root@OpenWrt:~# fw
fw3     fw4     fwtool
root@OpenWrt:~# opkg install uboot-envtools
Installing uboot-envtools (2024.07-r1) to root...
Downloading https://downloads.openwrt.org/releases/24.10.1/targets/ramips/mt7621/packages/uboot-envtools_2024.07-r1_mipsel_24kc.ipk
Configuring uboot-envtools.
cfg016cb7
cfg02a9ad
grep: /etc/fw_env.config: No such file or directory
grep: /etc/fw_sys.config: No such file or directory
```

### OpenWrt version

r29941-1304d8b064

### OpenWrt release

SNAPAHOT

### OpenWrt target/subtarget

ramips/mt7621

### Device

R3G v1

### Image kind

Self-built image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

map-b, 2025-06-12:
@efahl let's move here the discussion, this another issue but it's related, (excuse me by the messy description and my English, you can skip the first part and go to the Update header) 

Do you see differences when downloading a package using `apk fetch` versus `wget`(maybe that's expected)?  I checked them with `apk adbdump` and other mechanisms from apk but I chose to show `apk manifest` here because it's more concise.

Is it normal the different sha256 hashes?

efahl, 2025-06-12:
> Do you see differences when downloading a package using apk fetch versus wget(maybe that's expected)?

No, it is completely unexpected.  They should be exactly the same because `apk` finds the `wget` command and uses that to do its downloads internally.  It's almost always whatever `/usr/bin/wget` links to.

You can get it to show you the location from which it's downloading with `apk fetch --url pkg`.  Try that and see if it shows the same as you are using with your `wget` command.

The hashes should be the same, too...

map-b, 2025-06-14:
In today's snapshot I am getting the same "corrupted" package using `apk fetch` or `wget` but the good one in download on my x86_64 host machine.

```shell
root@OpenWrt:~# wget https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/base/ubox-2024.04.26~85f10530-r1.apk
--2025-06-13 23:39:59--  https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/base/ubox-2024.04.26~85f10530-r1.apk
Resolving downloads.openwrt.org... 151.101.2.132, 151.101.66.132, 151.101.130.132, ...
Connecting to downloads.openwrt.org|151.101.2.132|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 16799 (16K) [application/octet-stream]
Saving to: 'ubox-2024.04.26~85f10530-r1.apk'

ubox-2024.04.26~85f10530-r1.a 100%[===============================================>]  16.41K  --.-KB/s    in 0.003s

2025-06-13 23:40:00 (5.07 MB/s) - 'ubox-2024.04.26~85f10530-r1.apk' saved [16799/16799]

root@OpenWrt:~# ls -l
-rw-r--r--    1 root     root         15369 Jun 13 22:30 uboot-envtools-2025.01-r3 (1).apk
-rw-r--r--    1 root     root         13929 Jun 13 22:37 uboot-envtools-2025.01-r3.apk
-rw-r--r--    1 root     root         13929 Jun 13 22:23 uboot-envtools-2025.01-r3FETCHED.apk
-rw-r--r--    1 root     root         13929 Jun 13 22:27 uboot-envtools-2025.01-r3WGET.apk
-rw-r--r--    1 root     root         16799 Jun 13 05:09 ubox-2024.04.26~85f10530-r1.apk
-rw-r--r--    1 root     root         16873 Jun 13 22:40 ubox-2024.04.26~85f10530-r1EXTERNAL.apk
-rw-r--r--    1 root     root         16799 Jun 13 05:09 ubox-2024.04.26~85f10530-r1FETCHED.apk
root@OpenWrt:~#
```

commit dc80e698 (2025-06-25): uboot-envtools: set nonshared flag correctly per-subpackage

Currently, uboot-envtools is being built for a (shared) instruction
set (phase2 in buildbots) instead of target-specific (phase1 in builbots).
So the package does not contain the uci-defaults file specific for each
target_subtarget. (Only the fortunate target with coincidentally the same
instruction set and target as the SDK chosen by the buildbot will have the
uci-defaults file.)

This commit sets the nonshared flag correctly in the Makefile so that
uboot-envtools is built for the target-specific (phase 1).

This is done by using th [... cut]
PR #19180 (2025-06-25): uboot-envtools: set nonshared flag correctly
Currently, uboot-envtools is being built for a (shared) instruction set (phase2 in buildbots) instead of target-specific (phase1 in builbots). So the package does not contain the uci-defaults file specific for each target_subtarget. (Only the fortunate target with coincidentally the same instruction set as the SDK chosen by the buildbot will have the uci-defaults file.)

This commit sets the nonshared flag correctly in the Makefile so that uboot-envtools is built for the target-specific.

Fixes: #19040
Fixes: 46e376c ("uboot-tools: migrate uboot-env [... cut]
closed 2025-06-25 by openwrt-bot