openwrt/openwrt #19057: MPTCP is not working-linux6.12

sijidou, 2025-06-07:
### Describe the bug

MPTCP is not working. The tool used for testing is shadowsocks-rust.
Using the mainline kernel version, I enabled the following options when compiling the kernel:
`[*]   MPTCP: Multipath TCP  
[*]     MPTCP: IPv6 support for Multipath TCP  `

After the system boots,
I execute:
`ip mptcp endpoint flush  
ip mptcp limits set subflows 8 add_addr_accepted 8  
ip mptcp endpoint add xxxxxxxxxxxxxxxx  `

Even after adding the correct endpoints, MPTCP is not working. The symptoms are:

ip mptcp monitorshows no information.

When running a speed test (eg, with Speedtest), only one interface has traffic, while the other three interfaces have no traffic.

The setup and network configuration have not changed. This was working correctly with the previous Linux 6.6 kernel version.
I have tested this on both x86 and sunxi/cortex-a53 platforms, and the results are the same.

### OpenWrt version

r29909-ba674fea40

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

x86/64  and sunxi/cortexa53

### Device

OrangePi Zero2

### Image kind

Self-built image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-06-07:
Do you run multiwan? Youu should not.

sijidou, 2025-06-08:
> Do you run multiwan? Youu should not.

I have multiple broadband connections, and I found that the current version of iperf3 supports
`-m, --mptcp use MPTCP rather than plain TCP`
Using this option for testing yields the same result.
After using` ip mptcp endpoint delete/flush/change/add,` 
`ip mptcp monitor` stops showing any data after a while.

sijidou, 2025-06-08:
> to

I may not have expressed myself clearly. I'm not using mwan. I know how to use MPTCP and have been using it for a long time.

sijidou, 2025-06-08:
> You need to shut off mwan so that mptcp can bind outgoing connections to wan addresses.

I may not have expressed myself clearly. I'm not using mwan. I know how to use MPTCP and have been using it for a long time.

rmandrad66, 2025-06-12:
instead of overriding via kernel_menuconfig select

[image]

brada4, 2025-06-12:
Can you run `conntrack -E` to determine destiny of part-connections? You can filter it by [Conntrack Meta](https://wiki.nftables.org/wiki-nftables/index.php/Matching_connection_tracking_stateful_metainformation) if output is too verbose.