openwrt/openwrt #19062: Netgear WAX610y drops in U-Boot prompt after flashing sysupgrade image using TFTP Method.

vochong, 2025-06-07:
### Describe the bug

I tried to convert a Netgear WAX610Y to OpenWrt using the TFTP method. After booting the initramfs-uImage.itb into OpenWrt, I performed a sysupgrade. However, after rebooting, the device consistently drops into the U-Boot prompt. This is using the latest snapshot image.

I followed the instructions in this PR:   https://github.com/openwrt/openwrt/pull/18377

@ServError    @robimarko 

```
root@OpenWrt:~# ubus call system board
{
        "kernel": "6.12.32",
        "hostname": "OpenWrt",
        "system": "ARMv8 Processor rev 4",
        "model": "Netgear WAX610Y",
        "board_name": "netgear,wax610y",
        "rootfs_type": "initramfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "SNAPSHOT",
                "firmware_url": "https://downloads.openwrt.org/",
                "revision": "r29955-8b24289a52",
                "target": "qualcommax/ipq60xx",
                "description": "OpenWrt SNAPSHOT r29955-8b24289a52",
                "builddate": "1749156955"
        }
}
```

After booting using the initramfs image:
```
root@OpenWrt:/tmp# sysupgrade -n openwrt-qualcommax-ipq60xx-netgear_wax610y-squa
shfs-sysupgrade.bin 
Thu Jan  1 00:05:17 GMT 1970 upgrade: Commencing upgrade. Closing all shell sessions.
Watchdog handover: fd=3
- watchdog -
Watchdog did not previously reset the system
Thu Jan  1 00:05:18 GMT 1970 upgrade: Sending TERM to remaining processes ...
Thu Jan  1 00:05:22 GMT 1970 upgrade: Sending KILL to remaining processes ...
[  328.399126] stage2 (3554): drop_caches: 3
Thu Jan  1 00:05:28 GMT 1970 upgrade: Switching to ramdisk...
Thu Jan  1 00:05:29 UTC 1970 upgrade: Performing system upgrade...
verifying sysupgrade tar file integrity
[  330.095902] block ubiblock0_3: released
Volume ID 0, size 45 LEBs (5713920 bytes, 5.4 MiB), LEB size 126976 bytes (124.0 KiB), dynamic, name "kernel", alignment 1
[  330.424644] block ubiblock0_3: created from ubi0:3(rootfs)
Volume ID 3, size 54 LEBs (6856704 bytes, 6.5 MiB), LEB size 126976 bytes (124.0 KiB), dynamic, name "rootfs", alignment 1
Set volume size to 1396736
Volume ID 4, size 11 LEBs (1396736 bytes, 1.3 MiB), LEB size 126976 bytes (124.0 KiB), dynamic, name "rootfs_data", alignment 1
sysupgrade successful
umount: can't unmount /dev: Resource busy
umount: can't unmount /tmp: Resource busy
[  334.185660] leds blue:wlan-2: led_trigger_set: Error sending uevent
[  334.305659] leds blue:wlan-0: led_trigger_set: Error sending uevent
[  334.374075] remoteproc remoteproc0: stopped remote processor cd00000.remoteproc
[  335.086586] reboot: Restarting system

Format: Log Type - Time(microsec) - Message - Optional Info
Log Type: B - Since Boot(Power On Reset),  D - Delta,  S - Statistic
S - QC_IMAGE_VERSION_STRING=BOOT.XF.0.3-00077-IPQ60xxLZB-2
S - IMAGE_VARIANT_STRING=IPQ6018LA
S - OEM_IMAGE_VERSION_STRING=crm-ubuntu64
S - Boot Interface: NAND
S - Secure Boot: Off
S - Boot Config @ 0x000a602c = 0x000002e4
S - JTAG ID @ 0x000a607c = 0x0013a0e1
S - OEM ID @ 0x000a6080 = 0x00000000
S - Serial Number @ 0x000a4128 = 0x9c4da819
S - OEM Config Row 0 @ 0x000a4188 = 0x0000000000000000
S - OEM Config Row 1 @ 0x000a4190 = 0x0000000000000000
S - Feature Config Row 0 @ 0x000a4130 = 0x0000000008000001
S - Feature Config Row 1 @ 0x000a4138 = 0x02c3e83383000009
S - PBL Patch Ver: 1
S - I-cache: On
S - D-cache: On
B -      3413 - PBL, Start
B -       592 - bootable_media_detect_entry, Start
B -      4339 - bootable_media_detect_success, Start
B -      5147 - elf_loader_entry, Start
B -      5319 - auth_hash_seg_entry, Start
B -      7787 - auth_hash_seg_exit, Start
B -      8282 - elf_segs_hash_verify_entry, Start
B -    110429 - elf_segs_hash_verify_exit, Start
B -    114854 - auth_xbl_sec_hash_seg_entry, Start
B -    114998 - auth_xbl_sec_hash_seg_exit, Start
B -    121549 - xbl_sec_segs_hash_verify_entry, Start
B -    121550 - xbl_sec_segs_hash_verify_exit, Start
B -    122479 - PBL, End
B -    103273 - SBL1, Start
B -    243359 - GCC [RstStat:0x0, RstDbg:0x600000] WDog Stat : 0x4
B -    245799 - clock_init, Start
D -      2745 - clock_init, Delta
B -    254278 - boot_flash_init, Start
D -     29676 - boot_flash_init, Delta
B -    287249 - sbl1_ddr_set_default_params, Start
D -       244 - sbl1_ddr_set_default_params, Delta
B -    293867 - boot_config_data_table_init, Start
D -      4758 - boot_config_data_table_init, Delta - (575 Bytes)
B -    303688 - CDT Version:2,Platform ID:8,Major ID:3,Minor ID:2,Subtype:0
B -    308538 - Image Load, Start
D -      6618 - OEM_MISC Image Loaded, Delta - (0 Bytes)
B -    317871 - Image Load, Start
D -      5063 - PMIC Image Loaded, Delta - (0 Bytes)
B -    325740 - sbl1_ddr_set_params, Start
B -    330711 - CPR configuration: 0x366
B -    333914 - Pre_DDR_clock_init, Start
D -       183 - Pre_DDR_clock_init, Delta
D -         0 - sbl1_ddr_set_params, Delta
B -    370575 - Image Load, Start
D -       427 - APDP Image Loaded, Delta - (0 Bytes)
B -    383659 - Image Load, Start
D -       458 - QTI_MISC Image Loaded, Delta - (0 Bytes)
B -    386099 - Image Load, Start
D -       793 - Auth Metadata
D -       640 - Segments hash check
D -     22387 - QSEE Dev Config Image Loaded, Delta - (36354 Bytes)
B -    410438 - Image Load, Start
D -      6588 - Auth Metadata
D -     10553 - Segments hash check
D -    336324 - QSEE Image Loaded, Delta - (1470632 Bytes)
B -    747189 - Image Load, Start
D -       702 - Auth Metadata
D -      1037 - Segments hash check
D -     35075 - RPM Image Loaded, Delta - (102664 Bytes)
B -    783941 - Image Load, Start
D -       763 - Auth Metadata
D -      3050 - Segments hash check
D -    117944 - APPSBL Image Loaded, Delta - (541348 Bytes)
B -    917531 - SBL1, End
D -    814563 - SBL1, Delta
S - Flash Throughput, 4000 KB/s  (2152245 Bytes,  432298 us)
S - Core 0 Frequency, 800 MHz
S - DDR Frequency, 466 MHz

U-Boot 2016.01-V9.0.0.23 (May 21 2020 - 11:58:58 +0530)

DRAM:  smem ram ptable found: ver: 2 len: 4
512 MiB
NAND:  ONFI device found
ID = 9500a1ef
Vendor = ef
Device = a1
SPI_ADDR_LEN=3
SF: Detected MX25U6435F with page size 256 Bytes, erase size 64 KiB, total 8 MiB
ipq_spi: page_size: 0x100, sector_size: 0x10000, size: 0x800000
136 MiB
MMC:   sdhci: Node Not found, skipping initialization

PCI0 is not defined in the device tree
In:    serial@78B1000
Out:   serial@78B1000
Err:   serial@78B1000
Product ID: WAX610Y
HW Version: 1.0
machid: 8030200
Power source: Adaptor
eth2 MAC Address from ART is not valid
eth3 MAC Address from ART is not valid
eth4 MAC Address from ART is not valid
eth5 MAC Address from ART is not valid
Hit any key to stop autoboot:  0 
Erasing NAND...
Erasing at 0xde0000 -- 100% complete.
Writing to NAND... OK
setenv - set environment variables

Net:   MAC0 addr:44:a5:6e:cb:9c:df
PHY ID1: 0x4d
PHY ID2: 0xd101
EDMA ver 1 hw init
Num rings - TxDesc:1 (0-0) TxCmpl:1 (0-0)
RxDesc:1 (15-15) RxFill:1 (7-7)
ipq6018_edma_alloc_rings: successfull
ipq6018_edma_setup_ring_resources: successfull
ipq6018_edma_configure_rings: successfull
ipq6018_edma_hw_init: successfull
eth0

Net:   MAC0 addr:44:a5:6e:cb:9c:df
PHY ID1: 0x4d
PHY ID2: 0xd101
EDMA ver 1 hw init
Num rings - TxDesc:1 (0-0) TxCmpl:1 (0-0)
RxDesc:1 (15-15) RxFill:1 (7-7)
ipq6018_edma_alloc_rings: successfull
ipq6018_edma_setup_ring_resources: successfull
ipq6018_edma_configure_rings: successfull
ipq6018_edma_hw_init: successfull
, eth0
Warning: eth0 MAC addresses don't match:
Address in SROM is         44:a5:6e:cb:9c:df
Address in environment is  44:a5:6e:cb:9c:ef

IPQ6018#

```

### OpenWrt version

r29955-8b24289a52

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

qualcommax/ipq60xx

### Device

Netgear WAX610Y

### Image kind

Official downloaded image

### Steps to reproduce

Follow the TFTP instructions in this PR:   https://github.com/openwrt/openwrt/pull/18377

### Actual behaviour

The device drops back in the U-Boot prompt.  I saw some unexpected NAND erase after sysupgrade/reboot:

PCI0 is not defined in the device tree
In:    serial@78B1000
Out:   serial@78B1000
Err:   serial@78B1000
Product ID: WAX610Y
HW Version: 1.0
machid: 8030200
Power source: Adaptor
eth2 MAC Address from ART is not valid
eth3 MAC Address from ART is not valid
eth4 MAC Address from ART is not valid
eth5 MAC Address from ART is not valid
Hit any key to stop autoboot:  0 
**Erasing NAND...
Erasing at 0xde0000 -- 100% complete.
Writing to NAND... OK**
setenv - set environment variables

### Expected behaviour

The device should boot into OpenWrt successfully after the sysupgrade.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

ServError, 2025-06-23:
Resolved by #19215

Djfe, 2025-07-12:
please close this issue :)
and feel free to verify whether your issue has been fixed in OpenWrt snapshots

vochong, 2025-07-12:
Fixed in snapshots.  Thanks!

closed 2025-07-12 by vochong