openwrt/openwrt #19075: SELINUX unusable?  login: can't get SID for root

AnoBene, 2025-06-09:
### Describe the bug

Compiling OpenWRT with selinux options.
System boots up, but is unusable because pressing enter does not provide login with error:
login: can't get SID for root

### OpenWrt version

r28597-0425664679

### OpenWrt release

24.10.1

### OpenWrt target/subtarget

x86/64

### Device

PC

### Image kind

Official downloaded image

### Steps to reproduce

1. ubuntu 24.04 server default install
Install stuff..

``` sudo apt update
 sudo apt-get install build-essential subversion git-core git patch bzip2 flex bison \
 autoconf gettext libncurses5-dev ncurses-term zlib1g-dev gawk libz-dev libssl-dev \
 ack antlr3 asciidoc automake autopoint binutils \
 ccache cmake cpio device-tree-compiler fastjar clang g++ gcc-multilib g++-multilib \
 gperf haveged help2man intltool libc6-dev-i386 libelf-dev libglib2.0-dev libgmp3-dev libltdl-dev \
 libmpc-dev libmpfr-dev libncursesw5-dev libreadline-dev libtool lrzsz \
 qemu-utils scons squashfs-tools texinfo uglifyjs upx-ucl unzip \
 curl rsync swig  file wget sharutils fakeroot lzma xmlto xxd vim nano mkisofs msmtp ninja-build p7zip p7zip-full pkgconf \
 libncurses-dev libexpat1 libexpat1-dev libdaemon* libdaemon-generic-perl libdaemon0 libdaemon-dev libdaemon-doc libdaemon-control-perl \
 python3 python3-pyelftools python3-distutils-extra python3-setuptools libpython3-dev
```

git clone https://git.openwrt.org/openwrt/openwrt.git
cd openwrt
git pull
git checkout v24.10.1
wget -O feeds.conf.default https://downloads.openwrt.org/releases/24.10.1/targets/x86/64/feeds.buildinfo

Add SELINUX options to .config
```
CONFIG_PACKAGE_busybox=y
#CONFIG_PACKAGE_busybox-selinux=m
CONFIG_BUSYBOX_CUSTOM=y
CONFIG_KERNEL_LSM="lockdown,yama,loadpin,safesetid,integrity,selinux"
CONFIG_SELINUX=y
CONFIG_SELINUXTYPE_dssp=y
CONFIG_KERNEL_SECURITY_SELINUX=y
CONFIG_KERNEL_SECURITY_SELINUX_BOOTPARAM=y
CONFIG_KERNEL_SECURITY_SELINUX_DEVELOP=y
CONFIG_KERNEL_SECURITY_SELINUX_SID2STR_CACHE_SIZE=256
CONFIG_KERNEL_SECURITY_SELINUX_SIDTAB_HASH_BITS=9
CONFIG_PACKAGE_selinux-policy=y
CONFIG_PACKAGE_procd-selinux=y
CONFIG_PACKAGE_libselinux=y
CONFIG_PACKAGE_libf2fs-selinux=m
CONFIG_PACKAGE_mkf2fs-selinux=m
```

make menuconfig
make -j $(nproc) clean defconfig download
make -j 1 world V=99 -k 2>&1 | tee "/results_clean_world_99_$(date +"%Y-%m-%d.%H.%M.%S")_j01.log"

### Actual behaviour

System boots up, but is unusable because pressing enter does not provide login with error:
login: can't get SID for root

### Expected behaviour

Can log in to default root account after boot

### Additional info

From internets i only found this as kinda releevant maybe:
https://www.mail-archive.com/busybox@busybox.net/msg28806.html

But its from 2023 and what they refer is from 2020
I have SELINUX enabled compile working without this problem with 2022 made compile
with same SELINUX diffoptions but older OpenWRT 21.02.2

### Diffconfig

```text
Add SELINUX options to .config

CONFIG_PACKAGE_busybox=y
#CONFIG_PACKAGE_busybox-selinux=m
CONFIG_BUSYBOX_CUSTOM=y
CONFIG_KERNEL_LSM="lockdown,yama,loadpin,safesetid,integrity,selinux"
CONFIG_SELINUX=y
CONFIG_SELINUXTYPE_dssp=y
CONFIG_KERNEL_SECURITY_SELINUX=y
CONFIG_KERNEL_SECURITY_SELINUX_BOOTPARAM=y
CONFIG_KERNEL_SECURITY_SELINUX_DEVELOP=y
CONFIG_KERNEL_SECURITY_SELINUX_SID2STR_CACHE_SIZE=256
CONFIG_KERNEL_SECURITY_SELINUX_SIDTAB_HASH_BITS=9
CONFIG_PACKAGE_selinux-policy=y
CONFIG_PACKAGE_procd-selinux=y
CONFIG_PACKAGE_libselinux=y
CONFIG_PACKAGE_libf2fs-selinux=m
CONFIG_PACKAGE_mkf2fs-selinux=m
```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

ynezz, 2025-06-09:
I've just prepared https://github.com/openwrt/openwrt/pull/19080 so you can cherry-pick that fix https://github.com/openwrt/openwrt/pull/19080/commits/9d42d56c2efbc7c7ca13675261703f91aa744b07 into 24.10 try it out and report back.

brada4, 2025-06-10:
You have incomplete required packages for build system (50 extra copypasted does not help for one missing)
Please install as guided on the wiki and retry build on a clean tree.

PR #19542 (2025-07-27): [24.10] busybox: fix login applet on selinux
Currently the system boots up, but is unusable because pressing enter does not provide login with error:

  login: can't get SID for root

This is happenning, because login.c passes the Linux username directly to get_default_context(), while libselinux expects an SELinux user identity, causing the call to fail for users without a matching SELinux name (e.g., root) and aborting login on SELinux-enabled systems.

Fixes: #19075
Upstream-Status: Submitted [https://lists.busybox.net/pipermail/busybox/2025-April/091407.html]
Link: https://github.com/openw [... cut]
commit bfed41cc (2025-07-27): busybox: fix login applet on selinux

Currently the system boots up, but is unusable because pressing enter
does not provide login with error:

  login: can't get SID for root

This is happenning, because login.c passes the Linux username directly
to get_default_context(), while libselinux expects an SELinux user
identity, causing the call to fail for users without a matching SELinux
name (e.g., root) and aborting login on SELinux-enabled systems.

Fixes: #19075
Upstream-Status: Submitted [https://lists.busybox.net/pipermail/busybox/2025-April/091407.html]

Link: https://github.com/openwrt/ope [... cut]
closed 2025-07-26 by openwrt-bot