openwrt/openwrt #19078: BPi-R4: WED does not work on kernel 6.12

danpawlik, 2025-06-09:
### Describe the bug

Hi,
On testing kernel 6.12 on BPi-R4 WED does not work.

Dmesg:
```
[   59.230171] mt7996e 0000:01:00.0: Message 001a0002 (seq 4) timeout
[   79.639985] mt7996e 0000:01:00.0: Message 00130022 (seq 5) timeout
[  100.052145] mt7996e 0000:01:00.0: Message 00120035 (seq 6) timeout
[  120.504550] mt7996e 0000:01:00.0: Message 00130022 (seq 7) timeout
[  140.965926] mt7996e 0000:01:00.0: Message 00120035 (seq 8) timeout
```

As it was mentioned in the [PR](https://github.com/openwrt/openwrt/pull/18750) to ensure if that is working on latest snapshot build and I can confirm - it works normally:

```
# Limit 2GB for NAND:
root@BPI-R4:~# fw_printenv | grep bootargs
bootargs=console=ttyS0,115200n1 pci=pcie_bus_perf root=/dev/fit0 rootwait ubi.block=0,fit mem=2048M

root@BPI-R4:~# cat /etc/modules.conf
options mt7996e wed_enable=Y sr_scene_detect=Y

root@BPI-R4:~# cat /sys/module/mt7996e/parameters/wed_enable
Y

root@BPI-R4:~# uname -a
Linux BPI-R4 6.6.93 #0 SMP Sat Jun  7 19:13:23 2025 aarch64 GNU/Linux
```

### OpenWrt version

r29982-975f4de1c8

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

Banana Pi BPI-R4 (2x SFP+)b

### Image kind

Self-built image

### Steps to reproduce

Build image with `CONFIG_TESTING_KERNEL=y`.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

LorenzoBianconi, 2025-06-12:
@dangowrt I am afk too at the moment. I can take a look on Sunday

CoelacanthusHex, 2025-06-16:
On my BPI-R4, it also happened on 6.6 kernel.
```
[   92.633908] mt7996e 0000:01:00.0: Message 001a0002 (seq 3) timeout
[  113.114584] mt7996e 0000:01:00.0: Message 00120035 (seq 4) timeout
[  133.597224] mt7996e 0000:01:00.0: Message 00130022 (seq 5) timeout
[  154.077608] mt7996e 0000:01:00.0: Message 00120035 (seq 6) timeout
[  174.557637] mt7996e 0000:01:00.0: Message 00130022 (seq 7) timeout
[  195.037038] mt7996e 0000:01:00.0: Message 00120035 (seq 8) timeout
```
OpenWRT version: r29634-e3bec5692c
kernel version: Linux magnesium 6.6.90 #0 SMP Fri May 16 11:16:15 2025 aarch64 GNU/Linux

danpawlik, 2025-06-16:
> On my BPI-R4, it also happened on 6.6 kernel.
>
> ```
> [   92.633908] mt7996e 0000:01:00.0: Message 001a0002 (seq 3) timeout
> [  113.114584] mt7996e 0000:01:00.0: Message 00120035 (seq 4) timeout
> [ [...]

Stupid question and sorry for that, but just want to be sure: you have also limited RAM to 2GB, right?

brada4, 2025-06-18:
What is this? Has it worked before?
sr_scene_detect=Y

danpawlik, 2025-06-26:
After consulting @LorenzoBianconi  in a private conversation, where he recommended removing from `wan` the `br-wan` and just set one interface (in my case `sfp-wan`), it does not work.

```
root@BPI-R4:~# cat /etc/modules.conf
options mt7996e wed_enable=Y sr_scene_detect=Y

root@BPI-R4:~# free -m
              total        used        free      shared  buff/cache   available
Mem:        2040732      153176     1802932         416       84624     1836440
Swap:             0           0           0

# /etc/config/network
config device
        option name 'br-wan'
        option type 'bridge'
        option macaddr 'B6:XXX:34'
        list ports 'wan'

config device
        option name 'wan'
        option macaddr 'B6:XXX:34'

config device
        option name 'sfp-wan'
        option macaddr 'B6:XXX:34'

config interface 'wan'
        option device 'sfp-wan'
        option proto 'static'
        option ipaddr '8.XXX.7.2'
        option netmask '255.255.255.0'
        option gateway '8.XXX.7.1'
        option peerdns '0'
        option ipv6 '0'

config interface 'wan6'
        option device 'sfp-wan'
        option proto 'dhcpv6'
        option auto '0'
        option reqaddress 'try'
        option reqprefix 'auto'
        option norelease '1'
```

Even after complete removing `br-wan` and keep wan + sfp-wan:

```
config device
        option name 'wan'
        option macaddr 'B6:XXX:34'

config device
        option name 'sfp-wan'
        option macaddr 'B6:XXX:34'

config interface 'wan'
        option device 'sfp-wan'
        option proto 'static'
        option ipaddr '8.XXX.7.2'
        option netmask '255.255.255.0'
        option gateway '8.XXX.7.1'
        option peerdns '0'
        option ipv6 '0'

config interface 'wan6'
        option device 'sfp-wan'
        option proto 'dhcpv6'
        option auto '0'
        option reqaddress 'try'
        option reqprefix 'auto'
        option norelease '1'
```

it ends with (after reboot):

```
[  117.609081] mt7996e 0000:01:00.0: Message 001a0002 (seq 15) timeout
[  138.089137] mt7996e 0000:01:00.0: Message 00120035 (seq 1) timeout
[  158.569178] mt7996e 0000:01:00.0: Message 00130022 (seq 2) timeout
[  179.049180] mt7996e 0000:01:00.0: Message 00120035 (seq 3) timeout
[  199.529163] mt7996e 0000:01:00.0: Message 00130022 (seq 4) timeout
[  220.009148] mt7996e 0000:01:00.0: Message 00120035 (seq 5) timeout
[  240.489086] mt7996e 0000:01:00.0: Message 00130022 (seq 6) timeout
[  260.968982] mt7996e 0000:01:00.0: Message 00120035 (seq 7) timeout
```

danpawlik, 2025-07-10:
@LorenzoBianconi I was trying to switch mt76 project to earlier commit, but it raises issues with mac80211 when I was doing customize image. I don't have enough knowledge here, but maybe someone else have an image with kernel 6.12 where WED is working?

danpawlik, 2025-07-17:
```
curl -SL https://pastebin.com/raw/n5xW9ScK > target/linux/mediatek/patches-6.12/999-lorenzo-dts.patch
```

Wooo! It works @LorenzoBianconi , thank you 

```
root@BPI-R4:~# cat /sys/module/mt7996e/parameters/wed_enable
Y
```
and no timeout

```
root@BPI-R4:~# dmesg  | grep timeout
[    1.905000] mtk-wdt 1001c000.watchdog: Watchdog enabled (timeout=31 sec, nowayout=0)
```

Still 2GB probably is needed, but WED is working \o/

dangowrt, 2025-07-17:
@danpawlik When using OpenWrt all you stated above is currently true. There is another approach for layer 2 offloading pending upstream approval implemented by @ericwoud which doesn't require the `bridger` process to create conntrack entries for L2 streams, see: https://patchwork.kernel.org/project/netdevbpf/list/?series=951145&archive=both&state=*

mhalano, 2025-07-18:
Do we still need to set RAM to 2GB?

danpawlik, 2025-07-18:
@mhalano yes. I asked @LorenzoBianconi if he can take a look. Odd,that  build with MTK feed does not need such limitation,so it means they have working patch,but why don't they publish to kernel? Or maybe it is but it needs to be ported to openwrt.