openwrt/openwrt #19095: Essential downgrade in wifi perfomance for x86/64 + usb dongle Asus USB-AC51

icegood, 2025-06-11:
### Describe the bug

Same issue as in 
https://github.com/openwrt/openwrt/issues/17599
(very same spot device installed and very same client checks in)

### OpenWrt version

r28597-0425664679

### OpenWrt release

24.10.1

### OpenWrt target/subtarget

x86/64

### Device

ChangWang CW56-58

### Image kind

Official downloaded image

### Steps to reproduce

1. Driver is kmod-mt76x0u
2. Try to connect to wifi once AP is properly installed.

### Actual behaviour

Accessible only if the device is close enough

### Expected behaviour

Accessible if the device is in another room
like in 
https://github.com/morrownr/USB-WiFi/discussions/367

### Additional info

**root@ice_nas_new:/sys/kernel/debug/ieee80211/phy0/mt76# hexdump -C eeprom** 
```
00000000  10 76 01 02 7c 10 c9 75  96 d0 ff ff ff ff ff ff  |.v..|..u........|
00000010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00000020  ff ff ff 17 ef ff 50 3e  3c 9b 40 ff ff ff ff ff  |......P><.@.....|
00000030  ff ff ff ff 11 fd 04 30  1e 01 4a 01 55 55 99 bb  |.......0..J.UU..|
00000040  ff 08 ff ff 00 09 01 01  00 0c 00 00 00 0c 12 10  |................|
00000050  80 80 05 05 05 05 05 05  05 05 05 05 05 05 05 05  |................|
00000060  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff 86 f0  |................|
00000070  ff ff ff ff ff 24 00 00  05 05 05 05 05 05 05 05  |.....$..........|
00000080  05 05 05 05 05 05 05 05  05 05 05 05 05 05 05 05  |................|
00000090  05 05 05 05 05 05 05 05  05 05 05 ff ff ff ff ff  |................|
000000a0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
000000d0  1e f8 1c 00 40 74 95 ff  ff ff ff 00 64 9b 03 03  |....@t......d...|
000000e0  3e 3c 3e 3c 3f 3c 3e 3c  3e 3b 3e 3c 3e 3b 3e 3c  |><><?<><>;><>;><|
000000f0  b3 ba b8 b0 b6 b2 bf a2  ff ff ff ff ff ff ff ff  |................|
00000100  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00000110  ff ff ff ff ff ff ff ff  ff ff ff 1f 07 00 ff ff  |................|
00000120  0c 09 05 03 0c 09 05 03  09 07 02 01 01 01 ff ff  |................|
00000130  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00000140  12 03 4d 00 65 00 64 00  69 00 61 00 54 00 65 00  |..M.e.d.i.a.T.e.|
00000150  6b 00 0a 03 57 00 69 00  46 00 69 00 08 03 31 00  |k...W.i.F.i...1.|
00000160  2e 00 30 00 ff ff ff ff  ff ff ff ff ff ff ff ff  |..0.............|
00000170  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
000001c0  01 02 05 0b d1 17 00 01  4a 00 00 00 a0 50 01 00  |........J....P..|
000001d0  40 00 08 00 02 00 02 00  01 40 52 5c ff ff ff ff  |@........@R\....|
000001e0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00000200
```

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

icegood, 2025-06-11:
@DragonBluep , do you happen to know? Is it just devil place? Or report 
https://github.com/morrownr/USB-WiFi/discussions/367 is unfair?
But in my first bug it worked on 23.07 as well...

brada4, 2025-06-11:
Some power saving?
What is the purpose of hexdump-ing proprietary fw blob?

icegood, 2025-06-12:
> Some power saving? What is the purpose of hexdump-ing proprietary fw blob?

was asked in other ticket before

> That device doesn't really like USB 3.0 ports, try with 2.0.

seems it was already on 2.0 but let me recheck