openwrt/openwrt #19111: Linksys MR7350 Ethernet LEDs staying in "emergency dispatch" mode

snakesandstuff, 2025-06-12:
I asked for information over in the IRC channel and was directed to report an issue here. I apologize if I have posted this in the wrong place. If so, please advise and I'll make the post in the correct place.

I recently built an image (r29992-628b3ff2c3) for a MR7350 and the LEDs seem to flash in "emergency dispatch" mode (as described by someone on IRC in #openwrt).

As suggested, I am tagging @robimarko 

The output of cat /sys/class/leds/90000.mdio-1:00:green:lan/trigger is "none timer heartbeat default-on [netdev]" so that seems to be set correctly.

The emergency dispatch does not change even after plugging in an ethernet cable and ssh'ing into the device. The radios are disabled and it is not connected to anything on the WAN port.

Please advise.

snakesandstuff, 2025-06-12:
Additional info:

I tried to flash the other boot partition and test it with the snapshot build from the openwrt build server at https://downloads.openwrt.org/snapshots//targets/qualcommax/ipq60xx/openwrt-qualcommax-ipq60xx-linksys_mr7500-squashfs-sysupgrade.bin

I verified the checksum, did a sysupgrade -v [sysupgrade.bin file] and the router did *not* reboot and come back up.

I interrupted the boot process a few times to get it back to the previous in house boot image and it came back up on that image (the one producing the emergency dispatch lights).

I'll quit stabbing in the dark and await further info and will be glad to donate my time/hardware/output if it may help.

snakesandstuff, 2025-06-13:
I am not sure of the revision, but here are some pictures of the board:

Top:
https://snakesandstuff.com/share/openwrt/top.png

Bottom:
https://snakesandstuff.com/share/openwrt/bottom.png

And the flashing lights:
https://snakesandstuff.com/share/openwrt/blinky.mp4

closed 2025-06-16 by snakesandstuff

BreachableBase, 2025-06-18:
Also getting this issue, continuation of [this thread](https://forum.openwrt.org/t/linksys-mr7350-w-error-writing-image/221567) I think.. Any support on this?

Thue, 2025-06-29:
I think I might have the same issue - I created https://github.com/openwrt/openwrt/issues/19260

From my poking around, I think the problem is that the kernel "partition" on the preformatted flash is 8MiB, while the OpenWRT image is 15MiB.

Shine-, 2025-09-05:
Try [this](https://github.com/openwrt/openwrt/issues/19261#issuecomment-3249890408). If this is really due to the partition vs. image size mismatch, it should work (due to `kernel` and `rootfs` being adjacent, according to the boot log).