openwrt/openwrt #19121: 10G ports of QNAP QHora-301w Qualcommax not working properly with Main branch

sppmasterspp, 2025-06-13:
QNAP QHora-301w (Qualcommax) has two 10G ports Aquantia AQR113c.
This is dedicated device forum thread - https://forum.openwrt.org/t/adding-openwrt-support-for-qnap-qhora-301w/96934/922
There are several reports from different users that one or the other 10G ports just don't work properly and that changes with different commits of OpenWrt Main branch. They cause different issues and produce different error messages.
There were/are different issues with the AQR 10G Firmware not loading or simply the 10G-1 port gives errors.
Recently there was a commit that broke 10g-2 port with kernel 6.6 and later 10g-1 port with kernel 6.12.
https://github.com/openwrt/openwrt/commit/cc50cac8a0fd0e12fc9e2a7fa7d14f99a2a5ffae
reported here - https://forum.openwrt.org/t/adding-openwrt-support-for-qnap-qhora-301w/96934/868?u=sppmaster
and confirmed by several users.

Some examples from the forum. 
```
Error messages that can be seen:
[   30.117803] Aquantia AQR113C 90000.mdio-1:08: CRC mismatch: calculated 0x7011 PHY 0xffff
[   30.117853] Aquantia AQR113C 90000.mdio-1:08: firmware loading failed: -22
[   30.135101] Aquantia AQR113C 90000.mdio-1:08: Direct firmware load for marvell/AQR-G4_v5.4.C-AQR_CIG_WF-1945_0x8_ID44776_VER1630.cld failed with error -2
[   30.135166] Aquantia AQR113C 90000.mdio-1:08: Falling back to sysfs fallback for: marvell/AQR-G4_v5.4.C-AQR_CIG_WF-1945_0x8_ID44776_VER1630.cld
[   30.382519] kmodloader: loading kernel modules from /etc/modules.d/*
[   30.528283] urngd: v1.0.2 started.
[   40.853318] Aquantia AQR113C 90000.mdio-1:08: failed to find FW file marvell/AQR-G4_v5.4.C-AQR_CIG_WF-1945_0x8_ID44776_VER1630.cld (-12)
[   40.853383] Aquantia AQR113C 90000.mdio-1:08: probe with driver Aquantia AQR113C failed with error -12
```
```
[  267.771052] nss-dp 3a007000.dp6-syn 10g-2: PHY Link is down
[  267.771413] br-lan: port 2(10g-2) entered disabled state
[  302.672476] uniphy autoneg time out!
[  303.131636] nss-dp 3a001800.dp5 10g-1: PHY Link up speed: 2500
[  303.131902] br-lan: port 1(10g-1) entered blocking state
[  303.136387] br-lan: port 1(10g-1) entered forwarding state
[  309.371330] nss-dp 3a001800.dp5 10g-1: PHY Link is down
[  309.371690] br-lan: port 1(10g-1) entered disabled state
[  315.612185] nss-dp 3a001800.dp5 10g-1: PHY Link up speed: 2500
[  315.612451] br-lan: port 1(10g-1) entered blocking state
[  315.616932] br-lan: port 1(10g-1) entered forwarding state
[  315.632585] uniphy autoneg time out!
[  316.651357] nss-dp 3a001800.dp5 10g-1: PHY Link is down
[  316.651718] br-lan: port 1(10g-1) entered disabled state
[  322.891653] nss-dp 3a001800.dp5 10g-1: PHY Link up speed: 2500
[  322.891968] br-lan: port 1(10g-1) entered blocking state
[  322.896406] br-lan: port 1(10g-1) entered forwarding state
[  323.003107] uniphy autoneg time out!
```
Or even there was a case with the 10g-2 port cannot pass any traffic but there was no error message at all.

Ansuel, 2025-06-25:
Thanks for looking intro this... Honestly why we are using the ssdk aqr driver? I feel the dedicated one should be mature enough now?

sppmasterspp, 2025-10-06:
@robimarko @Ansuel 
Currently there is workaround provided by @rmandrad and tested from him. Working for me too,
https://forum.openwrt.org/t/adding-openwrt-support-for-qnap-qhora-301w/96934/944?u=sppmaster

robimarko, 2025-12-29:
That is quite literaly just the linux driver pulled and put there, completely useless

Sebdev43, 2026-10-25:
Test report: both 10G ports work on a QNAP QHora-301W with current main.

**Build:** official snapshot `r37657-b5bc4becb4`, kernel 6.18.55, qualcommax/ipq807x, profile `qnap_301w` (images from downloads.openwrt.org, no changes). Flashed from QuRouter 2.7.2 to entry 0 (`0:HLOS` + `rootfs`).

**AQR113C firmware:** loaded from NVMEM, i.e. the QNAP `.cld` images kept in `0:ethphyfw1` / `0:ethphyfw2`, no `aq_fw_load` in the bootcmd:

```
[    9.095055] Aquantia AQR113C 90000.mdio-1:00: loading firmware version 'v5.4.C CIG WF-1945_0x0 060120 02:47:48' from 'NVMEM'
[   22.885071] Aquantia AQR113C 90000.mdio-1:08: loading firmware version 'v5.4.C CIG WF-1945_0x8 060120 02:47:48' from 'NVMEM'
[   67.514454] qca-ppe 3a000000.ppe 10g-1: Link is Up - 10Gbps/Full - flow control off
[ 5249.273927] qca-ppe 3a000000.ppe 10g-2: Link is Up - 10Gbps/Full - flow control rx/tx
```

- `10g-1`: WAN to a Freebox (10GBASE-T), negotiates 10Gbps, DHCP and routing fine.
- `10g-2`: X540-T2 on a Proxmox host, negotiates 10Gbps after the host's power cycle.
- No `uniphy autoneg time out!` anymore. Up ~2 h so far, stable.

The `uniphy autoneg` failure is gone on this hardware with current main. Thanks @robimarko for 8e4892f1d0.

Side note, unrelated to this issue: with a VLAN-aware bridge, frames entering an untagged port leave a tagged port with VID 0 (#25904). That matches the fix pending in #24806.

Ansuel, 2026-10-25:
@Sebdev43 the VID 0 thing seems very relevant... @JuliusBairaktaris is it possible to have a fix for it outside the massive 56 commit pr?

JuliusBairaktaris, 2026-10-25:
Sure why not. I can check it next weekend. 

I will split out everything that doesn't need a patch.