openwrt/openwrt #19130: WiFi download speed degraded on Mediatek Filogic Xiaomi AX3000T with kernel 6.12 and VLANs, upload is fine

dadogroove, 2025-06-13:
### Describe the bug

I first noticed this when I upgraded my AX3000T from 6.6 to 6.12 keeping my config.  On 6.6 everything was working fine. After the upgrade the wifi download speed went from 300 (my ISP max) down to about 20. The upload side was unaffected. I use the AX3000T as an AP behind a x86 running 6.12. It has 3 VLANs. I also tried a fresh build and without the VLANs the speed is normal. As soon as I added a VLAN the attached wifi speed dropped.

### OpenWrt version

r30083-d7f9e240c2

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

Xiaomi AX3000T

### Image kind

Self-built image

### Steps to reproduce

Install fresh version of OpenWrt SNAPSHOT and set up VLAN, attach wifi interface, see download speed drop

### Actual behaviour

Wifi download speed drops when using VLAN

### Expected behaviour

Wifi speed should not change due to VLAN use

### Diffconfig

```text
# Use "make defconfig" to expand this to a full .config
CONFIG_TARGET_mediatek=y
CONFIG_TARGET_mediatek_filogic=y
CONFIG_TARGET_DEVICE_mediatek_filogic_DEVICE_wavlink_wl-wn586x3=y
CONFIG_TARGET_PER_DEVICE_ROOTFS=y

CONFIG_TESTING_KERNEL=y

# Add-on programs
CONFIG_PACKAGE_htop=y
CONFIG_PACKAGE_openssh-sftp-server=y

# Luci
CONFIG_PACKAGE_luci=y

# Luci statistics
CONFIG_PACKAGE_luci-app-statistics=y
CONFIG_PACKAGE_collectd-mod-thermal=y
CONFIG_PACKAGE_collectd-mod-wireless=y
```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

dadogroove, 2025-06-13:
Sorry for the mistake in the Diffconfig. It should be the Xiaomi AX3000T not the Wavlink but the rest is correct.

dadogroove, 2025-06-14:
It is an early version of the AX3000T. It has the Mediatek MT7531AE switch hardware. I am only noticing the speed issue on the 5G interface. The 2.4 interface is only for a thermostat and a link up to a camera system. These configs work fine with 6.6 but if I do a straight system upgrade to 6.12 everything works fine except for the 5G speed decrease. I also have the Wavlink WL-WN586X3 (Filogic & Mediatek MT7531AE) and see the same issue with it as well. I tried an upgrade w/o saving config (fresh install) and set up one VLAN and also see the same issue, 5G speed decrease. 

**This is the network config:**

config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fd13:cfa5:1e37::/48'
	option packet_steering '0'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'lan2'
	list ports 'lan3'
	list ports 'lan4'

config interface 'lan'
	option device 'br-lan.30'
	option proto 'static'
	option ipaddr '192.168.2.11'
	option netmask '255.255.255.0'
	option ip6assign '60'
	list dns '192.168.2.7'
	option gateway '192.168.2.7'
	option ip6hint '1'

config device
	option name 'wan'

config interface 'things'
	option proto 'none'
	option device 'br-lan.50'

config bridge-vlan
	option device 'br-lan'
	option vlan '30'
	list ports 'lan2'
	list ports 'lan3'
	list ports 'lan4'

config bridge-vlan
	option device 'br-lan'
	option vlan '50'
	list ports 'lan2:t'

config bridge-vlan
	option device 'br-lan'
	option vlan '70'
	list ports 'lan2:t'

config interface 'camera'
	option proto 'none'
	option device 'br-lan.70

**This is the wireless config:**

config wifi-device 'radio0'
	option type 'mac80211'
	option path 'platform/soc/18000000.wifi'
	option band '2g'
	option channel '1'
	option htmode 'HE20'
	option country 'US'
	option cell_density '0'
    option he_su_beamformee '1'
	option he_bss_color '25'
	option txpower '18'

config wifi-device 'radio1'
	option type 'mac80211'
	option path 'platform/soc/18000000.wifi+1'
	option band '5g'
	option channel '48'
	option htmode 'HE80'
	option country 'US'
	option cell_density '0'
    option he_su_beamformee '1'
	option he_bss_color '21'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option ssid 'bungalow_5G'
	option encryption 'psk2'
	option key '*****'

config wifi-iface 'wifinet3'
	option device 'radio0'
	option mode 'ap'
	option ssid 'things'
	option encryption 'psk2'
	option isolate '1'
	option key '*****'
	option network 'things'
	option macfilter 'allow'
	option hidden '1'
	
config wifi-iface 'wifinet4'
	option device 'radio1'
	option mode 'ap'
	option ssid 'bungalow_2'
	option encryption 'psk2'
	option isolate '1'
	option key '*****'
	option network 'things'
	option disabled '1'

config wifi-iface 'wifinet5'
	option device 'radio0'
	option mode 'ap'
	option ssid 'bungalow'
	option encryption 'psk2'
	option key '*****'
	option network 'lan'
	option disabled '1'

config wifi-iface 'wifinet6'
	option device 'radio0'
	option mode 'ap'
	option ssid 'low_fi'
	option encryption 'psk2'
	option key '*****'
	option network 'camera'
	option disabled '1'

csharper2005, 2025-06-15:
I noticed the issue on the 2.4 and 5 GHz interface both.

There is no speed degradation between:
- Wireless client and AP (1 tcp stream)
`[  4]   0.00-10.00  sec   330 MBytes   277 Mbits/sec                  receiver`
- AP and upstream hosts (1 tcp stream)
`[  5]   0.00-10.00  sec   724 MBytes   607 Mbits/sec                  receiver`

Degradation between wireless clients and upstream hosts depends on `iperf3` streams count:
- 1 tcp stream:
```
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec  1.20 MBytes  10.1 Mbits/sec                  
[  4]   1.00-2.01   sec   532 KBytes  4.34 Mbits/sec                  
[  4]   2.01-3.00   sec   907 KBytes  7.46 Mbits/sec                  
[  4]   3.00-4.01   sec   693 KBytes  5.65 Mbits/sec                  
[  4]   4.01-5.00   sec   865 KBytes  7.14 Mbits/sec                  
[  4]   5.00-6.01   sec   686 KBytes  5.59 Mbits/sec                  
[  4]   6.01-7.00   sec  41.3 KBytes   340 Kbits/sec                  
[  4]   7.00-8.00   sec  1.15 MBytes  9.64 Mbits/sec                  
[  4]   8.00-9.00   sec  1.82 MBytes  15.3 Mbits/sec                  
[  4]   9.00-10.00  sec  1.51 MBytes  12.7 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-10.00  sec  9.50 MBytes  7.97 Mbits/sec                  sender
[  4]   0.00-10.00  sec  9.45 MBytes  7.93 Mbits/sec                  receiver
```
- 5 tcp streams:
```
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-10.00  sec  14.6 MBytes  12.3 Mbits/sec                  sender
[  4]   0.00-10.00  sec  14.6 MBytes  12.3 Mbits/sec                  receiver
[  6]   0.00-10.00  sec  14.8 MBytes  12.4 Mbits/sec                  sender
[  6]   0.00-10.00  sec  14.6 MBytes  12.3 Mbits/sec                  receiver
[  8]   0.00-10.00  sec  14.4 MBytes  12.1 Mbits/sec                  sender
[  8]   0.00-10.00  sec  14.2 MBytes  11.9 Mbits/sec                  receiver
[ 10]   0.00-10.00  sec  14.5 MBytes  12.2 Mbits/sec                  sender
[ 10]   0.00-10.00  sec  14.3 MBytes  12.0 Mbits/sec                  receiver
[ 12]   0.00-10.00  sec  14.2 MBytes  12.0 Mbits/sec                  sender
[ 12]   0.00-10.00  sec  14.2 MBytes  11.9 Mbits/sec                  receiver
[SUM]   0.00-10.00  sec  72.5 MBytes  60.8 Mbits/sec                  sender
[SUM]   0.00-10.00  sec  72.0 MBytes  60.4 Mbits/sec                  receiver
```
- 10 tcp streams:
```
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-10.00  sec  11.8 MBytes  9.86 Mbits/sec                  sender
[  4]   0.00-10.00  sec  11.7 MBytes  9.78 Mbits/sec                  receiver
[  6]   0.00-10.00  sec  12.1 MBytes  10.2 Mbits/sec                  sender
[  6]   0.00-10.00  sec  11.9 MBytes  10.0 Mbits/sec                  receiver
[  8]   0.00-10.00  sec  12.0 MBytes  10.1 Mbits/sec                  sender
[  8]   0.00-10.00  sec  11.9 MBytes  9.95 Mbits/sec                  receiver
[ 10]   0.00-10.00  sec  12.2 MBytes  10.3 Mbits/sec                  sender
[ 10]   0.00-10.00  sec  12.0 MBytes  10.1 Mbits/sec                  receiver
[ 12]   0.00-10.00  sec  11.9 MBytes  9.96 Mbits/sec                  sender
[ 12]   0.00-10.00  sec  11.7 MBytes  9.78 Mbits/sec                  receiver
[ 14]   0.00-10.00  sec  12.4 MBytes  10.4 Mbits/sec                  sender
[ 14]   0.00-10.00  sec  12.2 MBytes  10.2 Mbits/sec                  receiver
[ 16]   0.00-10.00  sec  11.4 MBytes  9.54 Mbits/sec                  sender
[ 16]   0.00-10.00  sec  11.2 MBytes  9.38 Mbits/sec                  receiver
[ 18]   0.00-10.00  sec  12.0 MBytes  10.1 Mbits/sec                  sender
[ 18]   0.00-10.00  sec  11.9 MBytes  9.95 Mbits/sec                  receiver
[ 20]   0.00-10.00  sec  11.8 MBytes  9.86 Mbits/sec                  sender
[ 20]   0.00-10.00  sec  11.6 MBytes  9.76 Mbits/sec                  receiver
[ 22]   0.00-10.00  sec  10.5 MBytes  8.81 Mbits/sec                  sender
[ 22]   0.00-10.00  sec  10.4 MBytes  8.71 Mbits/sec                  receiver
[SUM]   0.00-10.00  sec   118 MBytes  99.0 Mbits/sec                  sender
[SUM]   0.00-10.00  sec   116 MBytes  97.6 Mbits/sec                  receiver
```

bmork, 2025-06-16:
Not sure it's the switch.  Ethernet or phy drivers maybe? 

I see the same extreme download speed degradation on both MT7981 (Unifi 6+) and MT7621 (Unifi 6 Lite) using a tagged lan port wifi bridge configuration on both.  AFAIK, the U6+ does't have a switch.  It uses an soc internal phy attached to a soc internal mac. 

iperf3 from wifi client on U6+ running v6.12.32:
```
bjorn@miraculix:~$ iperf3 -R -V -c canardo -4 192.168.10.1
iperf 3.12
Linux miraculix 6.1.0-37-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.140-1 (2025-05-22) x86_64
Control connection MSS 1448
Time: Mon, 16 Jun 2025 14:47:33 GMT
Connecting to host canardo, port 5201
Reverse mode, remote host canardo is sending
      Cookie: tpfjnc233qwb2hooi35ss74prigjegic5amd
      TCP MSS: 1448 (default)
[  5] local 192.168.10.186 port 41842 connected to 51.175.80.185 port 5201
Starting Test: protocol: TCP, 1 streams, 131072 byte blocks, omitting 0 seconds, 10 second test, tos 0
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec   288 KBytes  2.36 Mbits/sec                  
[  5]   1.00-2.00   sec   238 KBytes  1.95 Mbits/sec                  
[  5]   2.00-3.00   sec   255 KBytes  2.09 Mbits/sec                  
[  5]   3.00-4.00   sec   226 KBytes  1.85 Mbits/sec                  
[  5]   4.00-5.00   sec   215 KBytes  1.76 Mbits/sec                  
[  5]   5.00-6.00   sec   226 KBytes  1.85 Mbits/sec                  
[  5]   6.00-7.00   sec   221 KBytes  1.81 Mbits/sec                  
[  5]   7.00-8.00   sec   229 KBytes  1.88 Mbits/sec                  
[  5]   8.00-9.00   sec   221 KBytes  1.81 Mbits/sec                  
[  5]   9.00-10.00  sec   226 KBytes  1.85 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
Test Complete. Summary Results:
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  2.38 MBytes  2.00 Mbits/sec  415             sender
[  5]   0.00-10.00  sec  2.29 MBytes  1.92 Mbits/sec                  receiver
snd_tcp_congestion cubic
rcv_tcp_congestion cubic

iperf Done.
```
iperf3 from wifi client on U6+ running v6.6.90:
```
bjorn@miraculix:~$ iperf3 -R -V -c canardo -4 192.168.10.1
iperf 3.12
Linux miraculix 6.1.0-37-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.140-1 (2025-05-22) x86_64
iperf3: error - unable to connect to server: Resource temporarily unavailable
bjorn@miraculix:~$ 
bjorn@miraculix:~$ iperf3 -R -V -c canardo -4 192.168.10.1
iperf 3.12
Linux miraculix 6.1.0-37-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.140-1 (2025-05-22) x86_64
Control connection MSS 1448
Time: Mon, 16 Jun 2025 14:51:19 GMT
Connecting to host canardo, port 5201
Reverse mode, remote host canardo is sending
      Cookie: x4dwsbtv7wyl6vmojekebnpwoj7ycpjvszvm
      TCP MSS: 1448 (default)
[  5] local 192.168.10.186 port 37434 connected to 51.175.80.185 port 5201
Starting Test: protocol: TCP, 1 streams, 131072 byte blocks, omitting 0 seconds, 10 second test, tos 0
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  38.8 MBytes   325 Mbits/sec                  
[  5]   1.00-2.00   sec  41.1 MBytes   345 Mbits/sec                  
[  5]   2.00-3.00   sec  48.6 MBytes   408 Mbits/sec                  
[  5]   3.00-4.00   sec  53.3 MBytes   447 Mbits/sec                  
[  5]   4.00-5.00   sec  53.2 MBytes   446 Mbits/sec                  
[  5]   5.00-6.00   sec  53.6 MBytes   450 Mbits/sec                  
[  5]   6.00-7.00   sec  53.7 MBytes   451 Mbits/sec                  
[  5]   7.00-8.00   sec  52.8 MBytes   443 Mbits/sec                  
[  5]   8.00-9.00   sec  52.9 MBytes   444 Mbits/sec                  
[  5]   9.00-10.00  sec  51.8 MBytes   434 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
Test Complete. Summary Results:
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   505 MBytes   423 Mbits/sec  524             sender
[  5]   0.00-10.00  sec   500 MBytes   419 Mbits/sec                  receiver
snd_tcp_congestion cubic
rcv_tcp_congestion cubic

iperf Done.
```

Results on the MT7621 based U6 Lite are similar

bmork, 2025-06-17:
Definitely related to wifi + tagged vlan bridging.  A simplified version of my normal config has:
```
root@u6plus:~# brctl show
bridge name     bridge id               STP enabled     interfaces
br-lan          7fff.e43883e585d4       no              eth0.10
                                                        phy1-ap0
```

Downloading to wifi client directly from the AP is fine:
```
[  5]   0.00-10.00  sec   446 MBytes   374 Mbits/sec                  receiver
```

Downloading from to wifi client from a server on the eth0.10 vlan is terrible:
```
[  5]   0.00-10.00  sec  4.09 MBytes  3.43 Mbits/sec                  receiver
```

Configuring the remote switch port to use vid 10 as pvid+untagged and reconfiguring the brigde to
```
root@u6plus:~# brctl show
bridge name     bridge id               STP enabled     interfaces
br-lan          7fff.e43883e585d4       no              eth0
                                                        phy1-ap0
```
results in normal download from the lan server:
```
[  5]   0.00-10.00  sec   342 MBytes   287 Mbits/sec                  receiver
```

Note that there were no other changes than simply making the lan port untagged.

luizluca, 2025-06-17:
> Definitely related to wifi + tagged vlan bridging. 

That is odd. Is it really wifi or just anything that bridges through the CPU? Would an USB Ethernet also behave like the wifi?

csharper2005, 2025-06-17:
> anything that bridges through the CPU?

Exactly like that. There is also noticed between tagged and untagged eth ports.

csharper2005, 2025-06-17:
> Looks like untagging/tagging takes the software/slow path instead of being offloaded to Ethernet MAC or switch hardware.

Interesting that:
- There is no degradation if the destination is device itself (tagged port). The speed decrease is observed only when sending packets to the wireless interface or another port via the eth bridge. Does this agree with your assumption?
- Tagging works without any performance issues. Only untagging is working badly.

bmork, 2025-06-20:
The degradation is much worse that what would be expected from software forwarding.  It's due to packet corruption.  

Snooping I see lots of issues like dup acks, retransmissions, out-of-order packets etc. This is the reason for the low speed. I believe the root cause is that some of the frames are messed up like this one (which will be marked as invalid IP by wireshark):

```
0000  44 85 00 11 26 e6 00 1b 21 a7 98 bc 08 00 00 0a   D...&...!.......                                                                                                                                                                
0010  05 e0 85 66 05 dc 85 66 53 60 c0 a8 0a 01 c0 a8   ...f...fS`......                                                                                                                                                                
0020  0a ba c0 a8 0a ba 14 51 c8 fe 0b 2b 4b cf 8f b0   .......Q...+K...                                                                                                                                                                
0030  d9 bb 80 10 00 aa 56 2f 00 00 01 01 08 0a 54 f1   ......V/......T.                                                                                                                                                                
0040  93 b0 fd 6b f8 fb 3e d8 18 7d 88 4a d9 04 d1 22   ...k..>..}.J..."                                                                                                                                                                
0050  2c 51 69 58 73 41 d3 68 b2 e1 8a 95 85 63 5b fb   ,QiXsA.h.....c[.                                                                                                                                                                
0060  24 f6 7c ac 1d 59 a1 60 a8 38 03 1b 10 b1 32 2e   $.|..Y.`.8....2.                                                                                                                                                                
0070  67 04 f6 3a 0b 71 ca 82 17 f8 e2 82 9e 13 85 0d   g..:.q..........                                                                                                                                                                
etc
```
This is captured on the wifi client 44 85 00 11 26 e6 downloading from 00 1b 21 a7 98 bc.  The packet is forwarded by a U6+ AP bridging VID 10 on the lan side to wifi.  

As you can see, the 14 byte ethernet header looks fine, having ethertype 0x0800.  But the follwing data is not IPv4 (we expected 0x45 and god 0x00).  0x000a coud be the VID. But there is more corruption here.  This should have been part of an iperf3 tcp stream from port 5201 (0x1451).  We find that at offset 0x26, while it should have been at 0x22.  

Looking at the 4 byte from 0x10 on, they seem to be duplicated from 0x14.  Except that something has adjusted the IP packet length from 0x05dc (1500) to 0x05e0 (1504)  Hardly arbitrary numbers..

There is something very fishy going on in the bridging code

Interesting to learn that wifi is irrelevant.  But that's not something I have equipment to test. My affected devices only have one ethernet port.

nbd168, 2025-07-05:
I finally managed to debug this issue and figure out a way to fix it. I will push a fix soon.

nbd168, 2025-07-05:
The fix is in my staging tree at https://git.openwrt.org/?p=openwrt/staging/nbd.git;a=summary
("kernel: fix fraglist GRO on linux 6.12")
Please test, so that I can push it to main and set it upstream soon.

bmork, 2025-07-05:
https://git.openwrt.org/?p=openwrt/staging/nbd.git;a=commit;h=5149ede9a0db84ef882c94f7d6a154738dec9cb9 works for me.

Tested IPv4 and IPv6 udp and tcp on a "Ubiquiti UniFi 6 Plus"

closed 2025-07-05 by openwrt-bot

nbd168, 2025-07-05:
Thanks for testing

Mile-Lile, 2025-07-13:
@namiltd 
this is fixed, when can we expect switch to 6.12 so that we lower users can test it...