openwrt/openwrt #19147: Asus RT-AX53U unable to boot with openwrt after clean installation of 24.10.0

madhusudhanarya, 2025-06-15:
### Describe the bug

Bought a new Asus RT-AX53U router with the intention of flashing OpenWRT to replace my existing Archer C6 AC1200 running OpenWRT. 

Followed the instructions described in (https://openwrt.org/toh/asus/rt-ax53u) to install the latest factory image. Upon reboot, the router restarted with stock firmware. Tried different versions of OpenWRT (22.03.6/23.05.3) but no luck in rebooting the router with OpenWRT installed.

### OpenWrt version

NA

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

ramips/mt7621

### Device

ASUS RT-AX53U

### Image kind

Official downloaded image

### Steps to reproduce

* Download any supported OpenWRT version
* Copy the image to the router using `cat` command as described in the documentation
* Run `mtd-write -i IMAGE_NAME -d Kernel` and reboot (or power cycle if stuck)
* Router will take few seconds to boot and will come online with stock firmware instead of OpenWRT

### Actual behaviour

* Rebooting after the installation brings the router back with stock firmware

### Expected behaviour

* Upon successful installation of the firmware, the router should boot with OpenWRT installed

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

madhusudhanarya, 2025-06-15:
I couldn't include a version because the router didn't boot with openWRT installed.

madhusudhanarya, 2025-06-16:
While checking whether I am following the correct process for flashing, came across below comment from Reddit where the users are reporting that, after flashing, the router has booted with stock firmware instead of openWRT.

https://www.reddit.com/r/openwrt/comments/1hdhngk/how_to_install_openwrt_on_asus_rtax53u/

brada4, 2025-06-16:
> few seconds to boot

3 minutes according to their android app.

> any supported OpenWRT version

factory image for 24.10.1 or 23.05.5 , not any image.
and verify sha1sum of download file and one on the device. helps keeping on-device filenames memorably short.

> reboot (or power cycle)
type reboot command and power cycle only if leds show apparent boot loop after 10 minutes.

What is the model name and any fw version on the back sticker? Fw version installed via oem web or app?

brada4, 2025-06-16:
redfit users are not following install guide, oem web upgrade form rejects invalid image and nothing happens.

madhusudhanarya, 2025-06-17:
I followed the exact same step as described in the documentation and this time, used `wget` command instead of `cat` and downloaded `openwrt-23.05.5-ramips-mt7621-asus_rt-ax53u-squashfs-factory.bin` image and installed it via `mtd-write` command.

The installation was successful, and the router booted with OpenWRT. Later, I did a sysupgrade to 24.10.0 via luci web and rebooted the roter, came online with 24.10.0 installed.

But when i checked the wireless, i don't see any wireless devices in the router. Below is the screenshot of the router just after it rebooted from success upgrade to 24.10.0

[image]

can you advise what could be the issue with wireless and how i can resolve it?

brada4, 2025-06-17:
sha256sum is correct.
i got 2 of these working well
what is in `lspci` from `pciutils` output - should be 2 mediatek cards.

closed 2025-06-17 by madhusudhanarya

brada4, 2025-06-17:
But it being absent from default install is a big problem.