openwrt/openwrt #19151: [24.10]: MT6000 (possibly other devices) images have aquantia kmod - MT6000 has no aquantia

systemcrash, 2025-06-15:
### Describe the bug

I do not recall that the MT6000 has any aquantia hardware. But the 24.10.1 image has this kmod nonetheless. I cannot see where it gets added - the Makefiles look clean. 

It's set here in `target/linux/mediatek/mt7622/config-6.6`: `CONFIG_AQUANTIA_PHY=y`

Seemingly added there in cd509d739a8ec8a548fb7db62dab88ca13c38a7c ( ping @dangowrt ) 

Currently, it's harmless - but if it shouldn't be there, it shouldn't be there. 

### OpenWrt version

r28597-0425664679

### OpenWrt release

24.10.1

### OpenWrt target/subtarget

mediatek/filogic

### Device

GL.iNet GL-MT6000

### Image kind

Official downloaded image

### Steps to reproduce

`lsmod | grep aqua`

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

andrewjlamarche, 2025-06-16:
Filogic devices, such as https://wikidevi.wi-cat.ru/TP-LINK_Archer_BE805, may be AQR113C PHYs. I don't see a problem here, since most Filogic targets have plenty of storage. For older Mediatek chips, probably not necessary.

systemcrash, 2025-06-17:
Yeah, it's definitely not critical.

Consider the possible event that some bug creeps into this kernel mod which causes some kernel crash. Devices which don't require the module should be protected from such risks.

systemcrash, 2025-06-18:
@KA2107 I believe those devices depend on it due to aquantia hardware

systemcrash, 2025-06-21:
It's not included as a package I don't think, it's included in the kernel config. It looks like those devices which need the module have it as a +kmod.

dangowrt, 2025-06-25:
`kmod-phy-aquantia` was added to the target-wide set of default packages in order to always have it included also in initramfs images. This consideration is now obsolete as also initramfs filesystem is generated per device since commit 97fd059e7e6a75766da4b8c9649b105b17e3acb3. Hence we can remove it from `target.mk` and include it only for the boards actually coming with that phy.