openwrt/openwrt #19164: [24.10.1] mt76: mt7681/mt7915e PCI not probing after boot, no radio

f00b4r0, 2025-06-17:
### Describe the bug

Since 24.10, a new random bug is occurring on (at least) ramips devices featuring mt7621/mt7915e combination (e.g. Yuncore AX820). Every now and then (but not every time) after reboot the PCI probe fails to detect any hardware resulting in no wireless operation possible (radios are attached to PCI bus).

A full power cycle is required to restore normal operation, "reboot" is not enough. Also this doesn't seem to occur after a cold boot, suggesting the hardware may be left in some invalid state upon reboot. I wonder if it may be related to the firmware crashes workarounds on this platform.

@nbd168 

### OpenWrt version

r28597-0425664679

### OpenWrt release

24.10.1

### OpenWrt target/subtarget

ramips/mt7621

### Device

YunCore AX820

### Image kind

Official downloaded image

### Steps to reproduce

reboot device until occurrence

### Actual behaviour

Symptom is: randomly upon *reboot*, the PCI probe fails to detect any hardware (see attached dmesg):

```
[    0.625580] mt7621-pci 1e140000.pcie: host bridge /pcie@1e140000 ranges:
[    0.632298] mt7621-pci 1e140000.pcie:   No bus range found for /pcie@1e140000, using [bus 00-ff]
[    0.641042] mt7621-pci 1e140000.pcie:      MEM 0x0060000000..0x006fffffff -> 0x0060000000
[    0.649156] mt7621-pci 1e140000.pcie:       IO 0x001e160000..0x001e16ffff -> 0x0000000000
[    1.015941] mt7621-pci 1e140000.pcie: pcie0 no card, disable it (RST & CLK)
[    1.022838] mt7621-pci 1e140000.pcie: pcie1 no card, disable it (RST & CLK)
[    1.029808] mt7621-pci 1e140000.pcie: pcie2 no card, disable it (RST & CLK)
[    1.036695] mt7621-pci 1e140000.pcie: nothing connected in virtual bridges
```

### Expected behaviour

Contrast this to a "normal" boot:

```
[    0.626884] mt7621-pci 1e140000.pcie: host bridge /pcie@1e140000 ranges:
[    0.633523] mt7621-pci 1e140000.pcie:   No bus range found for /pcie@1e140000, using [bus 00-ff]
[    0.642312] mt7621-pci 1e140000.pcie:      MEM 0x0060000000..0x006fffffff -> 0x0060000000
[    0.650419] mt7621-pci 1e140000.pcie:       IO 0x001e160000..0x001e16ffff -> 0x0000000000
[    1.015845] mt7621-pci 1e140000.pcie: pcie2 no card, disable it (RST & CLK)
[    1.022748] mt7621-pci 1e140000.pcie: PCIE0 enabled
[    1.027626] mt7621-pci 1e140000.pcie: PCIE1 enabled
[    1.032595] PCI coherence region base: 0x60000000, mask/settings: 0xf0000002
[    1.039805] mt7621-pci 1e140000.pcie: PCI host bridge to bus 0000:00
```

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

neheb, 2025-06-22:
maybe @paraka knows

f00b4r0, 2025-06-25:
It's happening from 24.10 and *as far as I can tell* only after a firmware crash + attempted firmware restart has occurred prior to the reboot.

f00b4r0, 2025-06-27:
I'll begin testing #19239 to see if this issue goes away with it.

dfateyev, 2025-06-27:
Interesting, we discussed [a similar issue](https://github.com/openwrt/openwrt/pull/11220) with ports initialization delay 2.5 years ago — still actual.