openwrt/openwrt #19176: [24.10] Ethernet speed issue (800 Mbps to 100 Mbps) since between 2025-06-10 and 2025-06-15 on Linksys WRT3200ACM

skynet-gh, 2025-06-18:
### Describe the bug

Some commit on the [24.10 branch](https://github.com/openwrt/openwrt/commits/openwrt-24.10/) between [0db2af95bc](https://github.com/openwrt/openwrt/commit/0db2af95bcb9d82c2af2edf6c5401d3809881114) (last known good) and [26c2ff9e5d](https://github.com/openwrt/openwrt/commit/26c2ff9e5d341a88e7d6011982c3978b9444ea12) (first bad, through current) has degraded my (wired) network performance from roughly 800 Mbps down to 100 Mbps:

[image]

I don't know how to test versions between these 2 commits to narrow it down further.

### OpenWrt version

r28732-989b12999c

### OpenWrt release

24.10-SNAPSHOT

### OpenWrt target/subtarget

mvebu/cortexa9

### Device

Linksys WRT3200ACM

### Image kind

Official downloaded image

### Steps to reproduce

Network config:

```
# cat /etc/config/network 

config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'lan1'
	list ports 'lan2'
	list ports 'lan3'
	list ports 'lan4'

config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option ipaddr '192.168.1.1'
	option netmask '255.255.255.0'
	option ip6assign '60'
	option ipv6 '0'
	option delegate '0'

config device
	option name 'wan'
	option macaddr '<snip>'

config interface 'wan'
	option device 'wan'
	option proto 'dhcp'
	option ipv6 '0'

config interface 'wan6'
	option device 'wan'
	option proto 'dhcpv6'
	option disabled '1'
	option reqaddress 'none'
	option reqprefix 'auto'
```

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

skynet-gh, 2025-06-19:
> What is the indicated link speed now?

Oh under Port status it said `100 MbE`, I guess I should've checked that. I unplugged the WAN Ethernet cable on both ends and plugged it back in, now it's showing as `1 GbE` so maybe just a cable issue after all. Thanks!

brada4, 2025-06-19:
Sorry for the scare, please close this.

closed 2025-06-19 by skynet-gh