openwrt/openwrt #19202: SIMAX1800U : WAN not working

huytran456, 2025-06-22:
### Describe the bug

Hi All
Current i have big problem with SIMAX1800U. I am using SIMAX1800T lasted firmware from https://firmware-selector.openwrt.org/ and then i have installed on SIMAX1800U, Everything is working perfectly but WAN port not working, I have tested 5 SIMAX1800U, they not working, not connect internet

This document tell that SIMAX1800U the same SIMAX1800T: https://github.com/openwrt/openwrt/pull/10696

### OpenWrt version

r28597-0425664679

### OpenWrt release

24.10.1

### OpenWrt target/subtarget

ramips/mt7621

### Device

SIMAX1800U

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

DragonBluep, 2025-06-23:
I didn't see WAN link up event, check your cable first.

DragonBluep, 2025-06-23:
Are their ethernet ports in the same order? You can try inserting the cable into the LAN port.

huytran456, 2025-06-23:
Hi @DragonBluep 
I have checked, SIMAX1800U has the same CPU (MediaTek MT7621AT) and WLAN Hardware(MediaTek MT7905DAN, MediaTek MT7975DN) with SIMAX1800T.
I have tried update boot-env.bin and firmware.bin from SIMAX1800T to SIMAX1800U, but everything no changes, WAN port no working.
But i use TFTP with WAN port, i can flash OPENWRT to memory
Inside images of SIMAX1800U 

[image]

[image]

DragonBluep, 2025-06-24:
It seems that your device has different circuit designs. Unfortunately, the current firmware doesn't support this new revision SIMAX1800U.

huytran456, 2025-06-25:
> It seems that your device has different circuit designs. Unfortunately, the current firmware doesn't support this new revision SIMAX1800U.

Can you add support new revision SIMAX1800U?. Please give me your address and i will send you two units

Gibitwit, 2025-06-27:
Same problem on the Zbtlink ZBT-WE3526 (IQ Router V3), which also has a MT7621.

DragonBluep, 2025-06-28:
> > You can send email to me [yangshiji66@outlook.com](mailto:yangshiji66@outlook.com).
> 
> me, right?

yes

huytran456, 2025-06-28:
> > > You can send email to me [yangshiji66@outlook.com](mailto:yangshiji66@outlook.com).
> > 
> > 
> > me, right?
> 
> yes

Checking your IB, my friend

Gibitwit, 2025-12-23:
WAN is now working on the Zbtlink ZBT-WE3526 (IQ Router V3).  However a cold reboot was requiered after updating the system.  Without a cold reboot the WAN connection did not work.  OpenWRT version 24.10.4.