openwrt/openwrt #19207: dnsmasq `confdir`

G-M0N3Y-2503, 2025-06-22:
### Describe the bug

As of https://github.com/openwrt/openwrt/commit/bd81d97e19e6cc6e33dc5ff852ece95bbc6be01e the default `confdir` is now generated [for the default config of dhcp](https://github.com/openwrt/openwrt/blob/main/package/network/services/dnsmasq/files/dhcp.conf), this directory now has a [autogenerated ID/CFGID](https://openwrt.org/docs/guide-user/base-system/uci) component, which is programmatically inconvenient at best. This results in bugs in packages trying to do the right thing and not override the user's preferences. 

Perhaps a solution would be to set the generated value so it can at least be retrieved, if not a default assumed. 

### OpenWrt version

r28597-0425664679

### OpenWrt release

v24.10.0

### OpenWrt target/subtarget

24.10.1

### Device

QEMU Standard PC (i440FX + PIIX, 1996)

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

G-M0N3Y-2503, 2025-06-22:
Specifically this is causing issues with my package cache-domains-*

dibdot, 2025-06-25:
It's not that hard to determine the correct path, example from adblock:
https://github.com/openwrt/packages/blob/master/net/adblock/files/adblock.sh#L267-L276

G-M0N3Y-2503, 2025-08-09:
> It's not that hard

Difficulty is subjective, but I hope we can at least agree that the solution you've linked is indirect and more involved than the previous one. 

However, what your point misses is that the solution appears to be relying on the assumption that the value of the default confdir won't change or be generated differently. Which are the exact events, in part, that triggered this issue.

Though specifically, I advocate for no assumptions and always setting the default value so that any number of similar changes may occur without proliferating errors in any dependent packages.

dibdot, 2025-08-09:
> the solution appears to be relying on the assumption that the value of the default confdir won't change or be generated differently

Nope, it will take that one, if the user has defined a different confdir.

G-M0N3Y-2503, 2025-08-09:
> Nope, it will take that one, if the user has defined a different confdir.

Perhaps I could have phrased it better; I meant changes to what the default value is. not changes from the default value to a user-defined value.

It was previously assumed that the default `confdir` was `/tmp/dnsmasq.d`, but with the aforementioned change, it has been made clear that that was a bad assumption, hence the breaking of dependent packages. The solution you linked assumes that the default `confdir` is of the form `"/tmp/dnsmasq${instance_name}.d"`, which is the same assumption, just a different value. 

So if we can't predict the default value of `confdir` long-term, as these changes have shown, there should be a way to get the default value from the source of truth. The simplest way seems to be just to always set the `confdir` in this case within `dnsmasq`, that way they can change the default on a daily basis if they want to.