openwrt/openwrt #19223: ksoftirqd/1 at 50%

bioc21, 2025-06-24:
### Describe the bug

I don't know if this is a bug but I noticed in the processes that ksoftirqd/1 was running at 50%. I don't have much running on the router and I checked some settings. The system load was also higher than usual. Once I rebooted the router then it went back to normal. I did not see this on another openwrt router. 

### OpenWrt version

r28597-0425664679

### OpenWrt release

24.10.1

### OpenWrt target/subtarget

mvebu/cortexa9

### Device

Linksys WRT1900ACS

### Image kind

Official downloaded image

### Steps to reproduce

I went into the processes menu. 

### Actual behaviour

I don't know if this is a bug but I noticed in the processes that ksoftirqd/1 was running at 50%

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

DavidBeavan, 2025-08-09:
I had this issue, although ksoftirqd/0 at 50%. A bit of hunting led me to [this post about running one wifi band](https://forum.openwrt.org/t/divested-wrt-no-nonsense-hardened-builds-for-linksys-wrt-series/80522/1577).

I'm using that build, but I was running just 2.4GHz and had 5GHz disabled. After adding the other band, I'm back to nice looking idle.

Might not be the cause of this issue, but worth reporting in case it helps

variia, 2025-11-20:
This is super bad. I am on wrt1900acv2 and I have 5Ghz off all the time. During evening I take down wifi via cron to avoid EMF exposure...

I noticed this morning that ksoftirqd is actually taking one of the CPUs (first day after 24.10.4 upgrade!):
```
root@linksys:~# top -n1 | grep ksoftirqd
   20     2 root     RW       0   0%  50% [ksoftirqd/1]
```

If I enable both radios, take down wifi both CPUs will be pegged by ksoftirqd:
```
root@linksys:~# top -n1 | grep ksoftirqd
   20     2 root     RW       0   0%  50% [ksoftirqd/1]
   14     2 root     RW       0   0%  50% [ksoftirqd/0]
``` 

this returns to normal when I bring back up wifi but... not always otherwise I would not have got this issue this morning. Maybe the overnight +5hrs downtime sort makes this permanent and even if I bring up the wifi in the morning, the issue remains until I reboot.  

I tried various combinations, even having blank, stock default wireless, like I want to run the router without wifi, just cabled device. The issue remains!

The other thing I noticed that hostapd runs all the time, you cannot kill it, netifd (most likely) brings it back up straight away. 

Going back to 23.5.5 insta solves this, sadly.