openwrt/openwrt #19240: HPE MSM 430 missing from list of supported devices

G0rocks, 2025-06-26:
It seems according to [this post](https://forum.openwrt.org/t/hp-msm-support/82788/60) on the forum that you can use the HPE MSM 460 image on the MSM 430 and it works just fine.
But when you look at the [table of supported devices](https://openwrt.org/toh/start) then it's not there

[image]

I'm new to this and am probably about to try flashing the 460 image to my 430 but am a little hesitant. Though if it's fine then I suggest that the table be updated to include the 430 and it can link to the same image.

I know some stuff about computers and routers but am not the most knowledgeable but if there is something that needs to be done still I might be able to do that, just let me know here.

G0rocks, 2025-06-27:
Following [these instructions](https://github.com/blocktrron/msm460-flashing) worked all the way until the part that says "You get the message 'Transfer timed out.' To find out if you have been successfull please check the blinking LED Pattern." because the LED pattern did not appear. That was yesterday and I had to unplug everything and don't remember what the blinking pattern was.

G0rocks, 2025-06-27:
Tried it again, these are the lights that I get
[image]

But my laptop says the ethernet port is unplugged (which it is not) and pinging yields
```
Pinging 192.168.1.1 with 32 bytes of data:
Request timed out.
Request timed out.
Request timed out.
Request timed out.
```
Not sure what to do now.

G0rocks, 2025-07-08:
Thanks, I'll try that, let's see how it goes :)

brada4, 2025-07-08:
There is nothing to be addressed in here, you can close the issue while trying HPE service guide to get back to working factory state.

brada4, 2025-07-24:
Can you submit a PR with new device alias?

jtaczanowski, 2025-07-24:
PR submitted: https://github.com/openwrt/openwrt/pull/19519

Toxminn, 2026-02-17:
> Tried it again, these are the lights that I get
> [image]
> 
> But my laptop says the ethernet port is unplugged (which it is not) and pinging yields
> ```
> Pinging 192.168.1.1 with 32 bytes of data:
> Request timed out.
> Request timed out.
> Request timed out.
> Request timed out.
> ```
> Not sure what to do now.

Hey, I have the same problem.  Power and lan is green but it is not teachable :(