openwrt/openwrt #19261: Linksys MR7350 install failure

Thue, 2025-06-29:
### Describe the bug

Describe the bug

I tried following the instructions at https://openwrt.org/inbox/toh/linksys/mr7350_1.0 - but the router booted up in recovery mode. Poking around in recovery mode, it seems likely that the error is caused by the "kernel" and "alt_kernel" partitions being 8MiB - with most of the flash space being in the "rootfs" and "alt_rootfs" partitions.

It is not immediately obvious to me how to work around this?

[image]

### OpenWrt version

r30244-1831966406

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

qualcommax/ipq60xx

### Device

Linksys MR350

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

Thue, 2025-06-29:
script output attached as smth.txt

brada4, 2025-06-29:
Well, I just asked to rename to .txt to make it github-friendly ;-) But thanks...