openwrt/openwrt #19263: BananaPi R3 mini (filogic): both amber/green leds on both ethernet ports are constantly flashing

Hayqe, 2025-06-29:
### Describe the bug

upgraded two banana pi r3 minis and did a fresh install on the thrid one.
On all devices, the ethenet port leds (both yellow and green) are constantly flashing.

### OpenWrt version

r28739-d9340319c6

### OpenWrt release

24.10.2

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPi-R3 Mini

### Image kind

Official downloaded image

### Steps to reproduce

flash or upgrade to openwrt 24.10.2

### Actual behaviour

Look at ethernet ports: lights are constantly flashing

### Expected behaviour

one led constantly on for connection;
one led blinking on network traffic

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

glassd00r, 2025-06-30:
Reported a while back. But seems to have gone unnoticed.
[[]](https://github.com/openwrt/openwrt/issues/19178)(url)
[https://github.com/openwrt/openwrt/issues/19179](url)

Hayqe, 2025-06-30:
Can confirm that changing the device to br-lan instead of eth0/eth1 does the trick to -at least- stop the weird blinking, so I would assume that this has to do something with netdev and how it handles eth0/eth1 on these devices.

I'm not 100% up to speed on all of this, but it seems that with [this commit](https://lists.infradead.org/pipermail/lede-commits/2024-February/019987.html) from @dangowrt configuration of PHY/MAC interrupt structures have been changed. I can imagine that this could be involved in the current behavior.

```onfig led 'led_lan1'
        option name 'LAN'
        option sysfs 'mdio-bus:0e:green:lan'
        option trigger 'netdev'
#       option mode 'link tx rx'
#       option dev 'eth0'
        option dev 'br-lan'
        option mode 'link'

config led 'led_lan2'
        option name 'LAN'
        option sysfs 'mdio-bus:0e:yellow:lan'
        option trigger 'netdev'
#       option mode 'link tx rx'
#       option dev 'eth0'
        option dev 'br-lan'
        option mode 'tx rx'

config led 'led_wan1'
        option name 'WAN'
        option sysfs 'mdio-bus:0f:green:wan'
        option trigger 'netdev'
#       option mode 'link tx rx'
#       option dev 'eth1'
        option dev 'br-lan'
        option mode 'link'

config led 'led_wan2'
        option name 'WAN'
        option sysfs 'mdio-bus:0f:yellow:wan'
        option trigger 'netdev'
#       option mode 'link tx rx'
#       option dev 'eth1'
        option dev 'br-lan'
        option mode 'tr rx'
```

dangowrt, 2025-07-03:
This commit changes the definition of  what `blink_set` is supposed to do -- all LED drivers implementing `blink_set` assume that calling `blink_set` should set the LED into blinking state. torvalds/linux@c629c972b310af41e9e072febb6dae9a299edde6 however assumes that `blink_set` is used to setup the activity blinking intervals, which contradicts the previous definition. There is no easy fix.

Alexey71, 2025-07-04:
Made a patch with the changes reverted. Tested on kernel 6.12.35. Banana BPI-R4
[9999-revert-configure-led-blink-interval-for-hw-offload.patch](https://github.com/user-attachments/files/21051468/9999-revert-configure-led-blink-interval-for-hw-offload.patch)

nyanmisaka, 2025-07-07:
> Made a patch with the changes reverted. Tested on kernel 6.12.35. Banana BPI-R4 [9999-revert-configure-led-blink-interval-for-hw-offload.patch](https://github.com/user-attachments/files/21051468/9999-revert-configure-led-blink-interval-for-hw-offload.patch) \target\linux\mediatek\patches-6.12\

Considering this is unlikely to get a fix upstream anytime soon, would you like to submit a PR to get it included in OpenWRT?

Alexey71, 2025-07-07:
> Considering this is unlikely to get a fix upstream anytime soon, would you like to submit a PR to get it included in OpenWRT?
https://github.com/openwrt/openwrt/pull/19336

nyanmisaka, 2025-07-07:
> [#19336](https://github.com/openwrt/openwrt/pull/19336)

That was quick! We may also need a branch targeting 24.10.

dangowrt, 2025-07-07:
Did anyone report this upstream and suggest a revert there?

nyanmisaka, 2025-09-26:
@Hayqe I thinks this can be closed in 24.10.3.