openwrt/openwrt #19275: no easy way to disable ssh banner

attila-lendvai, 2025-07-01:
### Describe the bug

ssh banners litter the ssh command output. it just ruined a 9GB file transfer for me...

```
$ ssh root@router.lan "cat /etc/config/dropbear"
  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 24.10.1, r28597-0425664679
 -----------------------------------------------------

config dropbear 'main'
	option enable '1'
	option PasswordAuth 'on'
	option RootPasswordAuth 'off'
	option Port '22'
```

there's no easy way to disable the banner:

```
$ uci set dropbear.main.BannerFile=does-not-exist
$ uci commit dropbear
$ /etc/init.d/dropbear reload
[...]
dropbear.main.BannerFile=does-not-exist validates as file with false
[...]
validation failed
```

ok, let's try an empty value.

err, no: "The UCI (Unified Configuration Interface) in OpenWrt does not allow setting an empty string as an option value."

but it probably wouldn't validate either...

so, all i can do is delete/rename the `/etc/banner`, but what happens at an upgrade? if it gets recreated then the banner is silently restored.

### OpenWrt version

r28597-0425664679

### OpenWrt release

24.10.1

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R4

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-07-01:
Truncate banner file and add its full path to sysupgrade.conf.

brada4, 2025-07-03:
Example of ssh banner usage
https://linux-audit.com/ssh/configure-ssh-welcome-message-or-banner/

brada4, 2025-07-03:
Try this 
```diff
- [ -f /etc/banner ] && cat /etc/banner
-  :</dev/tty && [ -f /etc/banner ] && cat /etc/banner
```

attila-lendvai, 2025-07-03:
this kinda works (kinda, because it prints some error to `stderr`. also, i didn't check the normal login codepath, afraid of locking myself out):

```
$ ssh router.lan "echo foobar" >/tmp/x && echo file contents: && cat /tmp/x
/etc/profile: line 3: /dev/tty: No such device or address
file contents:
foobar
$ 
```

> You cannot get rid of busybox version line though.

not sure what you mean by that. i'm happy just by truncating `/etc/banner` to zero:

```
$ ssh router.lan "echo foobar"
foobar
```

brada4, 2025-07-03:
```bash
ssh -T openwrt.lan /bin/sh
uname -a
Linux sunshine 6.6.94 #0 SMP Mon Jun 30 09:58:32 2025 mips GNU/Linuxexit
```

attila-lendvai, 2025-07-03:
the bug here is not that i couldn't eventually reach my desired outcome (with your kind help)... but the multiple hours i have spent to be able to finally not have the banner in the output, and the need to add `/etc/banner` to `/etc/sysupgrade.conf` not to be bitten by it after a future upgrade.

i consider it a bug to unconditionally print the banner for non-interactive shells, but then i'm no unix authority.