openwrt/openwrt #19299: NAT Loopback not working on IPv6 port forwarding

jjm2473, 2025-07-05:
### Describe the bug

NAT Loopback not working on IPv6 port forwarding

### OpenWrt version

r28739-d9340319c6

### OpenWrt release

24.10.2

### OpenWrt target/subtarget

x86/64

### Device

innotek GmbH VirtualBox

### Image kind

Official downloaded image

### Steps to reproduce

1. Append to `/etc/config/firewall`:
```shell

config redirect
        option dest 'lan'
        option target 'DNAT'
        option name 'testnatloopback6'
        option family 'ipv6'
        option src 'wan'
        option src_dport '8888'
        option dest_ip 'fdfb:e363:ea7e:1::1'
        option dest_port '8888'
```

2. run `/etc/init.d/firewall reload`
3. run `nft -t list ruleset | grep 8888`

### Actual behaviour

```
> nft -t list ruleset | grep 8888
                meta nfproto ipv6 tcp dport 8888 counter packets 0 bytes 0 dnat ip6 to [fdfb:e363:ea7e:1::1]:8888 comment "!fw4: testnatloopback6"
                meta nfproto ipv6 udp dport 8888 counter packets 0 bytes 0 dnat ip6 to [fdfb:e363:ea7e:1::1]:8888 comment "!fw4: testnatloopback6"
```

NAT loopback rules not present.

### Expected behaviour

NAT loopback rules should present.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

jjm2473, 2025-07-15:
> Option reflection_zone is missing.

it's default true. so missing = true.

dongdigua, 2025-07-21:
same issue on OpenWrt 24.10.1 r28597-0425664679

doge-gif, 2026-01-16:
If anyone is curious why you'd want NAT loopback on IPv6, some ISPs only provide you a /64 delegation. (Yes, it sucks.)
In my case, it's NTT in Japan (biggest ISP) with their 1g fiber.
So basically, you cannot do proper firewalling across VLANs with SLAAC.

I had to use ULA+NAT66 for my internal network to separate VLANs, but that falls apart when I'm using public DNS and port forwarding my DMZ.

For now, I'm manually adding dns entries pointing to ULA addresses on openwrt for internal use to circumvent this problem, but it's far from ideal.

jjm2473, 2026-01-17:
@doge-gif  I believe what you're saying is irrelevant to this issue. This issue discusses "NAT loopback" not working, not NAT itself not working.

doge-gif, 2026-01-17:
@jjm2473 
Nope, the nat66 itself is fine. I was just trying to do the same thing as people would do in ipv4, since my isp sucks.
i.e., only have 1 global address, and do all other stuff with port forwarding.

I was using the "Enable NAT loopback" inside "Firewall - Port Forwards" using luci.

jjm2473, 2026-01-18:
@doge-gif OK, I see. NAT66 is useful. Besides the scenario you mentioned where the ISP only provides a 64-bit prefix, it's also useful if we only want to run DDNS on OpenWRT to bind IPv6 addresses.