openwrt/openwrt #19301: 24.10.2 Mercusys MR90X: Official sysupgrade image is unbootable

WerctFourth, 2025-07-05:
### Describe the bug

Hello.
I was setting up OpenWRT 24.10.2 on my Mercusys MR90X. I followed the instructions on the official wiki.
After the final step (flashing the sysupgrade image from the SSH command line), the router rebooted and got stuck in the boot process with the orange LED light on.
I was able to boot an initramfs-recovery image by pressing and holding the Reset button and serving the image from a TFTP server.
Then I tried to flash the same image again, but this time from LuCI web interface. No change.
Booted from the recovery image again and this time I flashed the 24.10.1 sysupgrade image. It booted normally.
I also noticed that the 24.10.2 sysupgrade image is much smaller than the 24.10.1 image (10.3 MB vs 5.41 MB).
I filed this bug report with information from the currently booted version (24.10.1), sorry if it's incorrect.
If you need any additional information, please ask.

### OpenWrt version

r28597-0425664679

### OpenWrt release

24.10.1

### OpenWrt target/subtarget

mediatek/filogic

### Device

MERCUSYS MR90X v1 (UBI)

### Image kind

Official downloaded image

### Steps to reproduce

Follow the steps to install OpenWRT 24.10.2 from the official wiki.

### Actual behaviour

The router is stuck at boot with the orange LED light.
It's possible to reboot into initramfs-recovery by pushing and holding the Reset button while powering the device up.

### Expected behaviour

The image should boot.

### Additional info

Nothing

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

csharper2005, 2025-07-05:
Nokia 24.10.2 image is also broken. https://forum.openwrt.org/t/openwrt-24-10-2-service-release/235046/108

Use https://firmware-selector.openwrt.org/ to build a good sysupgrade image.

WerctFourth, 2025-09-05:
@KneeCapStealer 
AFAIK the difference between UBI and non-UBI firmware is the amount of free space available for OpenWRT system image, configs and software. The UBI version has ~95MB available (from the top of my head, may be slightly incorrect), whereas the non-UBI version has ~55MB.
The exact list of files you should use is written at the beginning of the guide. You should use these files:
- openwrt-mediatek-filogic-mercusys_mr90x-v1-ubi-bl31-uboot.fip
- openwrt-mediatek-filogic-mercusys_mr90x-v1-ubi-initramfs-recovery.itb
- openwrt-mediatek-filogic-mercusys_mr90x-v1-ubi-preloader.bin
- openwrt-mediatek-filogic-mercusys_mr90x-v1-ubi-squashfs-sysupgrade.itb

for UBI firmware installation. Where exactly they are used is written in the guide. So I believe, you should use `openwrt-24.10.2-mediatek-filogic-mercusys_mr90x-v1-ubi-initramfs-recovery.itb` file for initial boot, if that's what your question was about.

WerctFourth, 2025-09-19:
@Ananas0631 
What exactly do you want to know? Regarding the main problem of this topic, I generated my own sysupgrade image (as mentioned [there](https://github.com/openwrt/openwrt/issues/19301#issuecomment-3038697749)) and flashed it instead of the official one. My router runs on OpenWRT 24.10.2 since then. 
The official "non-generated" image is still broken and I don't expect it to be fixed, but it's not a serious issue that blocks using OpenWRT altogether. Just generate your own and flash it instead.

WerctFourth, 2025-09-28:
With the release of 24.10.3 and the fact that its image appears to be valid, this issue can be closed, I suppose.

closed 2025-09-28 by WerctFourth