openwrt/openwrt #19312: Something between 94f043b and 604355e breaks 2,4 GHz Wifi on ath9k device (TP-Link Archer C7 v5)

schuettecarsten, 2025-07-06:
### Describe the bug

Something between 94f043b and 604355e breaks 2,4 GHz Wifi on ath9k device (TP-Link Archer C7 v5). I use ImageBuilder to create the image based and only add some LuCI things to the image, nothing special. In the logs I can see that PHY for radio1 is not found for 604355e while the same config works fine on 94f043b. radio0 and 5 GHz Wifi works fine. Resetting wireless config did not help.

### OpenWrt version

r30266-94f043bf87

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link Archer C7 v5

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

danpawlik, 2025-07-06:
@schuettecarsten you can revert https://github.com/openwrt/openwrt/commit/2ac791e87d3d09f2bb07d99c54f85b34297d8678 and all should be fine.

schuettecarsten, 2025-07-06:
> [@schuettecarsten](https://github.com/schuettecarsten) you can revert [2ac791e](https://github.com/openwrt/openwrt/commit/2ac791e87d3d09f2bb07d99c54f85b34297d8678) and all should be fine.

Yes, but this does not work if you use ImageBuilder to create your images. I switched from custom-compile to ImageBuilder some weeks ago, but it looks like building snapshot myself would be a better idea.

grosjo, 2025-07-06:
Same problem here, latest git breaks ath9k

```
Sun Jul  6 17:23:08 2025 daemon.notice netifd: radio0 (2143): WARNING: Variable 'data' does not exist or is not an array/object                                             
Sun Jul  6 17:23:08 2025 daemon.notice netifd: radio0 (2143): Phy not found                                                                                                 
Sun Jul  6 17:23:08 2025 daemon.notice netifd: radio0 (2143): Could not find PHY for device 'radio0'                                                                        
Sun Jul  6 17:23:08 2025 daemon.notice netifd: Wireless device 'radio0' set retry=0                                                                                         
Sun Jul  6 17:23:08 2025 daemon.crit netifd: Wireless device 'radio0' setup failed, retry=0                                                                                 
Sun Jul  6 17:23:09 2025 daemon.notice netifd: radio0 (2198): WARNING: Variable 'data' does not exist or is not an array/object                                             
Sun Jul  6 17:23:09 2025 daemon.notice netifd: radio0 (2198): Bug: PHY is undefined for device 'radio0'                                                                     
Sun Jul  6 17:23:09 2025 daemon.notice netifd: Wireless device 'radio0' is now down             
```

grosjo, 2025-07-06:
Earlier git worked fine

grosjo, 2025-07-06:
I had to do 
`git reset --hard 39580b1d7bf4062238be26d85bcd33cf55464ee8
`to recover a working device

schuettecarsten, 2025-07-06:
> Same problem here, latest git breaks ath9k

My log is the same.

danpawlik, 2025-07-06:
It is enough to 
```
git revert 2ac791e87d3d09f2bb07d99c54f85b34297d8678
```

schuettecarsten, 2025-07-06:
> > It is enough to
> > ```
> > git revert 2ac791e87d3d09f2bb07d99c54f85b34297d8678
> > ```
> 
> No, it does not. Only the reset to the previous resolves the situation

Still testing and investigation, seems that you need to revert https://github.com/openwrt/openwrt/commit/2ac791e87d3d09f2bb07d99c54f85b34297d8678 and https://github.com/openwrt/openwrt/commit/88f4c32060f79e5161a789e85da299cebbde68bf

nbd168, 2025-07-06:
ath9k regression should be fixed in d2b2670937f051bda7c8fbda8a6c0131977d5e68, please test

danpawlik, 2025-07-06:
works well @nbd168 thank you!

oldium, 2025-07-06:
Perfect timing, guys. Thanks! Will test C7 v4/v5 after local rebuild :-)

oldium, 2025-07-06:
Works fine. Thanks, Felix!

closed 2025-07-07 by nbd168

schuettecarsten, 2025-07-09:
No, it does not work on my devices:

[image]

nbd168, 2025-07-09:
@schuettecarsten it seems that somebody broke it again for you, right after I fixed it. Please test 118d8ee98132f4519cc208cbbaae6d1f8cb7d295, it should work now.

schuettecarsten, 2025-07-09:
@nbd168 Thank you, will test as soon as the ImageBuilder update is available.

schuettecarsten, 2025-07-10:
> @schuettecarsten it seems that somebody broke it again for you, right after I fixed it. Please test 118d8ee98132f4519cc208cbbaae6d1f8cb7d295, it should work now.

@nbd168 still no luck.

[image]

schuettecarsten, 2025-07-12:
@nbd168 works, thank you