openwrt/openwrt #19314: Compilation error : bash: line 1: *1024: arithmetic syntax error: operand expected (error token is "*1024")

grosjo, 2025-07-06:
### Describe the bug

With latest git, I see
```
make -f ./scripts/Makefile.build obj=scripts scripts/module.lds
make -f ./scripts/Makefile.modfinal
./scripts/misc-check
make[6]: Nothing to be done for 'modules'.
make[5]: Leaving directory '/data/openwrt/wr650/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.12.35'
# If .config did not change, use the previous timestamp to avoid package rebuilds
cmp -s /data/openwrt/wr650/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.12.35/.config /data/openwrt/wr650/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.12.35/.config.modules.save && mv /data/openwrt/wr650/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.12.35/.config.modules.save /data/openwrt/wr650/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.12.35/.config; cp -fpR /data/openwrt/wr650/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.12.35/.config /data/openwrt/wr650/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.12.35/.config.modules.save
touch /data/openwrt/wr650/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-6.12.35/.modules
make -C image compile TARGET_BUILD=
make[5]: Entering directory '/data/openwrt/wr650/target/linux/ath79/image'
bash: line 1: *1024: arithmetic syntax error: operand expected (error token is "*1024")
make[5]: Nothing to be done for 'compile'.
make[5]: Leaving directory '/data/openwrt/wr650/target/linux/ath79/image'
make[4]: Leaving directory '/data/openwrt/wr650/target/linux/ath79'
make[3]: Leaving directory '/data/openwrt/wr650/target/linux'

```

### OpenWrt version

r30239-ba2ae60a00

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ath79/generic

### Device

Comfast CF-WR650AC v2

### Image kind

Self-built image

### Steps to reproduce

Rebuild your kernel

### Actual behaviour

Display the error above

### Expected behaviour

There shall not be an error

### Additional info

Error does not stop the compilation, but obviously something is wrong

### Diffconfig

```text
CONFIG_TARGET_ath79=y
CONFIG_TARGET_ath79_generic=y
CONFIG_TARGET_ath79_generic_DEVICE_comfast_cf-wr650ac-v2=y
CONFIG_ATH10K_LEDS=y
CONFIG_ATH10K_THERMAL=y
# CONFIG_ATH9K_HWRNG is not set
# CONFIG_ATH9K_UBNTHSR is not set
# CONFIG_ATH_USER_REGD is not set
CONFIG_CLEAN_IPKG=y
[... 204 lines cut ...]
CONFIG_PROCD_SHOW_BOOT=y
# CONFIG_SIGNATURE_CHECK is not set
# CONFIG_SIGNED_PACKAGES is not set
# CONFIG_TARGET_ROOTFS_INITRAMFS is not set
CONFIG_TESTING_KERNEL=y
# CONFIG_USE_APK is not set
# CONFIG_USE_FS_ACL_ATTR is not set
CONFIG_USE_MKLIBS=y
# CONFIG_USE_SECCOMP is not set
# CONFIG_WPA_MBO_SUPPORT is not set
# CONFIG_OPENSSL_WITH_ERROR_MESSAGES is not set
# CONFIG_PACKAGE_dnsmasq_full_conntrack is not set
# CONFIG_PACKAGE_dnsmasq_full_dhcpv6 is not set
# CONFIG_PACKAGE_dnsmasq_full_nftset is not set
# CONFIG_PACKAGE_dnsmasq_full_noid is not set
# CONFIG_PACKAGE_dnsmasq_full_tftp is not set
# CONFIG_PACKAGE_kmod-nf-conntrack-netlink is not set
# CONFIG_PACKAGE_libnetfilter-conntrack is not set
# CONFIG_PACKAGE_libnfnetlink is not set
# CONFIG_PACKAGE_opkg is not set
```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2025-07-06:
Check https://github.com/openwrt/openwrt/commit/5adc663af6c9669493d4fe9bf8169d6debe7aab8

namiltd, 2025-07-06:
But the error was due to the analysis of its parameters even if it is not enabled.

closed 2025-07-06 by grosjo