openwrt/openwrt #19316: openwrt 24.10.2: opkg update Signature check failed for luci

jilroy, 2025-07-06:
### Describe the bug

Signature check fails for https://downloads.openwrt.org/releases/24.10.2/packages/aarch64_cortex-a53/luci/Packages.sig when doing an opkg update. This prevents package installs with opkg.

### OpenWrt version

stable

### OpenWrt release

24.10.2

### OpenWrt target/subtarget

mediatek/filogic

### Device

GL-MT6000

### Image kind

Official downloaded image

### Steps to reproduce

(1) opkg update is working correctly on openwrt 24.10.1: all signatures pass
(2) did a sysupgrade from 24.10.1 to 24.10.2 with image: https://downloads.openwrt.org/releases/24.10.2/targets/mediatek/filogic/openwrt-24.10.2-mediatek-filogic-glinet_gl-mt6000-squashfs-sysupgrade.bin
(3) opkg update returns a "Signature check failed." for htps://downloads.openwrt.org/releases/24.10.2/packages/aarch64_cortex-a53/luci/Packages.sig
(4) Note: other signatures pass, only the one mentionned above fails

### Actual behaviour

signature fails

### Expected behaviour

signature should pass

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

Lyude, 2025-09-13:
Also @jilroy you should edit your first comment to include the numeric version of openwrt under the "OpenWrt Release" header instead of `stable` (e.g. 24.10.2). The question you got before was from the Github workflow bot, and the bug is marked with the invalid issue label since the bot relies on a numeric version in the comment template

jilroy, 2025-09-13:
@Lyude Right.

xPXpanD, 2025-09-22:
I'm also seeing this, fresh install of 24.10.2 on an ASUS TUF-AX4200, build r28739-d9340319c6. I don't recall seeing this a few hours ago, but it's my first time on OpenWRT and I was doing a million things at once; chances are I just missed it.

This is probably unrelated, but just in case... luci-compat installed fine from LuCI's UI, but I ran into issues when grabbing p910nd:

Collected errors:
 * opkg_install_pkg: Checksum or size mismatch for package p910nd. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package p910nd.

This is when I first noticed the issue mentioned here.

At some point that package's flag also changed to "Not available" in the UI, which I saw when I tried to install luci-app-p910nd later. (which pulls p910nd in as a dependency)

The actual file downloaded fine when I visited the URL in a browser.

hauke, 2025-10-19:
Just try it again 1 hour later. 

The package feeds are build as a rolling release, when they get updated they are sometimes in an inconsistent state.