openwrt/openwrt #19333: Netgear R8000 - no more wifi in 24.10.2

hgomez, 2025-07-07:
I lost Wifi after install OpenWRT 24.10.2 on my Netgear R8000

I noticed this in dmesg 

`root@R8000:~# dmesg | grep brcmfmac
[   14.990213] usbcore: registered new interface driver brcmfmac
[   15.002011] brcmfmac 0000:01:00.0: enabling device (0140 -> 0142)
[   15.121055] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43602-pcie for chip BCM43602/1
[   15.147310] brcmfmac 0001:03:00.0: enabling device (0140 -> 0142)
[   15.154656] brcmfmac 0000:01:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.netgear,r8000.bin failed with error -2
[   15.271168] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43602-pcie for chip BCM43602/1
[   15.285939] brcmfmac 0001:04:00.0: enabling device (0140 -> 0142)
[   15.292735] brcmfmac 0001:03:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.netgear,r8000.bin failed with error -2
[   15.376217] brcmfmac 0000:01:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.txt failed with error -2
[   15.386288] brcmfmac 0001:03:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.txt failed with error -2
[   15.398987] brcmfmac 0001:03:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.clm_blob failed with error -2
[   15.409450] brcmfmac 0001:03:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.txcap_blob failed with error -2
[   15.441700] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43602-pcie for chip BCM43602/1
[   15.451537] brcmfmac 0001:04:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.netgear,r8000.bin failed with error -2
[   15.453947] brcmfmac 0000:01:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.clm_blob failed with error -2
[   15.473141] brcmfmac 0000:01:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.txcap_blob failed with error -2
[   15.512520] brcmfmac 0001:04:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.txt failed with error -2
[   15.553591] brcmfmac 0001:04:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.clm_blob failed with error -2
[   15.564026] brcmfmac 0001:04:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.txcap_blob failed with error -2
[   15.729242] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[   15.740382] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[   15.760976] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43602/1 wl0: Sep 18 2015 03:30:01 version 7.35.177.56 (r587209) FWID 01-6cb8e269
[   15.774116] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[   15.785229] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[   15.792946] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[   15.804036] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[   15.810870] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43602/1 wl0: Sep 18 2015 03:30:01 version 7.35.177.56 (r587209) FWID 01-6cb8e269
[   15.845202] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43602/1 wl0: Sep 18 2015 03:30:01 version 7.35.177.56 (r587209) FWID 01-6cb8e269
[  255.838555] brcmfmac 0001:03:00.0 phy1-ap0: renamed from wlan0
[  255.948269] brcmfmac 0001:03:00.0 phy1-ap0: entered allmulticast mode
[  255.955052] brcmfmac 0001:03:00.0 phy1-ap0: entered promiscuous mode
[  255.964755] brcmfmac 0001:03:00.0 phy1-ap0: left allmulticast mode
`

24.10.2 (24.10) is supposed to works on NetGear R8000 ?

Thanks for feedbacks

hgomez, 2025-07-08:
I reinstalled with 23.05.5 and I can't get signals on AC channels

[image]

hgomez, 2025-07-08:
Good news, on 23.05 I switched to US following [5G tips in documentation](https://openwrt.org/toh/netgear/r8000?s[]=r8000)

 _for 5 GHz work great load on radio2

    set country code to US
    set channel 48 to radio2
    set channel 149 to radio0→ SAVE → Reboot_

```
root@R8000:~# iw reg get
global
country US: DFS-FCC
	(902 - 904 @ 2), (N/A, 30), (N/A)
	(904 - 920 @ 16), (N/A, 30), (N/A)
	(920 - 928 @ 8), (N/A, 30), (N/A)
	(2400 - 2472 @ 40), (N/A, 30), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
	(5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
	(5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
	(5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
	(5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
	(5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
	(57240 - 71000 @ 2160), (N/A, 40), (N/A)

phy#2
country 99: DFS-UNSET
	(2402 - 2482 @ 40), (6, 20), (N/A)
	(2474 - 2494 @ 20), (6, 20), (N/A)
	(5140 - 5360 @ 160), (6, 20), (N/A)
	(5460 - 5860 @ 160), (6, 20), (N/A)

phy#0
country 99: DFS-UNSET
	(2402 - 2482 @ 40), (6, 20), (N/A)
	(2474 - 2494 @ 20), (6, 20), (N/A)
	(5140 - 5360 @ 160), (6, 20), (N/A)
	(5460 - 5860 @ 160), (6, 20), (N/A)

phy#1
country 99: DFS-UNSET
	(2402 - 2482 @ 40), (6, 20), (N/A)
	(2474 - 2494 @ 20), (6, 20), (N/A)
	(5140 - 5360 @ 160), (6, 20), (N/A)
	(5460 - 5860 @ 160), (6, 20), (N/A)
```

[image]

Both 5G channels are now up

hgomez, 2025-07-08:
It also works great on 24.10.2, I will close this issue.

R8000 users, be sure to set AC radio to US and channel 149/48

- set country code to US (radio0/radio2)
- set channel 48 to radio2
- set channel 149 to radio0→ SAVE → Reboot_

Abdoymani, 2025-10-07:
> I lost Wifi after install OpenWRT 24.10.2 on my Netgear R8000
> 
> I noticed this in dmesg
> 
> ```
> root@R8000:~# dmesg | grep brcmfmac
> [   14.990213] usbcore: registered new interface driver brcmfmac
> [   15.002011] brcmfmac 0000:01:00.0: enabling device (0140 -> 0142)
> [   15.121055] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43602-pcie for chip BCM43602/1
> [   15.147310] brcmfmac 0001:03:00.0: enabling device (0140 -> 0142)
> [   15.154656] brcmfmac 0000:01:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.netgear,r8000.bin failed with error -2
> [   15.271168] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43602-pcie for chip BCM43602/1
> [   15.285939] brcmfmac 0001:04:00.0: enabling device (0140 -> 0142)
> [   15.292735] brcmfmac 0001:03:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.netgear,r8000.bin failed with error -2
> [   15.376217] brcmfmac 0000:01:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.txt failed with error -2
> [   15.386288] brcmfmac 0001:03:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.txt failed with error -2
> [   15.398987] brcmfmac 0001:03:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.clm_blob failed with error -2
> [   15.409450] brcmfmac 0001:03:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.txcap_blob failed with error -2
> [   15.441700] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43602-pcie for chip BCM43602/1
> [   15.451537] brcmfmac 0001:04:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.netgear,r8000.bin failed with error -2
> [   15.453947] brcmfmac 0000:01:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.clm_blob failed with error -2
> [   15.473141] brcmfmac 0000:01:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.txcap_blob failed with error -2
> [   15.512520] brcmfmac 0001:04:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.txt failed with error -2
> [   15.553591] brcmfmac 0001:04:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.clm_blob failed with error -2
> [   15.564026] brcmfmac 0001:04:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.txcap_blob failed with error -2
> [   15.729242] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
> [   15.740382] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
> [   15.760976] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43602/1 wl0: Sep 18 2015 03:30:01 version 7.35.177.56 (r587209) FWID 01-6cb8e269
> [   15.774116] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
> [   15.785229] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
> [   15.792946] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
> [   15.804036] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
> [   15.810870] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43602/1 wl0: Sep 18 2015 03:30:01 version 7.35.177.56 (r587209) FWID 01-6cb8e269
> [   15.845202] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43602/1 wl0: Sep 18 2015 03:30:01 version 7.35.177.56 (r587209) FWID 01-6cb8e269
> [  255.838555] brcmfmac 0001:03:00.0 phy1-ap0: renamed from wlan0
> [  255.948269] brcmfmac 0001:03:00.0 phy1-ap0: entered allmulticast mode
> [  255.955052] brcmfmac 0001:03:00.0 phy1-ap0: entered promiscuous mode
> [  255.964755] brcmfmac 0001:03:00.0 phy1-ap0: left allmulticast mode
> ```
> 
> 24.10.2 (24.10) is supposed to works on NetGear R8000 ?
> 
> Chipset is BCM43602
> 
> Thanks for feedbacks

do you used the following steps to transfer the files:

On a local machine, download the files:
wget https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/brcm/brcmfmac43602-pcie.clm_blob
wget https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/brcm/brcmfmac43602-pcie.txcap_blob

Then transfer them to the router via SCP or another method.

On the router, copy the files to /lib/firmware/brcm/.

Then, reload the brcmfmac driver or reboot.

Let's check if the router has wget:

which wget or which curl

If not, install one of them:

opkg update && opkg install wget

Then, on the router:

cd /lib/firmware/brcm/
wget https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/brcm/brcmfmac43602-pcie.clm_blob
wget https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/brcm/brcmfmac43602-pcie.txcap_blob

Then, we can remove and reinsert the brcmfmac module or reboot.

However, note that the firmware file brcmfmac43602-pcie.bin might also be outdated. We can update it from the same repository:

wget https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/brcm/brcmfmac43602-pcie.bin -O [brcmfmac43602-pcie.bin.new](https://brcmfmac43602-pcie.bin.new/)

Then, we can replace the old one and reload the driver.

But note: the firmware version in the Linux firmware repository might be newer than the one we have (2015). We can check the date of the file in the repository.

Let's check the version of the firmware in the Linux firmware repository:

The file in the repository is updated from time to time. We can use the latest one.