openwrt/openwrt #19349: uhttpd keeps crashing when navigating menu options in LUCI

vochong, 2025-07-08:
### Describe the bug

I have experienced frequent uhttpd crashes with LuCI navigation over the past month when using snapshot-based images.  The problem seems most noticeable when quickly navigating through different LuCI menu options in the left-hand frame. Users are unexpectedly logged out and must sign in again, though the issue may reoccur randomly. In some cases, the interface becomes completely unresponsive, and logs show that uhttpd has crashed.

```
Tue Jul  8 07:37:58 2025 daemon.info procd: Instance uhttpd::instance1 s in a crash loop 8 crashes, 298 seconds since last crash

```

### OpenWrt version

r30302-bce0ad5563

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Linksys MX4300

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

paulcarroty, 2025-08-03:
Can't reproduce it on r30660-a2c361c520.

Sinbios, 2025-09-01:
Same issue here on MX4200:

Architecture | ARMv8 Processor rev 4
Target Platform | qualcommax/ipq807x
Firmware Version | OpenWrt SNAPSHOT r0-cc2a47e / LuCI Master 25.211.81098~b7c0847
Kernel Version | 6.12.40

```
Sun Aug 31 23:42:10 2025 daemon.notice procd: /etc/rc.d/S50uhttpd: Skipping invalid Lua prefix "/cgi-bin/luci=/usr/lib/lua/luci/sgi/uhttpd.lua"
Mon Sep  1 00:42:17 2025 daemon.info procd: Instance uhttpd::instance1 s in a crash loop 6 crashes, 1 seconds since last crash
Mon Sep  1 00:45:05 2025 daemon.info procd: Instance uhttpd::instance1 s in a crash loop 6 crashes, 1 seconds since last crash
```

Benqer0, 2025-12-28:
I have the same on my
Banana Pi BPI-R4

[28.12.2025, 12:53:54 MEZ] daemon.info: procd: Instance uhttpd::instance1 s in a crash loop 6 crashes, 294 seconds since last crash

uhttpd | 2025.10.03~ebb92e6b-r1

Any ideas?

robinZhao, 2026-01-22:
openwrt-25.12-rc3  rockchip
Thu Jan 22 08:29:02 2026 daemon.info procd: Instance
uhttpd::instance1 s in a crash loop 6 crashes, 160 se
conds since last crash
Thu Jan 22 08:29:32 2026