openwrt/openwrt #19355: HP MSM460: Wifi stops working after a few days

citronalco, 2025-07-09:
### Describe the bug

I'm  running a HPW MSM460 as access point and 802.11s mesh router.
After 1-3 days, wifi stops working completely.

iwinfo shows "Signal: unknown" for all wifi interfaces. There are no entries in dmesg.

After restarting wifi, everything starts working again. But after a 1-3 days,....

### OpenWrt version

r24190+11-1417fdfc90

### OpenWrt release

23.05-SNAPSHOT

### OpenWrt target/subtarget

mpc85xx/p1020

### Device

Hewlett-Packard MSM460

### Image kind

Self-built image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

Djfe, 2025-07-20:
I was looking into backporting this patch at some point which aims to finally fix this
https://patchwork.kernel.org/project/linux-wireless/patch/20241106-ath9k-deaf-detection-v1-1-736a150d2425@redhat.com/
This is pretty much a longstanding issue. It's some kind of hw bug apparently and now there is finally workaround/solid fix for it.
nbd already tried fixing this years ago but had to revert it again because it didn't catch the issue like expected
https://github.com/openwrt/openwrt/commit/33a840a3ff4cc15b45cf3fb47c772b5a44ecfe5c

citronalco, 2025-07-31:
Applied the patch (https://patchwork.kernel.org/project/linux-wireless/patch/20241106-ath9k-deaf-detection-v1-1-736a150d2425@redhat.com/) to 5.15 (only a few offset warnings). Kernel compiles, wifi works (at least since i installed the AP an hour ago).

I'm using Gluon: https://github.com/citronalco/gluon/blob/v2023.2.x_ath9k_deaf/patches/openwrt/0012-ath9k-add-rx-inactivity-detection-and-reset-chip-when-it-occurs.patch

Djfe, 2025-08-02:
That doesn't sound promising :/
Does "RX inactivity detected. Schedule chip reset" show up in your kernel log (dmesg)? (maybe ath9k debugging needs to be activated first somehow, I'm not sure)

You could test increasing/removing the max interval (if it takes more than 8 seconds it currently skips the check because the method was not called often enough):
https://github.com/citronalco/gluon/blob/v2023.2.x_ath9k_deaf/patches/openwrt/0012-ath9k-add-rx-inactivity-detection-and-reset-chip-when-it-occurs.patch#L69

You could also in a second step if removing the first conditional is not enough check whether it helps to increase how large "count" is required to be
`count >= 2 * interval / MSEC_PER_SEC)`
count must be larger than 8 in 4 seconds then, instead of 4 in 4 seconds
This means the patch would expect 1 interrupt per 500ms on average

citronalco, 2025-08-02:
Not a single "RX inactivity detected. Schedule chip reset" in dmesg.

You're probably right, looks like one has to enable debug output to see that message.
The message gets sent as "RESET" debug message, so according to [ath.h](https://github.com/torvalds/linux/blob/master/drivers/net/wireless/ath/ath.h), `echo 0x00000401 > /sys/kernel/debug/ieee80211/phy0/ath9k/debug` should be sufficient (default value is 0x00000400).
 I've never done anything with debugfs so far, so I could be wrong. Will try with that and report.

citronalco, 2025-08-03:
Can anyone tell me which interrupt is monitored?
Is it "TX" in /sys/kernel/debug/ieee80211/phy0/ath9k/interrupt ?

citronalco, 2025-08-04:
> Though: your issue is also fixed by running `wifi`

Yes, but I would not call that a fix.

Djfe, 2025-08-04:
yeah, I just meant that it's similar in that way. I knew of another wifi bugs that was not fixed by running `wifi` and required the kernel module to be reloaded (this was with mt76 wifi6 and has been fixed by now)

jtaczanowski, 2025-09-07:
Did you try to use latest OpenWrt 24.10.2? I'm using two HP MSM 460 access points. One is working as router with AP on phy1, and 802.11s on phy0 connected to with second HP MSM 460 (phy0) acting as repeater. I'm using 802.11s to create simple bridge over this two access points. It is working for me since week without any outage so far.

Maybe it matters that:
- I have little traffic on this network
- I'm not using encryption on this 802.11s link

citronalco, 2025-09-09:
>     * I have little traffic on this network

Have you experienced this issue before, e.g. with 23.05?

Can you try with medium traffic? It seems to happen more often the more traffic is on the link.
I'm having ~5 Mbit/s on average on the 802.11s link, and it happens 1-3 times every day.