openwrt/openwrt #19357: samba4: build failure with musl

graysky2, 2025-07-09:
### Describe the bug

Since [PR#19160](https://github.com/openwrt/openwrt/pull/19160) was merged, the samba4 package can only be build with glibc. It will not appear in a menuconfig when building with musl.

### OpenWrt version

N/A building from HEAD

### OpenWrt release

N/A building from HEAD

### OpenWrt target/subtarget

N/A building from HEAD

### Device

N/A building from HEAD

### Image kind

Official downloaded image

### Steps to reproduce

Try to set any component of samba4 with musl (default)

### Actual behaviour

The samba4 package is not available

### Additional info

The samba4 package should be available

### Diffconfig

```text
N/A
```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

rockdrilla, 2025-07-09:
Did you pulled recent commits for `base` and `packages` feeds? AFAIK `libcrypt-compat` doesn't have to be built with `musl` (only with `glibc`). I've built `musl`-based minimal build (mostly for compile tests) a day or two ago, and `dropbear` was built fine enough (which depends on `libcrypt.so.1`). Anyway, I/we need more info.

graysky2, 2025-07-09:
Yes, everything is updated. Can you try to even select samba4-server with a musl config? For me, it is not present unless I select glibc.

graysky2, 2025-07-09:
Actually, you can this reflected in the CI to a samba4 PR I created: https://github.com/openwrt/packages/actions/runs/16177725830/job/45666782320?pr=26947 ... look under annotations:

[image]

rockdrilla, 2025-07-15:
@graysky2 is problem still exists?

graysky2, 2025-07-15:
No, builds fine from latest HEAD