openwrt/openwrt #19359: "Drop invalid packets" actually drops many packets

grosjo, 2025-07-10:
### Describe the bug

When choosing that option in Luci/Firewall, the device let ping go through and some direct packet, but rerouted packets on port 80 or 443 are all blocked

### OpenWrt version

Git

### OpenWrt release

Git

### OpenWrt target/subtarget

ath79

### Device

Comfast wr 650

### Image kind

Official downloaded image

### Steps to reproduce

Click on "Drop invalid packet"

### Actual behaviour

Drop packets on port 80/443/22 but not ping requests 

### Expected behaviour

Just drop invalid packets....

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-07-10:
I dont know what "rerouted" means - "invalid" is classified between raw and mangle and persists until packets exit conntrack engine at postrouting,+32k.

grosjo, 2025-07-11:
rerouted are packets that come froma device with default gateway this device, and this device route the packet to the actual gateway to the internet

brada4, 2025-07-11:
OK, these are forwarded packets or more accurately masqueraded conntrack flows.
Can you enable sysctl and see on what topic the invalid markings are?
eg
```
[00] nf_ct_proto_6: 3312 bytes more than expected IN=wan OUT= MAC= SRC=149.112.112.112 DST= LEN=467 TOS=0x00 PREC=0x20 TTL=60 ID=27037 DF PROTO=TCP SPT=853 DPT=63223 SEQ=2599691418 ACK=339893725 WINDOW=22 RES=0x00 ACK PSH URGP=0 OPT () 
[00] nf_ct_proto_6: invalid tcp flag combination IN=wan OUT= MAC=  SRC=1.34.85.243 DST= LEN=60 TOS=0x00 PREC=0x20 TTL=53 ID=4441 DF PROTO=TCP SPT=51926 DPT=46228 SEQ=911901397 ACK=0 WINDOW=0 RES=0x00 URG ACK PSH RST SYN FIN URGP=17989 OPT () 

```

brada4, 2025-07-11:
NB sysctl -based logging works indifferently of firewall configuration with regard to invalid packet handling.

grosjo, 2025-07-12:
no, not masquerade

jsut routed to the actual gateway

brada4, 2025-07-12:
What is the logged reason packets are considered invalid?
Does .20.1 router know how to route.100.0/24 network?

grosjo, 2025-07-12:
In my example, all devices are on 192.168.0.0/16

Can you state the sysctl command you want me to execute ?

grosjo, 2025-07-12:
SOme more if of any use:

```
Jul 12 19:27:03 v1-main kernel: [11826.421188] nf_ct_proto_6: invalid new IN=LanPort OUT= MAC=40:a5:ef:5d:42:e0:fc:3c:d7:93:9d:ab:08:00 SRC=192.168.2.80 DST=52.35.54.49 LEN=40 TOS=0x00 PREC=0x00 TTL=255 ID=46746 PROTO=TCP SPT=53854 DPT=443 SEQ=2326645988 ACK=390563759 WINDOW=4380 RES=0x00 ACK RST URGP=0
Jul 12 11:27:04 v1-main kernel: nf_ct_proto_6: packet (index 3) in dir 0 invalid, state SYN_SENT IN=LanPort OUT= MAC=40:a5:ef:5d:42:e0:fc:3c:d7:93:9d:ab:08:00 SRC=192.168.2.80 DST=52.35.54.49 LEN=40 TOS=0x00 PREC=0x00 TTL=255 ID=46748 PROTO=TCP SPT=53866 DPT=443 SEQ=2335775401 ACK=869403456 WINDOW=4380 RES=0x00 ACK URGP=0 
Jul 12 11:27:04 v1-main kernel: nf_ct_proto_6: packet (index 3) in dir 0 invalid, state SYN_SENT IN=LanPort OUT= MAC=40:a5:ef:5d:42:e0:fc:3c:d7:93:9d:ab:08:00 SRC=192.168.2.80 DST=52.35.54.49 LEN=119 TOS=0x00 PREC=0x00 TTL=255 ID=46749 PROTO=TCP SPT=53866 DPT=443 SEQ=2335775401 ACK=869403456 WINDOW=4380 RES=0x00 ACK PSH URGP=0 
Jul 12 19:27:04 v1-main kernel: [11827.056839] nf_ct_proto_6: packet (index 3) in dir 0 invalid, state SYN_SENT IN=LanPort OUT= MAC=40:a5:ef:5d:42:e0:fc:3c:d7:10:d4:4f:08:00 SRC=192.168.2.84 DST=34.218.107.222 LEN=119 TOS=0x00 PREC=0x00 TTL=255 ID=4033 PROTO=TCP SPT=62683 DPT=443 SEQ=1715184 ACK=1770182451 WINDOW=4380 RES=0x00 ACK PSH URGP=0
Jul 12 19:27:04 v1-main kernel: [11827.084804] nf_ct_proto_6: packet (index 2) in dir 0 invalid, state SYN_SENT IN=LanPort OUT= MAC=40:a5:ef:5d:42:e0:fc:3c:d7:10:d4:4f:08:00 SRC=192.168.2.84 DST=34.218.107.222 LEN=40 TOS=0x00 PREC=0x00 TTL=255 ID=4034 PROTO=TCP SPT=62683 DPT=443 SEQ=1715263 ACK=1770182451 WINDOW=4380 RES=0x00 ACK FIN URGP=0
Jul 12 11:27:05 v1-main kernel: nf_ct_proto_6: packet (index 3) in dir 0 invalid, state SYN_SENT IN=LanPort OUT= MAC=40:a5:ef:5d:42:e0:fc:3c:d7:93:9d:ab:08:00 SRC=192.168.2.80 DST=52.35.54.49 LEN=40 TOS=0x00 PREC=0x00 TTL=255 ID=46750 PROTO=TCP SPT=53866 DPT=443 SEQ=2335775480 ACK=869403456 WINDOW=4380 RES=0x00 ACK URGP=0 
Jul 12 19:27:05 v1-main kernel: [11827.862561] nf_ct_proto_6: packet (index 3) in dir 0 invalid, state SYN_SENT IN=LanPort OUT= MAC=40:a5:ef:5d:42:e0:fc:3c:d7:93:9d:ab:08:00 SRC=192.168.2.80 DST=34.218.107.222 LEN=119 TOS=0x00 PREC=0x00 TTL=255 ID=46751 PROTO=TCP SPT=53858 DPT=443 SEQ=2329688710 ACK=3665011144 WINDOW=4380 RES=0x00 ACK PSH URGP=0
Jul 12 19:27:05 v1-main kernel: [11827.890749] nf_ct_proto_6: packet (index 2) in dir 0 invalid, state SYN_SENT IN=LanPort OUT= MAC=40:a5:ef:5d:42:e0:fc:3c:d7:93:9d:ab:08:00 SRC=192.168.2.80 DST=34.218.107.222 LEN=40 TOS=0x00 PREC=0x00 TTL=255 ID=46752 PROTO=TCP SPT=53858 DPT=443 SEQ=2329688789 ACK=3665011144 WINDOW=4380 RES=0x00 ACK FIN URGP=0
Jul 12 11:27:05 v1-main kernel: nf_ct_proto_6: packet (index 3) in dir 0 invalid, state SYN_SENT IN=LanPort OUT= MAC=40:a5:ef:5d:42:e0:fc:3c:d7:93:9d:ab:08:00 SRC=192.168.2.80 DST=52.35.54.49 LEN=119 TOS=0x00 PREC=0x00 TTL=255 ID=46753 PROTO=TCP SPT=53866 DPT=443 SEQ=2335775401 ACK=869403456 WINDOW=4380 RES=0x00 ACK PSH URGP=0 
Jul 12 11:27:07 v1-main kernel: nf_ct_proto_6: packet (index 3) in dir 0 invalid, state SYN_SENT IN=LanPort OUT= MAC=40:a5:ef:5d:42:e0:fc:3c:d7:93:9d:ab:08:00 SRC=192.168.2.80 DST=52.35.54.49 LEN=40 TOS=0x00 PREC=0x00 TTL=255 ID=46754 PROTO=TCP SPT=53866 DPT=443 SEQ=2335775480 ACK=869403456 WINDOW=4380 RES=0x00 ACK URGP=0 

```

brada4, 2025-07-12:
Looks like route mismatch - synack never gets back and client stops trying.

grosjo, 2025-07-12:
maybe, but removing teh filter gets back everything to work

brada4, 2025-07-12:
Ok, disable drop invalid for now.

Please add nftables tracing (change filter to at least 443 tcp) , in sysctl-d log it looks weird that packets to internet enter as if they are directed to router like not intended to be forwarded.🦧

https://github.com/openwrt/firewall4/blob/6b72f7149de2bda65486eb51602992e321cbd832/root/etc/nftables.d/10-custom-filter-chains.nft#L39

grosjo, 2025-07-12:
Not sure I am doing it righm but I took the example, change 53 with 443 and typed "# nft monitor trace"  -> nothing goes out

Should I do that while enable the firewall filter ?

grosjo, 2025-07-12:
ah yes, I have now :

```
trace id 9c07fc2b inet fw4 prerouting policy accept 
trace id 9c07fc2b inet fw4 mangle_forward packet: iif "LanPort" oif "LanPort" ether saddr fc:3c:d7:10:d4:4f ether daddr 40:a5:ef:5d:42:e0 ip saddr 192.168.2.84 ip daddr 34.218.107.222 ip dscp cs0 ip ecn not-ect ip ttl 254 ip id 6687 ip protocol tcp ip length 40 tcp sport 62702 tcp dport 443 tcp flags == ack tcp window 4380 
trace id 9c07fc2b inet fw4 mangle_forward policy accept 
trace id 9c07fc2b inet fw4 forward packet: iif "LanPort" oif "LanPort" ether saddr fc:3c:d7:10:d4:4f ether daddr 40:a5:ef:5d:42:e0 ip saddr 192.168.2.84 ip daddr 34.218.107.222 ip dscp cs0 ip ecn not-ect ip ttl 254 ip id 6687 ip protocol tcp ip length 40 tcp sport 62702 tcp dport 443 tcp flags == ack tcp window 4380 
trace id 9c07fc2b inet fw4 forward unknown rule handle 134 (verdict drop)
trace id f42c9a1d inet fw4 trace_chain packet: iif "LanPort" ether saddr fc:3c:d7:10:d4:4f ether daddr 40:a5:ef:5d:42:e0 ip saddr 192.168.2.84 ip daddr 34.218.107.222 ip dscp cs0 ip ecn not-ect ip ttl 255 ip id 6688 ip protocol tcp ip length 119 tcp sport 62702 tcp dport 443 tcp flags == 0x18 tcp window 4380 
trace id f42c9a1d inet fw4 trace_chain unknown rule handle 127 (verdict continue)
trace id f42c9a1d inet fw4 raw_prerouting policy accept 
trace id f42c9a1d inet fw4 mangle_prerouting packet: iif "LanPort" ether saddr fc:3c:d7:10:d4:4f ether daddr 40:a5:ef:5d:42:e0 ip saddr 192.168.2.84 ip daddr 34.218.107.222 ip dscp cs0 ip ecn not-ect ip ttl 255 ip id 6688 ip protocol tcp ip length 119 tcp sport 62702 tcp dport 443 tcp flags == 0x18 tcp window 4380 
trace id f42c9a1d inet fw4 mangle_prerouting policy accept 
trace id f42c9a1d inet fw4 prerouting packet: iif "LanPort" ether saddr fc:3c:d7:10:d4:4f ether daddr 40:a5:ef:5d:42:e0 ip saddr 192.168.2.84 ip daddr 34.218.107.222 ip dscp cs0 ip ecn not-ect ip ttl 255 ip id 6688 ip protocol tcp ip length 119 tcp sport 62702 tcp dport 443 tcp flags == 0x18 tcp window 4380 
trace id f42c9a1d inet fw4 prerouting unknown rule handle 142 (verdict jump helper_FWLAN)
trace id f42c9a1d inet fw4 prerouting policy accept 
trace id f42c9a1d inet fw4 mangle_forward packet: iif "LanPort" oif "LanPort" ether saddr fc:3c:d7:10:d4:4f ether daddr 40:a5:ef:5d:42:e0 ip saddr 192.168.2.84 ip daddr 34.218.107.222 ip dscp cs0 ip ecn not-ect ip ttl 254 ip id 6688 ip protocol tcp ip length 119 tcp sport 62702 tcp dport 443 tcp flags == 0x18 tcp window 4380 
trace id f42c9a1d inet fw4 mangle_forward policy accept 
trace id f42c9a1d inet fw4 forward packet: iif "LanPort" oif "LanPort" ether saddr fc:3c:d7:10:d4:4f ether daddr 40:a5:ef:5d:42:e0 ip saddr 192.168.2.84 ip daddr 34.218.107.222 ip dscp cs0 ip ecn not-ect ip ttl 254 ip id 6688 ip protocol tcp ip length 119 tcp sport 62702 tcp dport 443 tcp flags == 0x18 tcp window 4380 
trace id f42c9a1d inet fw4 forward unknown rule handle 134 (verdict drop)
trace id b4ba0bd7 inet fw4 trace_chain packet: iif "LanPort" ether saddr fc:3c:d7:10:d4:4f ether daddr 40:a5:ef:5d:42:e0 ip saddr 192.168.2.84 ip daddr 34.218.107.222 ip dscp cs0 ip ecn not-ect ip ttl 255 ip id 6689 ip protocol tcp ip length 40 tcp sport 62702 tcp dport 443 tcp flags == ack tcp window 4380 
trace id b4ba0bd7 inet fw4 trace_chain unknown rule handle 127 (verdict continue)
trace id b4ba0bd7 inet fw4 raw_prerouting policy accept 
trace id b4ba0bd7 inet fw4 mangle_prerouting packet: iif "LanPort" ether saddr fc:3c:d7:10:d4:4f ether daddr 40:a5:ef:5d:42:e0 ip saddr 192.168.2.84 ip daddr 34.218.107.222 ip dscp cs0 ip ecn not-ect ip ttl 255 ip id 6689 ip protocol tcp ip length 40 tcp sport 62702 tcp dport 443 tcp flags == ack tcp window 4380 
trace id b4ba0bd7 inet fw4 mangle_prerouting policy accept 
trace id b4ba0bd7 inet fw4 prerouting packet: iif "LanPort" ether saddr fc:3c:d7:10:d4:4f ether daddr 40:a5:ef:5d:42:e0 ip saddr 192.168.2.84 ip daddr 34.218.107.222 ip dscp cs0 ip ecn not-ect ip ttl 255 ip id 6689 ip protocol tcp ip length 40 tcp sport 62702 tcp dport 443 tcp flags == ack tcp window 4380 
trace id b4ba0bd7 inet fw4 prerouting unknown rule handle 142 (verdict jump helper_FWLAN)
trace id b4ba0bd7 inet fw4 prerouting policy accept 
trace id b4ba0bd7 inet fw4 mangle_forward packet: iif "LanPort" oif "LanPort" ether saddr fc:3c:d7:10:d4:4f ether daddr 40:a5:ef:5d:42:e0 ip saddr 192.168.2.84 ip daddr 34.218.107.222 ip dscp cs0 ip ecn not-ect ip ttl 254 ip id 6689 ip protocol tcp ip length 40 tcp sport 62702 tcp dport 443 tcp flags == ack tcp window 4380 
trace id b4ba0bd7 inet fw4 mangle_forward policy accept 
trace id b4ba0bd7 inet fw4 forward packet: iif "LanPort" oif "LanPort" ether saddr fc:3c:d7:10:d4:4f ether daddr 40:a5:ef:5d:42:e0 ip saddr 192.168.2.84 ip daddr 34.218.107.222 ip dscp cs0 ip ecn not-ect ip ttl 254 ip id 6689 ip protocol tcp ip length 40 tcp sport 62702 tcp dport 443 tcp flags == ack tcp window 4380 
trace id b4ba0bd7 inet fw4 forward unknown rule handle 134 (verdict drop)

```

grosjo, 2025-07-13:
Meaning ?

grosjo, 2025-07-13:
- only 1 IP on LanPort
- this box is the default gateway for all devices in the particular building, and forward to 192.168.1.11 for internet
- FWLAN is just the name of the firewall zone for the lan

grosjo, 2025-07-13:
Again, all works perfectly fine when un-clicking the "drop invalid packets"  (connections go through, high speed internet for humans, cameras and IoT properly conencted, etc...). When clicking that field, ping goes trhough but ssh/http(s) are stopped by the firewall

brada4, 2025-07-14:
So you have everything set up
subnet1->interface1->contains route to subnet2 and firewall permitting
subnet2 ->interface2 ->contains route and no specific firewall rule so that connect responses pass
Can you run tcpdump -nvv on the involved interfaces to see packets exitting and returning? THere is no reason for invalid packets to appear in normal flat setup, invalid packets usually come from internet, or from nmap if you run that...

grosjo, 2025-07-15:
```
21:31:56.803493 IP (tos 0x0, ttl 254, id 28065, offset 0, flags [none], proto TCP (6), length 237)
    192.168.2.80.40285 > 34.210.19.179.8886: Flags [P.], cksum 0xfb31 (correct), seq 197:394, ack 70, win 3966, length 197
21:31:56.859183 IP (tos 0x0, ttl 255, id 64854, offset 0, flags [none], proto UDP (17), length 232)
    192.168.1.82.59729 > 255.255.255.255.6667: [udp sum ok] UDP, length 204
21:31:56.884638 IP (tos 0x48, ttl 64, id 57123, offset 0, flags [DF], proto TCP (6), length 192)
21:31:56.884995 IP (tos 0x48, ttl 64, id 57124, offset 0, flags [DF], proto TCP (6), length 800)
21:31:56.887209 IP (tos 0x48, ttl 64, id 57125, offset 0, flags [DF], proto TCP (6), length 480)
21:31:56.889849 IP (tos 0x48, ttl 64, id 57126, offset 0, flags [DF], proto TCP (6), length 1080)
21:31:56.894514 IP (tos 0x48, ttl 64, id 40742, offset 0, flags [DF], proto TCP (6), length 52)
21:31:56.896307 IP (tos 0x48, ttl 64, id 40743, offset 0, flags [DF], proto TCP (6), length 52)
21:31:56.897004 IP (tos 0x48, ttl 64, id 40744, offset 0, flags [DF], proto TCP (6), length 52)
21:31:56.899665 IP (tos 0x48, ttl 64, id 40745, offset 0, flags [DF], proto TCP (6), length 52)
21:31:56.960247 IP (tos 0x0, ttl 255, id 11849, offset 0, flags [none], proto UDP (17), length 232)
    192.168.2.83.59727 > 255.255.255.255.6667: [udp sum ok] UDP, length 204
21:31:56.994788 IP (tos 0x0, ttl 1, id 0, offset 0, flags [DF], proto UDP (17), length 668)
    192.168.1.21.8197 > 239.1.1.100.8197: [udp sum ok] UDP, length 640
21:31:57.169143 IP (tos 0x0, ttl 255, id 28066, offset 0, flags [none], proto TCP (6), length 40)
    192.168.2.80.40285 > 34.210.19.179.8886: Flags [.], cksum 0x0aa6 (correct), seq 394, ack 139, win 3897, length 0
21:31:57.169259 IP (tos 0x0, ttl 254, id 28066, offset 0, flags [none], proto TCP (6), length 40)
    192.168.2.80.40285 > 34.210.19.179.8886: Flags [.], cksum 0x0aa6 (correct), seq 394, ack 139, win 3897, length 0
21:31:57.224895 IP (tos 0x0, ttl 255, id 63611, offset 0, flags [none], proto UDP (17), length 232)
    192.168.1.86.59733 > 255.255.255.255.6667: [udp sum ok] UDP, length 204
21:31:57.589351 IP (tos 0x0, ttl 255, id 41625, offset 0, flags [none], proto UDP (17), length 232)
    192.168.1.83.59731 > 255.255.255.255.6667: [udp sum ok] UDP, length 204
21:31:57.776915 IP (tos 0x0, ttl 255, id 24718, offset 0, flags [none], proto UDP (17), length 232)
    192.168.1.80.59728 > 255.255.255.255.6667: [udp sum ok] UDP, length 204
21:31:57.927181 IP (tos 0x48, ttl 64, id 57127, offset 0, flags [DF], proto TCP (6), length 664)
21:31:57.927524 IP (tos 0x48, ttl 64, id 57128, offset 0, flags [DF], proto TCP (6), length 384)
21:31:57.929550 IP (tos 0x48, ttl 64, id 57129, offset 0, flags [DF], proto TCP (6), length 184)
21:31:57.931160 IP (tos 0x48, ttl 64, id 57130, offset 0, flags [DF], proto TCP (6), length 864)
21:31:57.936150 IP (tos 0x48, ttl 64, id 40746, offset 0, flags [DF], proto TCP (6), length 52)
21:31:57.936254 IP (tos 0x48, ttl 64, id 40747, offset 0, flags [DF], proto TCP (6), length 52)
21:31:57.936506 IP (tos 0x48, ttl 64, id 40748, offset 0, flags [DF], proto TCP (6), length 52)
21:31:57.940222 IP (tos 0x48, ttl 64, id 40749, offset 0, flags [DF], proto TCP (6), length 52)
21:31:58.107461 ARP, Ethernet (len 6), IPv4 (len 4), Request who-has 192.168.1.101 tell 192.168.1.101, length 46
21:31:58.201615 IP (tos 0x0, ttl 255, id 1656, offset 0, flags [none], proto UDP (17), length 232)
    192.168.1.85.59735 > 255.255.255.255.6667: [udp sum ok] UDP, length 204
21:31:58.297514 IP (tos 0x0, ttl 255, id 28067, offset 0, flags [none], proto UDP (17), length 232)
    192.168.2.80.59738 > 255.255.255.255.6667: [udp sum ok] UDP, length 204

```

grosjo, 2025-07-19:
@brada4 any clue ?

brada4, 2025-07-19:
I dont think you can be helped, you tirim IP addresses from your outputs making any diagnostics completely impoossible.

brada4, 2025-07-20:
21:31:57.927181 - 21:31:57.940222 IP packets without addresses.

grosjo, 2025-07-20:
I am not doign anything special here, just clicking on the box in Luci about invalid packets. I don' t understand what do you refer to

business-tomcat, 2025-10-21:
I have  a similar problem, where "drop invalid packets" drop packets, which are (in my case) valid.

Here is my setup:

[image]

WiFi IoT is used for a cam in a different IP zone (192.168.20.0/24) to block all requests from cam to internet.
But access from my home zone (192.168.0.0/24) should be allowed to the cam (obviously ;) )

So my server make access to the cam via RTMP to the cam IP.
Because the cam IP 192.168.20.10 is not in the zone of my server, the packets are routed to the standard gateway 192.168.0.1 (nanopi).
In my nanopi, there is a static IPv4 route to reroute packets "192.168.20.0/24 -> 192.168.0.101" to the WiFi AP.
Finally this WiFi AP routes the packet to the cam an everything is fine.

But if I activate "Drop invalid packets" at my nanopi (the standard gateway), the static IPv4 route is kind of blocked and not working, instead the RTMP requests ends in nirvana.
Here is the output from my server:

```
user@192.168.0.2:/$ ffmpeg -v verbose -re -i "rtmp://192.168.20.10/bcs/channel0_sub.bcs?channel=0&stream=0&user=user&password=pass" test.mp4
ffmpeg version 6.1.1-3ubuntu5+esm5 Copyright (c) 2000-2023 the FFmpeg developers
...
[tcp @ 0xaaaac4566b10] Starting connection attempt to 192.168.20.10 port 1935
[tcp @ 0xaaaac4566b10] Successfully connected to 192.168.20.10 port 1935
[rtmp @ 0xaaaac45a3e80] Cannot read RTMP handshake response
[in#0 @ 0xaaaac45a3850] Error opening input: Immediate exit requested
Error opening input file rtmp://192.168.20.10/bcs/channel0_sub.bcs?channel=0&stream=0&user=user&password=pass.
Exiting normally, received signal 2.
```

Not sure if this is an expected behaviour but it took my some time to figure out, why my RTMP access wasn't working.

If you think my LAN setup is bad, feel free to make improvements :) (I don't have managed switches, so I can't create VLANs).

PS:
More strange, if I change the access. I added a secondary wifi router to the unmanaged switch and the new wifi network is standard (IP from my nanopi via DHCP, same lan zone as the server). A computer connected to the wifi has no problems to make RTMP requests to the cam with "Drop invalid packets" activated. Why is there a difference between wifi and lan?!