openwrt/openwrt #19405: bisected: OpenWRT master branch - setsockopt(SO_ATTACH_FILTER) failed after git pull to latest git head

ZerBea, 2025-07-14:
### Describe the bug

After update to latest git head, it is not longer to attach a BPF to a RAW socket and result in an unspecified ERROR:
```
$ hcxdumptool -i rogue1 --bpf=test.bpf
failed to attach BPF (SO_ATTACH_FILTER): No error information
failed to open raw packet socket

1 ERROR(s) during runtime (mostly caused by a broken driver)
0 Packet(s) captured by kernel
0 Packet(s) dropped by kernel
exit on error
```

### OpenWrt version

r30136-cb1b656027

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

bcm27xx/bcm2708

### Device

Raspberry Pi Model B Plus Rev 1.2

### Image kind

Self-built image

### Steps to reproduce

$ hcxdumptool -i rogue1 --bpf=/root/filter/own_ext.bpf
failed to attach BPF (SO_ATTACH_FILTER): No error information
failed to open raw packet socket

1 ERROR(s) during runtime (mostly caused by a broken driver)
0 Packet(s) captured by kernel
0 Packet(s) dropped by kernel
exit on error

### Actual behaviour

Trying to attach a BPF to a RAW socket results in in unspecified ERROR.

### Expected behaviour

BPF successfully attached to RAW socket.

### Additional info

Bisected faulty commit as:
```
cb1b65602712a090539de29aefa38eb194c185d4 is the first bad commit
commit cb1b65602712a090539de29aefa38eb194c185d4
Author: Shiji Yang <yangshiji66@outlook.com>
Date:   Fri Jun 20 00:01:11 2025 +0800

    kernel: bump 6.12 to 6.12.34
    
    Changelog: https://cdn.kernel.org/pub/linux/kernel/v6.x/ChangeLog-6.12.34
    
    Remove upstreamed patches:
      generic/backport-6.12/421-01-v6.16-spi-bcm63xx-spi-fix-shared-reset.patch [1]
      generic/backport-6.12/421-02-v6.16-spi-bcm63xx-hsspi-fix-shared-reset.patch [2]
      generic/backport-6.12/610-06-v6.16-net-dsa-b53-do-not-enable-RGMII-delay-on-bcm63xx.patch [3]
      generic/backport-6.12/610-08-v6.16-net-dsa-b53-allow-RGMII-for-bcm63xx-RGMII-ports.patch [4]
      generic/backport-6.12/610-09-v6.16-net-dsa-b53-do-not-touch-DLL_IQQD-on-bcm53115.patch [5]
      generic/backport-6.12/611-v6.16-net-dsa-tag_brcm-legacy-fix-pskb_may_pull-length.patch [6]
    
    Manually rebased patches:
      bcm27xx/patches-6.12/950-0665-drm-vc4-tests-Drop-drm-parameter-for-vc4_find_crtc_f.patch [7]
    
    [1] https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?h=v6.12.34&id=408ca1d1803b223d615f9021055f9ccb4f4863ea
    [2] https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?h=v6.12.34&id=2a98786e258718ff93ef6d6bd26a9a39076e0cb7
    [3] https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?h=v6.12.34&id=6d1c93a5c6b0ae87bb7001d8d6fdef3b3be9c6c6
    [4] https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?h=v6.12.34&id=1aa31695bf0dc1ee3e6c559c14db7fd05b6bb102
    [5] https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?h=v6.12.34&id=b2fc08d276797e529cacad6fa9d704a7367090b5
    [6] https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?h=v6.12.34&id=2c32fc56c05aa69439fdfd5e0b25f57e2a158627
    [7] https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?h=v6.12.34&id=aba86d49e5ac3700295ab8c417436abacc19cc32
    
    Signed-off-by: Shiji Yang <yangshiji66@outlook.com>
    Link: https://github.com/openwrt/openwrt/pull/19184
    Signed-off-by: Hauke Mehrtens <hauke@hauke-m.de>

 ...-hid-Reduce-default-mouse-polling-interval-to-60Hz.patch |  4 +-
 ...m-panel-simple-Add-a-timing-for-the-Raspberry-Pi-7.patch | 16 ++++----
 ...-page_alloc-cma-introduce-a-customisable-threshold.patch |  2 +-
 ...22-Bluetooth-hci_sync-Add-fallback-bd-address-prop.patch |  4 +-
 .../patches-6.12/950-0344-sdhci-Add-SD-Express-hook.patch   |  2 +-
 ...58-mmc-sdhci-of-dwcmshc-define-sdio-timeout-clocks.patch | 16 ++++----
 .../950-0379-drivers-iommu-Add-BCM2712-IOMMU.patch          |  2 +-
 ...0-0502-Bluetooth-hci_sync-Fix-crash-on-NULL-parent.patch |  2 +-
 ...9-firmware-Add-an-RP1-firmware-interface-over-mbox.patch |  2 +-
 ...m-vc4-tests-Drop-drm-parameter-for-vc4_find_crtc_f.patch | 20 ++++-----
 ...950-0666-drm-vc4-tests-Return-the-allocated-output.patch |  2 +-
 ...xup-mmc-sdhci-of-dwcmshc-define-sdio-timeout-clock.patch |  2 +-
 .../421-01-v6.16-spi-bcm63xx-spi-fix-shared-reset.patch     | 32 ---------------
 .../421-02-v6.16-spi-bcm63xx-hsspi-fix-shared-reset.patch   | 32 ---------------
 ...-net-dsa-b53-bcm_sf2-implement-.support_eee-method.patch |  6 +--
 ...03-v6.15-net-dsa-b53-mdio-add-support-for-BCM53101.patch |  4 +-
 ...04-v6.16-net-dsa-b53-implement-setting-ageing-time.patch |  4 +-
 ...-05-v6.16-net-dsa-b53-do-not-enable-EEE-on-bcm63xx.patch |  4 +-
 ...6-net-dsa-b53-do-not-enable-RGMII-delay-on-bcm63xx.patch | 49 ----------------------
 ...t-dsa-b53-do-not-configure-bcm63xx-s-IMP-port-inte.patch |  8 ++--
 ...16-net-dsa-b53-allow-RGMII-for-bcm63xx-RGMII-ports.patch | 36 ----------------
 ...6.16-net-dsa-b53-do-not-touch-DLL_IQQD-on-bcm53115.patch | 59 ---------------------------
 ...6-net-dsa-tag_brcm-legacy-fix-pskb_may_pull-length.patch | 31 --------------
 .../612-03-v6.17-net-dsa-b53-support-legacy-FCS-tags.patch  |  2 +-
 .../612-04-v6.17-net-dsa-b53-detect-BCM5325-variants.patch  |  4 +-
 ...net-dsa-b53-prevent-DIS_LEARNING-access-on-BCM5325.patch |  2 +-
 ...-net-phy-support-active-high-property-for-PHY-LEDs.patch |  4 +-
 ...t-page_pool-try-to-free-deferred-skbs-while-waitin.patch |  4 +-
 target/linux/generic/hack-6.12/904-debloat_dma_buf.patch    |  2 +-
 target/linux/generic/kernel-6.12                            |  4 +-
 .../703-phy-add-detach-callback-to-struct-phy_driver.patch  |  8 ++--
 .../739-net-add-negotiation-of-in-band-capabilities.patch   |  6 +--
 .../706-include-linux-add-phy-ops-for-rtl838x.patch         |  2 +-
 .../patches-6.12/1011-net-stmmac-use-GFP_DMA32.patch        |  4 +-
 34 files changed, 71 insertions(+), 310 deletions(-)
 delete mode 100644 target/linux/generic/backport-6.12/421-01-v6.16-spi-bcm63xx-spi-fix-shared-reset.patch
 delete mode 100644 target/linux/generic/backport-6.12/421-02-v6.16-spi-bcm63xx-hsspi-fix-shared-reset.patch
 delete mode 100644 target/linux/generic/backport-6.12/610-06-v6.16-net-dsa-b53-do-not-enable-RGMII-delay-on-bcm63xx.patch
 delete mode 100644 target/linux/generic/backport-6.12/610-08-v6.16-net-dsa-b53-allow-RGMII-for-bcm63xx-RGMII-ports.patch
 delete mode 100644 target/linux/generic/backport-6.12/610-09-v6.16-net-dsa-b53-do-not-touch-DLL_IQQD-on-bcm53115.patch
 delete mode 100644 target/linux/generic/backport-6.12/611-v6.16-net-dsa-tag_brcm-legacy-fix-pskb_may_pull-length.patch
```

```
$ git bisect log
git bisect start
# good: [7f2d7fbc2357c0826db3e44717325107b1236f2a] starfive: drop 6.6 support
git bisect good 7f2d7fbc2357c0826db3e44717325107b1236f2a
# bad: [583f81ef0ee152404b633810607bbe15b0dcf9cf] wifi-scripts: enforce management frame protection on 6 GHz
git bisect bad 583f81ef0ee152404b633810607bbe15b0dcf9cf
# good: [a51359aafde18b7e809842b4b4c31c78f5fa0c51] kernel: mtdsplit_uimage: return 0 if not fatal
git bisect good a51359aafde18b7e809842b4b4c31c78f5fa0c51
# bad: [cb1b65602712a090539de29aefa38eb194c185d4] kernel: bump 6.12 to 6.12.34
git bisect bad cb1b65602712a090539de29aefa38eb194c185d4
# good: [1e1e91a213afc046e930f83bf4cfbcadbfa6d9d3] kernel/mvebu: Create kernel files for v6.12 (from v6.6)
git bisect good 1e1e91a213afc046e930f83bf4cfbcadbfa6d9d3
# good: [7e014bbe0b1591dada6ea5abfd70a2177d1a2747] mvebu: add 6.12 testing kernel
git bisect good 7e014bbe0b1591dada6ea5abfd70a2177d1a2747
# good: [985005219020ac1446e28217bfdd7a0a7a7a59d0] hostapd: fix generating own neighbor report for dynamically added/changed interfaces
git bisect good 985005219020ac1446e28217bfdd7a0a7a7a59d0
# good: [bbd95c0523174dc724a40b4d2483cadf7dc9907c] ucode: add ubus refcounting fix
git bisect good bbd95c0523174dc724a40b4d2483cadf7dc9907c
# first bad commit: [cb1b65602712a090539de29aefa38eb194c185d4] kernel: bump 6.12 to 6.12.34
```

Additional information
24.10 branch is affected, too!

### Diffconfig

```text
From my point of view this is not necessary.
```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

ZerBea, 2025-07-14:
To cross reference this report:
https://github.com/ZerBea/hcxdumptool/discussions/519

nbd168, 2025-07-22:
I'm pretty sure the faulty commit is commit 2bc6dffb4b72d53d6a6ada510269bf548c3f7ae0 ("bpf: Avoid __bpf_prog_ret0_warn when jit fails")
errno 524 is ENOTSUPP, which is returned from the function that this change touches

nbd168, 2025-07-22:
@ZerBea the issue might be architecture specific. It probably only triggers if the kernel can't JIT compile the BPF program.

ZerBea, 2025-07-22:
@nbd168 Thanks for pointing me into that direction. This could be possible.
Maybe other users running a different  architecture (than armv6) can confirm/deny this.

Unfortunately this patch breaks the BPF function completely.  
As a workaround, I'll add some code (like tcpdump does) to ignore errno 524.
From my point of vie this is not a good idea (tcpdump = passive / hcxdumptool = active).

nbd168, 2025-07-23:
Please try this patch to see if it makes things work again: https://nbd.name/p/1e3afdf5

ZerBea, 2025-07-23:
Thanks, I'll revert 2bc6dffb4b72d53d6a6ada510269bf548c3f7ae0 as you suggested by your patch.

BTW:
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?id=2bc6dffb4b72d53d6a6ada510269bf548c3f7ae0
```
When creating bpf program, 'fp->jit_requested' depends on bpf_jit_enable.
This issue is triggered because of CONFIG_BPF_JIT_ALWAYS_ON is not set
and bpf_jit_enable is set to 1, causing the arch to attempt JIT the prog,
but jit failed due to FAULT_INJECTION. As a result, incorrectly
treats the program as valid, when the program runs it calls
`__bpf_prog_ret0_warn` and triggers the WARN_ON_ONCE(1).

Maybe that's the difference (CONFIG_BPF_JIT_ALWAYS_ON=y):
```

Arch (setsockopt(SO_ATTACH_FILTER is working)
```
CONFIG_BPF=y
CONFIG_ARCH_WANT_DEFAULT_BPF_JIT=y
CONFIG_BPF_SYSCALL=y
CONFIG_BPF_JIT=y
CONFIG_BPF_JIT_ALWAYS_ON=y
CONFIG_BPF_JIT_DEFAULT_ON=y
CONFIG_BPF_UNPRIV_DEFAULT_OFF=y
CONFIG_BPF_PRELOAD=y
CONFIG_BPF_PRELOAD_UMD=m
CONFIG_BPF_LSM=y
CONFIG_CGROUP_BPF=y
CONFIG_IPV6_SEG6_BPF=y
CONFIG_NETFILTER_BPF_LINK=y
CONFIG_NETFILTER_XT_MATCH_BPF=m
CONFIG_NET_CLS_BPF=m
CONFIG_NET_ACT_BPF=m
CONFIG_BPF_STREAM_PARSER=y
CONFIG_LWTUNNEL_BPF=y
CONFIG_BPF_LIRC_MODE2=y
CONFIG_HID_BPF=y
CONFIG_BPF_EVENTS=y
CONFIG_BPF_KPROBE_OVERRIDE=y
# CONFIG_TEST_BPF is not set
```

Debian (setsockopt(SO_ATTACH_FILTER is working)
```
CONFIG_BPF=y
CONFIG_BPF_JIT=y
# CONFIG_BPF_JIT_ALWAYS_ON is not set
CONFIG_BPF_JIT_DEFAULT_ON=y
# CONFIG_BPF_LSM is not set
# CONFIG_BPF_PRELOAD is not set
# CONFIG_BPF_STREAM_PARSER is not set
CONFIG_BPF_SYSCALL=y
CONFIG_BPF_UNPRIV_DEFAULT_OFF=y
# CONFIG_HID_BPF is not set
# CONFIG_NETFILTER_XT_MATCH_BPF is not set
# CONFIG_NET_ACT_BPF is not set
# CONFIG_NET_CLS_BPF is not set
# CONFIG_TEST_BPF is not set
```

OpenWRT:
```
CONFIG_BPF=y
CONFIG_BPF_JIT=y
# CONFIG_BPF_JIT_ALWAYS_ON is not set
CONFIG_BPF_JIT_DEFAULT_ON=y
# CONFIG_BPF_LSM is not set
# CONFIG_BPF_PRELOAD is not set
# CONFIG_BPF_STREAM_PARSER is not set
CONFIG_BPF_SYSCALL=y
CONFIG_BPF_UNPRIV_DEFAULT_OFF=y
# CONFIG_HID_BPF is not set
# CONFIG_NETFILTER_XT_MATCH_BPF is not set
# CONFIG_NET_ACT_BPF is not set
# CONFIG_NET_CLS_BPF is not set
# CONFIG_TEST_BPF is not set

```

nbd168, 2025-07-23:
@ZerBea this is not a selectable option, and I think it's actually filtered out from the build system config files. You need to check the .config in your kernel source directory to see the effective value. I'd expect it to be set to =y though.
I still think this is related to JIT not being able to compile some programs, and the stable patch making that fail the entire attach.
By the way, I also contacted the author of that patch and the lists it was sent to. Hopefully we'll get a response soon

nbd168, 2025-07-24:
@mannkafai either way, I think your patch needs to be reverted or fixed immediately, since it introduces a significant regression.

nbd168, 2025-07-24:
@mannkafai in the function that you changed, I found this comment:
```
    /* eBPF JITs can rewrite the program in case constant
     * blinding is active. However, in case of error during
     * blinding, bpf_int_jit_compile() must always return a
     * valid program, which in this case would simply not
     * be JITed, but falls back to the interpreter.
     */
```
To me this strongly suggests that there is supposed to be a fallback to the interpreter if JIT fails. A few years ago, I ran into other architecture specific eBPF JIT failures, and my programs always worked anyway.

So what's the reason behind not doing graceful fallback here?

nbd168, 2025-07-24:
@mannkafai so let me try to summarize my understanding of the current situation, please correct me if I'm wrong:

1. The above commit catches a corner case for some programs that require JIT. When JIT is not requested, those programs are rejected right away. When JIT is requested and those specific programs are not JITed, a warning is printed. So far there is still a valid fallback path for all programs that don't hit this corner case.
2. In order to eliminate potential warnings for this particular corner case, you eliminated the fallback path entirely.

Is this summary correct, or am I missing something?

mannkafai, 2025-07-25:
If `fallback path` means `fallback to interpreter`, I think it's not, the above commit change the default fp->bpf_func which BPF programs executed if CONFIG_BPF_JIT_ALWAYS_ON not set.

I agree this should be fixd. Could you share the reproducer on x86_64?

nbd168, 2025-07-25:
@mannkafai The above commit only changes fp->bpf_func for the case where the stack size is too large. For all other cases, fallback to interpreter in case of JIT failure still works.
Only after your commit does it reject programs in all cases of JIT failure.
I think your patch should be reworked to only reject fallback from JIT failure in cases of large stack size, not in all other cases.

mannkafai, 2025-07-26:
@nbd168 @ZerBea 
> The above commit only changes fp->bpf_func for the case where the stack size is too large. 

No, fp->bpf_func is modified when fp->jit_requested is set to true. 

Could you provide some examples JIT fails but works fine under the interpreter? This would help to identify the root cause.

ZerBea, 2025-07-26:
@mannkafai for sure I can add some examples.

Environment:
Latest git head OpenWRT 24.10 (compiled via toolchain) (because master branch has some additional problems):
```
uname -a
Linux OpenWrt 6.6.99 #0 Thu Jul 24 01:46:38 2025 armv6l GNU/Linux
```
Test system:
```
cat /sys/firmware/devicetree/base/model
Raspberry Pi Model B Plus Rev 1.2
``
The tests are done via more common and passive tcpdump Instead of active hcxdumptool/hcxlabtool:
```
tcpdump -h
tcpdump version 4.99.5
libpcap version 1.10.5 (with TPACKET_V3)
32-bit build, 64-bit time_t
```

Filter examples:
```
tcpdump -i mon1 "not type ctl subtype ack and not type ctl subtype cts"

tcpdump -i mon1 "not type ctl subtype ack and not type ctl subtype cts and not wlan addr3 3c37121a2b3c"

tcpdump -i mon1 "not type ctl subtype ack and not type ctl subtype cts and not wlan addr3 3c37121a2b3c and not wlan addr2 184ecbca2a3a"

tcpdump -i mon1 "not type ctl subtype ack and not type ctl subtype cts and not wlan addr3 3c37121a2b3c and not wlan addr2 184ecbca2a3a and not wlan addr2 14130b4d3f47"

tcpdump -i mon1 "not type ctl subtype ack and not type ctl subtype cts and not wlan addr3 3c37121a2b3c and not wlan addr2 184ecbca2a3a and not wlan addr2 14130b4d3f47 and not wlan addr2 f0f61cf440b7"

tcpdump -i mon1 "not type ctl subtype ack and not type ctl subtype cts and not wlan addr3 3c37121a2b3c and not wlan addr2 184ecbca2a3a and not wlan addr2 14130b4d3f47 and not wlan addr2 f0f61cf440b7 and not wlan addr3 a84b4dedf471"

tcpdump -i mon1 "not type ctl subtype ack and not type ctl subtype cts and not wlan addr3 3c37121a2b3c and not wlan addr2 184ecbca2a3a and not wlan addr2 14130b4d3f47 and not wlan addr2 f0f61cf440b7 and not wlan addr3 a84b4dedf471 and not wlan addr3 d022be17e1d7"

tcpdump -i mon1 "not type ctl subtype ack and not type ctl subtype cts and not wlan addr3 3c37121a2b3c and not wlan addr2 184ecbca2a3a and not wlan addr2 14130b4d3f47 and not wlan addr2 f0f61cf440b7 and not wlan addr3 a84b4dedf471 and not wlan addr3 d022be17e1d7 and not wlan addr3 5c497967208b"

tcpdump -i mon1 "not type ctl subtype ack and not type ctl subtype cts and not wlan addr3 3c37121a2b3c and not wlan addr2 184ecbca2a3a and not wlan addr2 14130b4d3f47 and not wlan addr2 f0f61cf440b7 and not wlan addr3 a84b4dedf471 and not wlan addr3 d022be17e1d7 and not wlan addr3 5c497967208b and not wlan addr2 706655784d5b"
Warning: Kernel filter failed: No error information

tcpdump -i mon1 "not wlan addr3 3c37121a2b3c and not wlan addr2 184ecbca2a3a and not wlan addr2 14130b4d3f47 and not wlan addr2 f0f61cf440b7 and not wlan addr3 a84b4dedf471 and not wlan addr3 d022be17e1d7 and not wlan addr3 5c497967208b and not wlan addr2 706655784d5b"
Warning: Kernel filter failed: No error information
```

mannkafai, 2025-08-11:
Hi, @ZerBea, I'm working the patch on kernel, and need a test for it.

> tcpdump -i mon1 "not type ctl subtype ack and not type ctl subtype cts and not wlan addr3 3c37121a2b3c and not wlan addr2 184ecbca2a3a and not wlan addr2 14130b4d3f47 and not wlan addr2 f0f61cf440b7 and not wlan addr3 a84b4dedf471 and not wlan addr3 d022be17e1d7 and not wlan addr3 5c497967208b and not wlan addr2 706655784d5b"
> tcpdump -i mon1 "not wlan addr3 3c37121a2b3c and not wlan addr2 184ecbca2a3a and not wlan addr2 14130b4d3f47 and not wlan addr2 f0f61cf440b7 and not wlan addr3 a84b4dedf471 and not wlan addr3 d022be17e1d7 and not wlan addr3 5c497967208b and not wlan addr2 706655784d5b"

These tests are fine, but only works whit wlan interface which support 802.11, I'm not sure is ok on kernel selftest.  I do not have any ARMv6 devices,  could you help to check if the following example works on armv6,  or provide some examples  which not work on Ethernet (802.3)?

> tcpdump -i wlp7s0 "not ether host 3c37121a2b3c and not ether host 184ecbca2a3a and not ether host 14130b4d3f47 and not ether host f0f61cf440b7 and not ether host a84b4dedf471 and not ether host d022be17e1d7 and not ether host 5c497967208b and not ether host 706655784d5b"

ZerBea, 2025-08-11:
That will take awhile. Right now, I can't compile latest git head of master branch due to this ERROR:
```
CFLAGS="-O2 -I/home/zerobeat/temp/openwrt/staging_dir/host/include " CPPFLAGS="-I/home/zerobeat/temp/openwrt/staging_dir/host/include " CXXFLAGS="-O2 -I/home/zerobeat/temp/openwrt/staging_dir/host/include " LDFLAGS="-L/home/zerobeat/temp/openwrt/staging_dir/host/lib " make  -C /home/zerobeat/temp/openwrt/build_dir/host/m4-1.4.19/.  
make[4]: Entering directory '/home/zerobeat/temp/openwrt/build_dir/host/m4-1.4.19'
make  all-recursive
make[5]: Entering directory '/home/zerobeat/temp/openwrt/build_dir/host/m4-1.4.19'
Making all in .
make[6]: Entering directory '/home/zerobeat/temp/openwrt/build_dir/host/m4-1.4.19'
make[6]: Nothing to be done for 'all-am'.
make[6]: Leaving directory '/home/zerobeat/temp/openwrt/build_dir/host/m4-1.4.19'
Making all in examples
make[6]: Entering directory '/home/zerobeat/temp/openwrt/build_dir/host/m4-1.4.19/examples'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/zerobeat/temp/openwrt/build_dir/host/m4-1.4.19/examples'
Making all in lib
make[6]: Entering directory '/home/zerobeat/temp/openwrt/build_dir/host/m4-1.4.19/lib'
make  all-am
make[7]: Entering directory '/home/zerobeat/temp/openwrt/build_dir/host/m4-1.4.19/lib'
/home/zerobeat/temp/openwrt/staging_dir/host/bin/gcc  -I.   -I/home/zerobeat/temp/openwrt/staging_dir/host/include   -O2 -I/home/zerobeat/temp/openwrt/staging_dir/host/include  -c -o gl_avltree_oset.o gl_avltree_oset.c
In file included from gl_avltree_oset.h:21,
                 from gl_avltree_oset.c:21:
gl_oset.h:275:1: warning: 'nodiscard' attribute ignored [-Wattributes]
  275 | GL_OSET_INLINE _GL_ATTRIBUTE_NODISCARD int
      | ^~~~~~~~~~~~~~
gl_oset.h:275:40: error: expected identifier or '(' before 'int'
  275 | GL_OSET_INLINE _GL_ATTRIBUTE_NODISCARD int
      |                                        ^~~
make[7]: *** [Makefile:2871: gl_avltree_oset.o] Error 1
make[7]: Leaving directory '/home/zerobeat/temp/openwrt/build_dir/host/m4-1.4.19/lib'
make[6]: *** [Makefile:2481: all] Error 2
make[6]: Leaving directory '/home/zerobeat/temp/openwrt/build_dir/host/m4-1.4.19/lib'
make[5]: *** [Makefile:2018: all-recursive] Error 1
make[5]: Leaving directory '/home/zerobeat/temp/openwrt/build_dir/host/m4-1.4.19'
make[4]: *** [Makefile:1974: all] Error 2
make[4]: Leaving directory '/home/zerobeat/temp/openwrt/build_dir/host/m4-1.4.19'
make[3]: *** [Makefile:28: /home/zerobeat/temp/openwrt/build_dir/host/m4-1.4.19/.built] Error 2
make[3]: Leaving directory '/home/zerobeat/temp/openwrt/tools/m4'
time: tools/m4/compile#0.12#0.05#0.18
    ERROR: tools/m4 failed to build.
```
compiling of m4 failed epically.

I'm trying 24.10 branch, now.

ZerBea, 2025-08-11:
Tested, running latest git head 24.10 branch:
```
$ uname -a
Linux (none) 6.6.100 #0 Sun Aug 10 19:53:43 2025 armv6l GNU/Linux
$ uname -v
#0 Sun Aug 10 19:53:43 2025
```

As expected, broken on 802.3 too, in configurations with CONFIG_BPF_JIT_DEFAULT_ON=y when programs can't be JITed:
```
$ tcpdump -i eth0 "not ether host 3c37121a2b3c and not ether host 184ecbca2a3a an
d not ether host 14130b4d3f47 and not ether host f0f61cf440b7 and not ether host a84b4dedf471 
and not ether host d022be17e1d7 and not ether host 5c497967208b and not ether host 706655784d5
b"
Warning: Kernel filter failed: No error information
```

Also expected, a filter code with less entries is working.
Four entries are working:
```
$ tcpdump -i eth0 "not ether host 3c37121a2b3c and not ether host 184ecbca2a3a an
d not ether host 14130b4d3f47 and not ether host f0f61cf440b7"
tcpdump: verbose output suppressed, use -v[v]... for full protocol decode
listening on eth0, link-type EN10MB (Ethernet), snapshot length 262144 bytes
```
Five or more entries are broken:
```
$ tcpdump -i eth0 "not ether host 3c37121a2b3c and not ether host 184ecbca2a3a an
d not ether host 14130b4d3f47 and not ether host f0f61cf440b7 and not ether host a84b4dedf471"
Warning: Kernel filter failed: No error information
```

ZerBea, 2025-09-10:
Great work. Everything is running as expected.
Thanks for your effort.