openwrt/openwrt #19449: Hardware flow offloading not working in MediaTek MT7981BA

AfsanXD, 2025-07-18:
I'm currently using OpenWRT 24.10.1 with Cudy WR3000s v1 which has MediaTek MT7981BA ( Filogic 820 ). I've noticed that HW flow offloading doesn't work intended, when using HW flow offloading at max load (942Mbps LAN-WAN), the router experiences a maximum latency of 3ms, whereas using software flow offloading keeps it below 1ms.  I also noticed that bandwidth flow is much smoother using software flow offloading, with no bumps. Where as bandwidth flow throttles using HW flow offloading. Is there any way to fix it ?? HW flow offloading seems like a superior option over software flow offloading after everything I've read about it.

nbd168, 2025-07-18:
In my opinion, if the CPU is able to handle the given throughput via software offloading, then that should be preferred over hardware offloading. The software is able to manage latency well through proper qdisc management like fq_codel. That necessarily gets thrown out when you enable HW flow offloading, so you see higher latency when running at capacity limit.

lorand-horvath, 2025-07-18:
I also thought that hardware flow offloading (HFO) was superior to software offloading (SFO) if (this is important to specify) if SQM is not used. When SQM is enabled, which is NOT compatible with HFO, obviously the latency is lower.

But without using SQM, HFO *should* be better. I hope I'm understanding this correctly...

lorand-horvath, 2025-07-18:
> It is superior, but when it is un-offloaded on timer...

What do you mean by this? Can you give more details?

brada4, 2025-07-18:
>  mean by this? 

It is well documented, eg kernel docs. It adds more variability due to lower latency in offloaded path.

AfsanXD, 2025-07-20:
Is there any way to make use of the hardware flow offloading on this soc :(

closed 2025-07-22 by AfsanXD