openwrt/openwrt #19469: Problem compiling Linux kernel with GCC 15

mhalano, 2025-07-20:
### Describe the bug

I got an error when compiling the image from snapshot using GCC 15.0.

I confess I did some other changes, but it seems to me that is related to me using GCC 15.0.

Here is the relevant snippet:
```
# CC [M]  /home/mhalano/code/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mac80211-regular/backports-6.14.11/net/mac80211/cfg.o
  aarch64-openwrt-linux-musl-gcc -Wp,-MMD,/home/mhalano/code/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mac80211-regular/backports-6.14.11/net/mac80211/.cfg.o.d -nostdinc -I/home/mhalano/code/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mac80211-regular/backports-6.14.11/backport-include/ -I/home/mhalano/code/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mac80211-regular/backports-6.14.11/backport-include/uapi -I/home/mhalano/code/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mac80211-regular/backports-6.14.11/include/ -I/home/mhalano/code/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mac80211-regular/backports-6.14.11/include/uapi -include /home/mhalano/code/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mac80211-regular/backports-6.14.11/backport-include/backport/backport.h -Wno-unused-but-set-variable -DCPTCFG_VERSION=\""v6.1.110-1-43-g2e32643b84e8"\" -DCPTCFG_KERNEL_VERSION=\""v6.14.11-0-gb9d5d463c216"\" -DCPTCFG_KERNEL_NAME=\""Linux"\"   -I./arch/arm64/include -I./arch/arm64/include/generated  -I./include -I./arch/arm64/include/uapi -I./arch/arm64/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/compiler-version.h -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -mlittle-endian -DKASAN_SHADOW_SCALE_SHIFT= -fmacro-prefix-map=./= -Werror -std=gnu11 -fshort-wchar -funsigned-char -fno-common -fno-PIE -fno-strict-aliasing -mgeneral-regs-only -DCONFIG_CC_HAS_K_CONSTRAINT=1 -Wno-psabi -mabi=lp64 -fno-asynchronous-unwind-tables -fno-unwind-tables -mbranch-protection=none -Wa,-march=armv8.5-a -DARM64_ASM_ARCH='"armv8.5-a"' -DKASAN_SHADOW_SCALE_SHIFT= -fno-delete-null-pointer-checks -O2 -fno-allow-store-data-races -fstack-protector -fno-omit-frame-pointer -fno-optimize-sibling-calls -fno-stack-clash-protection -fmin-function-alignment=4 -fstrict-flex-arrays=3 -fno-strict-overflow -fno-stack-check -fconserve-stack -fno-builtin-wcslen -fno-builtin-wcslen -Wall -Wextra -Wundef -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Werror=strict-prototypes -Wno-format-security -Wno-trigraphs -Wno-frame-address -Wno-address-of-packed-member -Wmissing-declarations -Wmissing-prototypes -Wframe-larger-than=2048 -Wno-main -Wno-dangling-pointer -Wvla -Wno-pointer-sign -Wcast-function-type -Wno-unterminated-string-initialization -Wno-array-bounds -Wno-stringop-overflow -Wno-alloc-size-larger-than -Wimplicit-fallthrough=5 -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -Wenum-conversion -fzero-init-padding-bits=all -Wunused -Wno-unused-but-set-variable -Wno-unused-const-variable -Wno-packed-not-aligned -Wno-format-overflow -Wno-format-truncation -Wno-stringop-truncation -Wno-override-init -Wno-missing-field-initializers -Wno-type-limits -Wno-shift-negative-value -Wno-maybe-uninitialized -Wno-sign-compare -Wno-unused-parameter -g -fno-var-tracking -femit-struct-debug-baseonly -ffile-prefix-map=/home/mhalano/code/openwrt/build_dir/target-aarch64_cortex-a53_musl=target-aarch64_cortex-a53_musl -fno-caller-saves -mstack-protector-guard=sysreg -mstack-protector-guard-reg=sp_el0 -mstack-protector-guard-offset=1160 -fno-pie -no-pie -std=gnu11 -DDEBUG -I/home/mhalano/code/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mac80211-regular/backports-6.14.11/include -ffile-prefix-map=/home/mhalano/code/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mac80211-regular/backports-6.14.11=backports-6.14.11  -DMODULE  -DKBUILD_BASENAME='"cfg"' -DKBUILD_MODNAME='"mac80211"' -D__KBUILD_MODNAME=kmod_mac80211 -c -o /home/mhalano/code/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mac80211-regular/backports-6.14.11/net/mac80211/cfg.o /home/mhalano/code/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mac80211-regular/backports-6.14.11/net/mac80211/cfg.c
In file included from ./include/linux/string.h:390,
                 from /home/mhalano/code/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mac80211-regular/backports-6.14.11/backport-include/linux/string.h:3,
                 from ./include/linux/bitmap.h:13,
                 from /home/mhalano/code/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mac80211-regular/backports-6.14.11/backport-include/linux/bitmap.h:3,
                 from ./include/linux/cpumask.h:12,
                 from ./arch/arm64/include/asm/cpufeature.h:27,
                 from ./arch/arm64/include/asm/hwcap.h:9,
                 from ./arch/arm64/include/asm/arch_timer.h:12,
                 from ./arch/arm64/include/asm/timex.h:8,
                 from ./include/linux/timex.h:67,
                 from ./include/linux/time32.h:13,
                 from ./include/linux/time.h:60,
                 from /home/mhalano/code/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mac80211-regular/backports-6.14.11/backport-include/linux/time.h:3,
                 from ./include/linux/skbuff.h:15,
                 from /home/mhalano/code/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mac80211-regular/backports-6.14.11/backport-include/linux/skbuff.h:3,
                 from ./include/linux/if_ether.h:19,
                 from /home/mhalano/code/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mac80211-regular/backports-6.14.11/backport-include/linux/if_ether.h:3,
                 from /home/mhalano/code/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mac80211-regular/backports-6.14.11/include/linux/ieee80211.h:19,
                 from /home/mhalano/code/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mac80211-regular/backports-6.14.11/net/mac80211/cfg.c:11:
In function 'fortify_memcpy_chk',
    inlined from 'copy_mesh_setup' at /home/mhalano/code/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mac80211-regular/backports-6.14.11/net/mac80211/cfg.c:2534:2,
    inlined from 'ieee80211_join_mesh' at /home/mhalano/code/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mac80211-regular/backports-6.14.11/net/mac80211/cfg.c:2687:8:
./include/linux/fortify-string.h:571:25: error: call to '__write_overflow_field' declared with attribute warning: detected write beyond size of field (1st parameter); maybe use struct_group()? [-Werror=attribute-warning]
  571 |                         __write_overflow_field(p_size_field, size);
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[9]: *** [scripts/Makefile.build:229: /home/mhalano/code/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mac80211-regular/backports-6.14.11/net/mac80211/cfg.o] Error 1
make[8]: *** [scripts/Makefile.build:478: /home/mhalano/code/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mac80211-regular/backports-6.14.11/net/mac80211] Error 2
make[7]: *** [Makefile:1945: /home/mhalano/code/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mac80211-regular/backports-6.14.11] Error 2
make[6]: *** [Makefile.build:13: modules] Error 2
make[5]: *** [Makefile.real:105: modules] Error 2
make[4]: *** [Makefile:121: modules] Error 2
make[3]: *** [Makefile:403: /home/mhalano/code/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mac80211-regular/backports-6.14.11/.built] Error 2
make[3]: Leaving directory '/home/mhalano/code/openwrt/package/kernel/mac80211'
time: package/kernel/mac80211/regular/compile#3.83#1.14#4.95
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[2]: *** [package/Makefile:183: package/kernel/mac80211/compile] Error 1
make[2]: Leaving directory '/home/mhalano/code/openwrt'
[... cut ...]
CONFIG_VERSION_FILENAMES=y
CONFIG_VERSION_FIRMWARE_URL=""
CONFIG_VERSION_HOME_URL=""
CONFIG_VERSION_HWREV=""
CONFIG_VERSION_MANUFACTURER=""
CONFIG_VERSION_MANUFACTURER_URL=""
CONFIG_VERSION_NUMBER=""
CONFIG_VERSION_PRODUCT=""
CONFIG_VERSION_REPO="https://downloads.openwrt.org/releases/24.10.1"
CONFIG_VERSION_SUPPORT_URL=""
CONFIG_WOLFSSL_HAS_NO_HW=y
CONFIG_WPA_WOLFSSL=y
CONFIG_KERNEL_WERROR=y
# CONFIG_VERSION_CODE_FILENAMES is not set
```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

mhalano, 2025-07-20:
I will try compiling bumping the binutils to version 2.44. Maybe it helps, maybe it does not.

castiel652, 2025-07-21:
https://marc.info/?l=linux-wireless&m=175008144021893&w=2
Looks like it's because of GCC15

andros-ua, 2025-08-02:
Same here with GCC15:

```
time: package/base-files/compile#1.81#0.56#2.36
In file included from ./include/linux/string.h:390,
                 from ../mac80211-regular/backports-6.14.11/backport-include/linux/string.h:3,
                 from ./include/linux/bitmap.h:13,
                 from ../mac80211-regular/backports-6.14.11/backport-include/linux/bitmap.h:3,
                 from ./include/linux/cpumask.h:12,
                 from ./arch/arm64/include/asm/cpufeature.h:27,
                 from ./arch/arm64/include/asm/hwcap.h:9,
                 from ./arch/arm64/include/asm/arch_timer.h:12,
                 from ./arch/arm64/include/asm/timex.h:8,
                 from ./include/linux/timex.h:67,
                 from ./include/linux/time32.h:13,
                 from ./include/linux/time.h:60,
                 from ../mac80211-regular/backports-6.14.11/backport-include/linux/time.h:3,
                 from ./include/linux/skbuff.h:15,
                 from ../mac80211-regular/backports-6.14.11/backport-include/linux/skbuff.h:3,
                 from ./include/linux/if_ether.h:19,
                 from ../mac80211-regular/backports-6.14.11/backport-include/linux/if_ether.h:3,
                 from ../mac80211-regular/backports-6.14.11/include/linux/ieee80211.h:19,
                 from ../mac80211-regular/backports-6.14.11/net/mac80211/cfg.c:11:
In function 'fortify_memcpy_chk',
    inlined from 'copy_mesh_setup' at ../mac80211-regular/backports-6.14.11/net/mac80211/cfg.c:2534:2,
    inlined from 'ieee80211_join_mesh' at ../mac80211-regular/backports-6.14.11/net/mac80211/cfg.c:2687:8:
./include/linux/fortify-string.h:571:25: error: call to 'write_overflow_field' declared with attribute warning: detected write beyond size of field (1st parameter); maybe use struct_group()? [-Werror=attribute-warning]
  571 |                         write_overflow_field(p_size_field, size);
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[9]:  [scripts/Makefile.build:229: /home/andros/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mac80211-regular/backports-6.14.11/net/mac80211/cfg.o] Error 1
make[8]:  [scripts/Makefile.build:478: /home/andros/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mac80211-regular/backports-6.14.11/net/mac80211] Error 2
make[7]:  [Makefile:1945: /home/andros/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mac80211-regular/backports-6.14.11] Error 2
make[6]:  [Makefile.build:13: modules] Error 2
make[5]:  [Makefile.real:105: modules] Error 2
make[4]:  [Makefile:121: modules] Error 2
make[4]: Leaving directory '/home/andros/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mac80211-regular/backports-6.14.11'
make[3]:  [Makefile:403: /home/andros/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mac80211-regular/backports-6.14.11/.built] Error 2
make[3]: Leaving directory '/home/andros/openwrt/package/kernel/mac80211'
time: package/kernel/mac80211/regular/compile#3.76#0.51#4.03
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[2]:  [package/Makefile:183: package/kernel/mac80211/compile] Error 1
make[2]: Leaving directory '/home/andros/openwrt'
make[1]:  [package/Makefile:177: /home/andros/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/andros/openwrt'
make:  [/home/andros/openwrt/include/toplevel.mk:233: world] Error 2
```