openwrt/openwrt #19482: Firewall4: Time-based rule: Error: Range negative size

DazzyWalkman, 2025-07-21:
### Describe the bug

I have a time based firewall rule, with my timezone in /etc/config/system set to "CST-8":
`config rule
	option src '*'
	option dest '*'
	option name 'TR4mbh'
	list src_mac 'YY:XX:YY:XX:YY:XX'
	list src_ip '192.168.1.178'
	option target 'REJECT'
	option start_time '00:00:00'
	option stop_time '08:00:00'
`
It's fine on e1d39bdbdb, but since bce0ad5563 upto 06267823a3 when boot or firewall reload, there's an error stating:
_/dev/stdin:43:83-103: Error: Range negative size
		meta l4proto udp ip saddr 192.168.1.178 ether saddr YY:XX:YY:XX:YY:XX meta hour "00:00:00"-"08:00:00" counter jump handle_reject comment "!fw4: TR4mbh"
		                                                                                ^^^^^^^^^^^^^^^^^^^^^_
Adding rules in /etc/config/firewall to nftables subsequently failed. 

### OpenWrt version

r30385-06267823a3

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

x86/64

### Device

Generic

### Image kind

Official downloaded image

### Actual behaviour

My observations:
1. When I changed the timezone from "CST-8" to "UTC", the error disapeared, rules added normally.
2. With CST-8 timezone, I changed the time range to "00:00:00"-"07:59:59",  the error disapeared, rules added normally, now as a workaround for me.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-07-21:
Reported upstream (originally reported in forums by @thess )
```
https://bugzilla.netfilter.org/show_bug.cgi?id=1805
```
Workaround ...
/sbin/fw4
```diff
- nft
+ TZ=Z nft
```

thess, 2025-08-01:
I can confirm patch applies cleanly and times in fw4 rules now appear to be correct. I spun it up yesterday with 1.1.3 and put it to work on house router with no errors.

brada4, 2025-08-01:
nftables HEAD works correctly too. Thanks for rushing 2nd part.

dave14305, 2025-08-07:
nftables 1.1.4 has been released with the hour fix.

closed 2025-09-20 by thess