openwrt/openwrt #19483: kmod-mac80211-hwsim not create wlan interfaces

Raizo62, 2025-07-21:
### Describe the bug

Hi

I use "openwrt-24.10.1-x86-64-generic-ext4-combined.img.gz"

Loading the kmod-mac80211-hwsim module should create virtual wlan0 interfaces, but this module only creates the "hwsim0" interface.
I tested it with the following commands:
- Classic installation: `opkg install kmod-mac80211-hwsim`
- Classic loading :
```
rmmod mac80211-hwsim
modprobe mac80211-hwsim
```
- Loading by specifying the number of interfaces :
```
rmmod mac80211-hwsim
modprobe mac80211-hwsim radios=2
```

It worked with OpenWRT 23.05.5 (x86-64-generic-ext4-combined.img)

### OpenWrt version

r28739-d9340319c6

### OpenWrt release

24.10.2

### OpenWrt target/subtarget

x86/64

### Device

QEMU Standard PC (i440FX + PIIX, 1996)

### Image kind

Official downloaded image

### Steps to reproduce

`root@OpenWrt:~# opkg update`

`root@OpenWrt:~# opkg install kmod-mac80211-hwsim`

### Actual behaviour

We don't have wlan interfaces : 
```
root@OpenWrt:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel master br-lan state UP qlen 1000
    link/ether 0c:9b:5e:ac:00:00 brd ff:ff:ff:ff:ff:ff
3: eth1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP qlen 1000
    link/ether 0c:9b:5e:ac:00:01 brd ff:ff:ff:ff:ff:ff
    inet 10.145.147.11/24 brd 10.145.147.255 scope global eth1
       valid_lft forever preferred_lft forever
    inet6 fe80::e9b:5eff:feac:1/64 scope link
       valid_lft forever preferred_lft forever
4: br-lan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP qlen 1000
    link/ether 0c:9b:5e:ac:00:00 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.1/24 brd 192.168.1.255 scope global br-lan
       valid_lft forever preferred_lft forever
    inet6 fd0f:5211:53a6::1/60 scope global noprefixroute
       valid_lft forever preferred_lft forever
    inet6 fe80::e9b:5eff:feac:0/64 scope link
       valid_lft forever preferred_lft forever
5: hwsim0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN qlen 1000
    link/[803] 12:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
```

### Expected behaviour

On "openwrt-23.05.5-x86-64-generic-ext4-combined.img", i have 2 wlans interfaces : 
```
root@OpenWrt:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel master br-lan state UP qlen 1000
    link/ether 0c:f0:bc:cb:00:00 brd ff:ff:ff:ff:ff:ff
3: eth1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP qlen 1000
    link/ether 0c:f0:bc:cb:00:01 brd ff:ff:ff:ff:ff:ff
    inet 10.145.147.12/24 brd 10.145.147.255 scope global eth1
       valid_lft forever preferred_lft forever
    inet6 fe80::ef0:bcff:fecb:1/64 scope link
       valid_lft forever preferred_lft forever
4: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN qlen 1000
    link/ether 02:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
5: wlan1: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN qlen 1000
    link/ether 02:00:00:00:01:00 brd ff:ff:ff:ff:ff:ff
6: hwsim0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN qlen 1000
    link/[803] 12:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
7: br-lan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP qlen 1000
    link/ether 0c:f0:bc:cb:00:00 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.1/24 brd 192.168.1.255 scope global br-lan
       valid_lft forever preferred_lft forever
    inet6 fd21:1524:f246::1/60 scope global noprefixroute
       valid_lft forever preferred_lft forever
    inet6 fe80::ef0:bcff:fecb:0/64 scope link
       valid_lft forever preferred_lft forever
```

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-07-21:
netdev-s are created once you bring up access point.

Raizo62, 2025-07-22:
I'm surprised. The `kmod-mac80211-hwsim` module is used to create fake Wi-Fi interfaces to simulate Wi-Fi. A Wi-Fi interfaces is visible before the AP is created.

For networking courses, we use this module with the https://github.com/Raizo62/vwifi software (and doc https://github.com/Raizo62/vwifi/wiki/Install-on-OpenWRT-X86_64) to teach Wi-Fi configuration on OpenWRT. This software works thanks to this `kmod-mac80211-hwsim` module and creates its own interfaces to have a more realistic MAC address.

When we configure the module with 0 Wi-Fi interfaces ,
`echo 'mac80211_hwsim radios=0' > /etc/modules.d/mac80211-hwsim`
or with the command : `modprobe  mac80211-hwsim radios=0`
OpenWRT seems to create two wifi interfaces.
We have to destroy them with the `uci delete wireless` command.

Is there a command or option to return to a more traditional operation similar to a real Wi-Fi card?

brada4, 2025-07-22:
I think hwsim is controlled via if phy phy0 hwsim now, the netdevs are not created even for real wifi unless the function is added.