openwrt/openwrt #19498: IPSET Config Not Creating an `nftables set` With `flags interval` for CIDR support

ronhombre, 2025-07-23:
### Describe the bug

I'm creating an ipset config using fw4 so I can explicitly only allow Cloudflare IPs to access my web server.

The bug is that you need the `flags interval` when creating an nftables set in order for CIDRs to work which are included in https://www.cloudflare.com/ips-v4 and https://www.cloudflare.com/ips-v6.

Expected behavior is for my web server to be accessible by port forward through Cloudflare IPs only.

### OpenWrt version

r28597-0425664679

### OpenWrt release

24.10.1

### OpenWrt target/subtarget

ramips/mt7621

### Device

TP-Link ER605 v2

### Image kind

Official downloaded image

### Steps to reproduce

1. Run in SSH `wget -q https://www.cloudflare.com/ips-v4 -O /etc/luci-uploads/ips-v4.txt`
2. Create an IP set in Luci
3. Set family to ipv4
4. Set packet field match to src_ip
5. Include the file in step 1.
6. Create a port forward rule for either 80 or 443 (depending on your cloudflare config)
7. Reference the ipset we created
8. Save and Apply
9. Visit the website through cloudflare in an external IP

### Actual behaviour

* Web server is down according to cloudflare when I try accessing it
* Web server is also down when attempted to access by my public ip

### Expected behaviour

* Web server is accessible through Cloudflare
* Web server is down when attempted to access by my public ip

### Additional info

Generated nftables set missing `flags interval` and CIDR notations in the IPs
```
 set cloudflare_v4 {
                type ipv4_addr
                elements = {
                        173.245.48.0,
                        103.21.244.0,
                        103.22.200.0,
                        103.31.4.0,
                        141.101.64.0,
                        108.162.192.0,
                        190.93.240.0,
                        188.114.96.0,
                        197.234.240.0,
                        198.41.128.0,
                        162.158.0.0,
                        104.16.0.0,
                        104.24.0.0,
                        172.64.0.0,
                        131.0.72.0,
                }
        }

        set cloudflare_v6 {
                type ipv6_addr
                elements = {
                        2400:cb00::,
                        2606:4700::,
                        2803:f800::,
                        2405:b500::,
                        2405:8100::,
                        2a06:98c0::,
                        2c0f:f248::,
                }
        }
```

Also, am I crazy or what? I feel like this shouldn't be a bug

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-07-23:
If it is not obvious from luci ui- set type with subnets is src_net.

ronhombre, 2025-07-23:
> If it is not obvious from luci ui- set type with subnets is src_net. Thats a radical 10 year step away from ipset which happily expanded cidr to list of 2^n addresses.

I swear I have tried everything. I've spent at least 8 hours trying to debug this, and I have given up. I will try again later. I'm too exhausted for now

ronhombre, 2025-07-23:
For the record, I have also tried Cloudflared but there seems to be an issue with the package on my system. It was a build issue upstream. Currently, I just have Nginx dropping connections with `return 444`

ronhombre, 2025-07-23:
> nginx error 444 means no tcp connection.

Did I stutter in text? I meant, I'm resorting to that which is not ideal because it still tells threat actors something exists in the port. The ideal way is to drop or reject the connection at the router

brada4, 2025-07-23:
I understand the big picture. I an just explaiining that with your (mis-)configuration cloudflare is unable to connect your web server.
You can run tcpdump -nvv -iwan port 443 to see packets arriving and currently rejected.

e9hack, 2025-07-23:
For me, it does work.
/etc/config/firewall contains:
```
config	ipset
	option enabled '1'
	option name 'drop-ssh-static'
	option match 'src_net'
	option loadfile '/var/dropcidr-ssh.txt'
```
/var/dropcidr-ssh.txt contains:
```
120.24.0.0/14
141.98.11.0/24
31.41.244.0/24
45.140.17.0/24
49.212.220.0/24
```
this results in:
```
root@MY-ROUTER:~# nft list set inet fw4 drop-ssh-static
table inet fw4 {
        set drop-ssh-static {
                type ipv4_addr
                flags interval
                auto-merge
                elements = { 31.41.244.0/24, 45.140.17.0/24,
                             49.212.220.0/24, 120.24.0.0/14,
                             141.98.11.0/24 }
        }
}
```

brada4, 2025-07-24:
Could you apply a working example provided first, then if the rules dont work right - get them from /etc/config/firewall and lets sift throu them whether it is a mistake  or indeed a bug or implementation gap.

brada4, 2025-07-24:
If you like clicking - 2 rules one ip4 other ip6 should be dnat, zone wan src dport 443 tcp/udp, rewrite destination to lan/dmz webserver.