openwrt/openwrt #19505: realtek: 10G SFPs on TP-Link TL-ST1008F v2.0

jwh, 2025-07-23:
### Describe the bug

Having a look at this, while 1G LX and 1G copper work ok (sometimes, there is weirdness with tags and drops, like other realtek chips), the 10G LR sfps I have handy dump this:

[  156.295291] sfp sfp-p8: module FS               SFP-10GLR-31     rev      sn G2502094504      dc 250226
[  156.306060] rtl83xx-switch switch@1b000000 lan8: unsupported SFP module: validation with support 00,00000000,00001000,00006440 failed

@plappermaul is there something I need to do with these?

thanks

### OpenWrt version

r1148+447-c9e934ffd8

### OpenWrt release

24.10

### OpenWrt target/subtarget

realtek/rtl930x

### Device

TP-Link TL-ST1008F v2.0

### Image kind

Self-built image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

jwh, 2025-07-25:
Thanks - re mac addresses, they should probably be explicitly set for all ports I think, if they are not set by firmware/uboot etc, as not all DSA ports may be bridged ports, perhaps something that should be added to default generated configs?

jwh, 2025-07-26:
There is something really fun with these:

[  667.002704] sfp sfp-p1: module TP-Link          SM311LS(UN)      rev 5.0  sn 224B362000316    dc 241123
[  667.074868] rtl83xx-switch switch@1b000000 lan1: Link is Up - 1Gbps/Full - flow control off
[  667.084335] switch: port 1(lan1) entered blocking state
[  667.090180] switch: port 1(lan1) entered forwarding state

Settings for lan1:
        Supported ports: [ FIBRE ]
        Supported link modes:   1000baseX/Full
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  1000baseX/Full
        Advertised pause frame use: Symmetric Receive-only
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Speed: 1000Mb/s
        Duplex: Full
        Auto-negotiation: on
        Port: FIBRE
        PHYAD: 0
        Transceiver: internal
        Supports Wake-on: d
        Wake-on: d
        Link detected: yes

        Laser output power                        : 0.2799 mW / -5.53 dBm
        Receiver signal average optical power     : 0.0000 mW / -inf dBm

There is no pair plugged in, so... if I remove the SFP, the link is no longer detected but 1000/full remains

jwh, 2025-07-27:
That would probably make sense, could it rely on something else to retrieve actual state perhaps? It's mostly cosmetic, the LEDs don't light up, but it can be confusing as the netdev reports up.  If it's limited by hardware then is it even worth trying to make it accurate?

>

plappermaul, 2025-07-27:
Just a wild guess. Maybe something like [sfp_upstream_ops for RTL8214FC](https://github.com/openwrt/openwrt/blob/c2066fcb95d48eb3d8cddb7076c71a4f763b0349/target/linux/realtek/files-6.12/drivers/net/phy/rtl83xx-phy.c#L3690) needs to be implemented as a fallback. But no idea now.

closed 2025-08-30 by jwh