openwrt/openwrt #19559: uqmi reports wrong data-status

fid0did0, 2025-07-27:
### Describe the bug

I'm using LTE interface to access the wwan throught qmi protocol.
The connection is working, but uqmi report a wrong status:
```
root@OpenWrt:~# ping openwrt.org
PING openwrt.org (2a03:b0c0:3:d0::1a51:c001): 56 data bytes
64 bytes from 2a03:b0c0:3:d0::1a51:c001: seq=0 ttl=49 time=21.741 ms
64 bytes from 2a03:b0c0:3:d0::1a51:c001: seq=1 ttl=49 time=20.621 ms
64 bytes from 2a03:b0c0:3:d0::1a51:c001: seq=2 ttl=49 time=20.301 ms
64 bytes from 2a03:b0c0:3:d0::1a51:c001: seq=3 ttl=49 time=20.101 ms
64 bytes from 2a03:b0c0:3:d0::1a51:c001: seq=4 ttl=49 time=21.421 ms
^C
--- openwrt.org ping statistics ---
5 packets transmitted, 5 packets received, 0% packet loss
round-trip min/avg/max = 20.101/20.837/21.741 ms
root@OpenWrt:~# uqmi -t 1000 --device=/dev/cdc-wdm0 --get-data-status
"disconnected"
root@OpenWrt:~# 
``` 

Any idea?

### OpenWrt version

r28739-d9340319c6

### OpenWrt release

24.10.2

### OpenWrt target/subtarget

ramips/mt7620

### Device

D-Link DWR-921 C1

### Image kind

Official downloaded image

### Steps to reproduce

uqmi -t 1000 --device=/dev/cdc-wdm0 --get-data-status

### Actual behaviour

"disconnected"

### Expected behaviour

I was expecting "connected" or equivalent.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

jjurski, 2025-07-28:
I am hitting a similar (same?)issue when attempting an upgrade from 23.5.5 to 24.10.0 (or 24.10.2 - I tried both with the same result) on D-Link DWR-921 C3. _uqmi_ always returns some errors (always after a prolonged period of time). Luci shows the following (wwan0 seems to be up and running but wwan_4 which is automatically created is inactive):

[image]

If this is of any help, I noticed the following difference in eeprom load in the 24.10.x logs (no such error shows in 23.5.5 logs):

[image]

In my case, I am unable to establish LTE connectivity to the internet. Downgrading back to 23.5.5 solved the problem for now.

fid0did0, 2025-07-28:
Can you please try to execute the following command:
uqmi -t 1000 --device=/dev/cdc-wdm0 --get-data-status

when you are sure that the LTE connection is established.

jjurski, 2025-07-28:
@fid0did0 - I have just created issue #19585

fid0did0, 2025-08-02:
I close the issue and move it to the uqmi repo.