openwrt/openwrt #19566: Dropbear : Compilatilation error

grosjo, 2025-07-28:
### Describe the bug

src/loginrec.c: In function 'lastlog_openseek':
src/loginrec.c:1307:16: error: too few arguments to function 'lastlog_filetype'
 1307 |         type = lastlog_filetype(LASTLOG_FILE);
      |                ^~~~~~~~~~~~~~~~
src/loginrec.c:1281:1: note: declared here
 1281 | lastlog_filetype(char *filename)
      | ^~~~~~~~~~~~~~~~
src/loginrec.c:1310:59: error: expected expression before ',' token
 1310 |                         strlcpy(lastlog_file, LASTLOG_FILE, sizeof(lastlog_file));
      |                                                           ^
src/loginrec.c:1314:46: error: expected expression before ',' token
 1314 |                                  LASTLOG_FILE, li->username);
      |                                              ^
src/loginrec.c:1318:41: error: expected expression before ')' token
 1318 |                             LASTLOG_FILE);
      |                                         ^
src/loginrec.c: In function 'lastlog_perform_login':
src/loginrec.c:1360:33: error: expected expression before ',' token
 1360 |                     LASTLOG_FILE, strerror(errno));
      |                                 ^

### OpenWrt version

r30512-ac9a97e49b

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ath79/generic

### Device

Yuncore CPE830

### Image kind

Self-built image

### Steps to reproduce

Just compile dropbear

### Actual behaviour

Not compiling

### Expected behaviour

Compiling

### Diffconfig

```text
# ./scripts/diffconfig.sh
CONFIG_TARGET_ath79=y
CONFIG_TARGET_ath79_generic=y
CONFIG_TARGET_ath79_generic_DEVICE_yuncore_cpe830=y
# CONFIG_ATH9K_HWRNG is not set
# CONFIG_ATH9K_UBNTHSR is not set
# CONFIG_ATH_USER_REGD is not set
CONFIG_CLEAN_IPKG=y
# CONFIG_DOWNLOAD_CHECK_CERTIFICATE is not set
# CONFIG_DROPBEAR_CURVE25519 is not set
# CONFIG_DROPBEAR_DBCLIENT is not set
# CONFIG_DROPBEAR_ED25519 is not set
CONFIG_DROPBEAR_LASTLOG=y
CONFIG_DROPBEAR_LASTLOG_PATH="/var/log/lastlog"
# CONFIG_DROPBEAR_LEGACY_COMPAT is not set
CONFIG_DROPBEAR_LOG_COMMANDS=y
# CONFIG_DROPBEAR_REEXEC is not set
# CONFIG_DROPBEAR_SCP is not set
# CONFIG_DROPBEAR_SFTPSERVER is not set
CONFIG_DROPBEAR_SHA2_512_HMAC=y
# CONFIG_DROPBEAR_SNTRUP761 is not set
# CONFIG_DROPBEAR_SVR_AGENTFWD is not set
# CONFIG_DROPBEAR_SVR_LOCALSTREAMFWD is not set
# CONFIG_DROPBEAR_SVR_LOCALTCPFWD is not set
# CONFIG_DROPBEAR_SVR_REMOTETCPFWD is not set
CONFIG_DROPBEAR_ZLIB=y
# CONFIG_FEED_luci is not set
# CONFIG_FEED_packages is not set
# CONFIG_FEED_routing is not set
# CONFIG_FEED_telephony is not set
# CONFIG_JSON_OVERVIEW_IMAGE_INFO is not set
# CONFIG_KERNEL_BTRFS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_CGROUPS is not set
# CONFIG_KERNEL_CIFS_ACL is not set
# CONFIG_KERNEL_DEBUG_FS is not set
# CONFIG_KERNEL_DEBUG_INFO is not set
CONFIG_KERNEL_DEVTMPFS=y
CONFIG_KERNEL_DEVTMPFS_MOUNT=y
# CONFIG_KERNEL_ELF_CORE is not set
# CONFIG_KERNEL_EROFS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_EROFS_FS_SECURITY is not set
# CONFIG_KERNEL_EROFS_FS_XATTR is not set
# CONFIG_KERNEL_EXT4_FS_POSIX_ACL is not set
# CONFIG_KERNEL_EXT4_FS_SECURITY is not set
# CONFIG_KERNEL_F2FS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_F2FS_FS_SECURITY is not set
# CONFIG_KERNEL_FS_POSIX_ACL is not set
# CONFIG_KERNEL_HFSPLUS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_HFS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_IPV6_MROUTE is not set
# CONFIG_KERNEL_IPV6_SEG6_LWTUNNEL is not set
# CONFIG_KERNEL_IP_MROUTE is not set
# CONFIG_KERNEL_JFFS2_FS_POSIX_ACL is not set
# CONFIG_KERNEL_JFFS2_FS_SECURITY is not set
# CONFIG_KERNEL_JFS_POSIX_ACL is not set
# CONFIG_KERNEL_KALLSYMS is not set
# CONFIG_KERNEL_KEYS is not set
# CONFIG_KERNEL_MAGIC_SYSRQ is not set
# CONFIG_KERNEL_MPTCP is not set
# CONFIG_KERNEL_NAMESPACES is not set
# CONFIG_KERNEL_NET_L3_MASTER_DEV is not set
# CONFIG_KERNEL_NFS_ACL_SUPPORT is not set
# CONFIG_KERNEL_NF_CONNTRACK_TIMEOUT is not set
# CONFIG_KERNEL_POSIX_MQUEUE is not set
CONFIG_KERNEL_PROC_PAGE_MONITOR=y
CONFIG_KERNEL_PROC_STRIPPED=y
# CONFIG_KERNEL_REISER_FS_POSIX_ACL is not set
# CONFIG_KERNEL_SECCOMP is not set
# CONFIG_KERNEL_SWAP is not set
# CONFIG_KERNEL_TMPFS_POSIX_ACL is not set
# CONFIG_KERNEL_UBIFS_FS_SECURITY is not set
# CONFIG_KERNEL_WERROR is not set
# CONFIG_KERNEL_XFS_POSIX_ACL is not set
CONFIG_LINUX_6_12=y
CONFIG_LUCI_SRCDIET=y
# CONFIG_PACKAGE_ATH_DFS is not set
# CONFIG_PACKAGE_MAC80211_DEBUGFS is not set
# CONFIG_PACKAGE_MAC80211_MESH is not set
# CONFIG_PACKAGE_apk-mbedtls is not set
# CONFIG_PACKAGE_ca-bundle is not set
CONFIG_PACKAGE_cgi-io=y
# CONFIG_PACKAGE_dnsmasq is not set
# CONFIG_PACKAGE_getrandom is not set
CONFIG_PACKAGE_kmod-ag71xx=y
# CONFIG_PACKAGE_kmod-ag71xx-legacy is not set
# CONFIG_PACKAGE_kmod-gpio-button-hotplug is not set
# CONFIG_PACKAGE_kmod-lib-crc-ccitt is not set
CONFIG_PACKAGE_kmod-net-selftests=y
CONFIG_PACKAGE_kmod-nft-bridge=y
CONFIG_PACKAGE_kmod-phylink=y
# CONFIG_PACKAGE_kmod-ppp is not set
# CONFIG_PACKAGE_knot-resolver_dnstap is not set
CONFIG_PACKAGE_libiptext=y
CONFIG_PACKAGE_libiptext-nft=y
CONFIG_PACKAGE_libiptext6=y
CONFIG_PACKAGE_libiwinfo-lua=y
CONFIG_PACKAGE_liblua=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-lua=y
CONFIG_PACKAGE_liblucihttp-ucode=y
CONFIG_PACKAGE_libmbedtls=m
CONFIG_PACKAGE_libubus-lua=y
# CONFIG_PACKAGE_libuclient is not set
CONFIG_PACKAGE_libustream-mbedtls=m
CONFIG_PACKAGE_libxtables=y
CONFIG_PACKAGE_lua=y
CONFIG_PACKAGE_luci=y
CONFIG_PACKAGE_luci-app-filebrowser=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-package-manager=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-lib-base=y
CONFIG_PACKAGE_luci-lib-ip=y
CONFIG_PACKAGE_luci-lib-jsonc=y
CONFIG_PACKAGE_luci-lib-nixio=y
CONFIG_PACKAGE_luci-light=y
CONFIG_PACKAGE_luci-lua-runtime=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
# CONFIG_PACKAGE_odhcp6c is not set
# CONFIG_PACKAGE_odhcpd-ipv6only is not set
# CONFIG_PACKAGE_openwrt-keyring is not set
# CONFIG_PACKAGE_ppp is not set
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
# CONFIG_PACKAGE_uclient-fetch is not set
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-log=y
CONFIG_PACKAGE_ucode-mod-lua=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
# CONFIG_PACKAGE_urandom-seed is not set
# CONFIG_PACKAGE_urngd is not set
# CONFIG_PACKAGE_usign is not set
CONFIG_PACKAGE_wpad-basic=y
# CONFIG_PACKAGE_wpad-basic-mbedtls is not set
CONFIG_PACKAGE_zoneinfo-asia=y
CONFIG_PACKAGE_zoneinfo-core=y
CONFIG_PROCD_SHOW_BOOT=y
# CONFIG_SIGNATURE_CHECK is not set
# CONFIG_SIGNED_PACKAGES is not set
# CONFIG_TARGET_ROOTFS_INITRAMFS is not set
CONFIG_TESTING_KERNEL=y
# CONFIG_USE_APK is not set
# CONFIG_USE_FS_ACL_ATTR is not set
CONFIG_USE_MKLIBS=y
# CONFIG_USE_SECCOMP is not set
# CONFIG_WPA_MBO_SUPPORT is not set
# CONFIG_PACKAGE_libpthread is not set
# CONFIG_PACKAGE_libuclient is not set
# CONFIG_PACKAGE_opkg is not set
# CONFIG_PACKAGE_uclient-fetch is not set
```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

rockdrilla, 2025-07-28:
It looks like I've did mistake here:

```diff
--- a/package/network/services/dropbear/Makefile
+++ b/package/network/services/dropbear/Makefile
@@ -138,7 +138,7 @@ DB_OPT_COMMON = \
        MAX_UNAUTH_PER_IP,$(CONFIG_DROPBEAR_MAX_UNAUTH_PER_IP) \
        UNAUTH_CLOSE_DELAY,$(CONFIG_DROPBEAR_UNAUTH_CLOSE_DELAY) \
        $(if $(CONFIG_DROPBEAR_RSA),DROPBEAR_DEFAULT_RSA_SIZE$(comma)$(CONFIG_DROPBEAR_DEFAULT_RSA_SIZE)) \
-       $(if $(CONFIG_DROPBEAR_LASTLOG),LASTLOG_FILE$(comma)$(CONFIG_DROPBEAR_LASTLOG_FILE)) \
+       $(if $(CONFIG_DROPBEAR_LASTLOG),LASTLOG_FILE$(comma)$(CONFIG_DROPBEAR_LASTLOG_PATH)) \
        $(if $(CONFIG_DROPBEAR_UTMP),UTMP_FILE$(comma)$(CONFIG_DROPBEAR_UTMP_FILE)) \
        $(if $(CONFIG_DROPBEAR_WTMP),WTMP_FILE$(comma)$(CONFIG_DROPBEAR_WTMP_FILE)) \

```

@grosjo can you try this fix?

rockdrilla, 2025-07-28:
Looks interesting. I'll investigate it a bit later.

grosjo, 2025-07-29:
@rockdrilla any update ? this is really blocking. Thank you

grosjo, 2025-07-29:
I did
`rm -rf build_dir/target-mips_24kc_musl/dropbear-2025.88/
`
I get
```
src/svr-authpubkey.c: In function 'checkpubkey':
src/svr-authpubkey.c:541:60: error: 'struct AuthState' has no member named 'pubkey_info'
  541 |                         keyblob, keybloblen, &ses.authstate.pubkey_info);
      |                                                            ^

```

This is really blocking. Can you kindly help ?

rockdrilla, 2025-07-29:
Please provide output of following command:

`grep -i -e libc -e dropbear .config`

grosjo, 2025-07-30:
so ?

rockdrilla, 2025-07-31:
Got it: https://github.com/mkj/dropbear/pull/374

rockdrilla, 2025-08-01:
@oldium does it works "fine" too? :)

grosjo, 2025-08-03:
resolved with https://github.com/openwrt/openwrt/pull/19611https://github.com/openwrt/openwrt/pull/19611

grosjo, 2025-08-04:
Now, I get

```
src/common-algo.c:297:101: error: 'sha512_desc' undeclared here (not in a function); did you mean 'sha256_desc'?
  297 | static const struct dropbear_kex kex_sntrup761 = {DROPBEAR_KEX_PQHYBRID, NULL, 0, &sntrup761_desc, &sha512_desc };
      |                                                                                                     ^~~~~~~~~~~
      |                                                                                                     sha256_desc

```

oldium, 2025-08-06:
@grosjo Can you please send the problematic diffconfig again? When I tested compilation with your config from the original report, I got `CONFIG_DROPBEAR_WTMP` disabled for some reason. When I enabled it, the patched `dropbear` correctly set `WTMP_FILE` in `localoptions.h`, so it should use it too.

Are you sure you had all 3 patches from #19611 applied to your tree when building `dropbear` by yourself?

You can verify the values yourself when doing `grep WTMP build_dir/target-mips_24kc_musl/dropbear-2025.88/localoptions.h`. This should be the output:

```
#define WTMP_FILE "/var/log/wtmp"
```

grosjo, 2025-08-06:
diffconfig same as already written

yes, I applied the PR diff file to include the whole thing

no, it does not pick the right location for the file

oldium, 2025-08-06:
Why I am asking for the fresh diff config? You wrote that you have a [compilation error](https://github.com/openwrt/openwrt/issues/19566#issuecomment-3149847214) and in the next message 15 minutes later you [wrote](https://github.com/openwrt/openwrt/issues/19566#issuecomment-3149895862) that the `/var/log/wtmp` is not used. So there are three possibilities:

1. you changed the configuration to make dropbear compile, or
2. you fixed the compilation by yourself somehow - please tell how, or
3. you do not have the fixed dropbear at all, because the compilation failed, and you are testing old (broken) version of dropbear.

I am just trying to get where you are 😊

grosjo, 2025-08-07:
> Why I am asking for the fresh diff config? You wrote that you have a [compilation error](https://github.com/openwrt/openwrt/issues/19566#issuecomment-3149847214) and in the next message 15 minutes [...]

As mentioend, the initial bug is fixed,

the second compilation error is : CONFIG_DROPBEAR_SNTRUP761 shall be disabled properly

grosjo, 2025-08-07:
```
Aug  7 20:25:07 cpe-terrace dropbear[2252]: Password auth succeeded for 'root' from 192.168.101.192:33074
Aug  7 20:25:07 cpe-terrace dropbear[2252]: User root executing login shell
Aug  7 20:25:07 cpe-terrace dropbear[2253]: wtmp_write: problem writing /dev/null/wtmp: Not a directory

```

grosjo, 2025-08-08:
```
# grep WTMP build_dir/target-mips_24kc_musl/dropbear-2025.88/localoptions.h
#define WTMP_FILE "/var/log/wtmp"
# grep dropbear bin/targets/ath79/generic/openwrt-ath79-generic-yuncore_cpe830.manifest
dropbear - 2025.88-r3

```

oldium, 2025-08-08:
Musl-libc does not include utmp/wtmp implementation, please see https://wiki.musl-libc.org/faq.html#Q:_Why_is_the_utmp/wtmp_functionality_only_implemented_as_stubs?

grosjo, 2025-08-11:
> [@rockdrilla](https://github.com/rockdrilla) Definitely it is completely on the app-side, but you also need some support from the OS. ChatGPT-5 thinks the Musl-based distros (like [Alpine](https://gitlab.alpinelinux.org/alpine/aports/-/issues/5075)) do not handle utmp/wtmp and _also_ lastlog. So it makes sense to simply disable this functionality completely for Musl.

So if ChatGPT says it :D :D :D :D

closed 2025-08-11 by grosjo

oldium, 2025-08-11:
> So if ChatGPT says it :D :D :D :D

I verified the claim, I even put the actual link to Alpine. Musl does not support it, that is the main point of the discussion. Your comment does not add any information and is only offensive.

grosjo, 2025-08-11:
> > > [@rockdrilla](https://github.com/rockdrilla) Definitely it is completely on the app-side, but you also need some support from the OS. ChatGPT-5 thinks the Musl-based distros (like [Alpine](https://gitlab.alpinelinux.org/alpine/aports/-/issues/5075)) do not handle utmp/wtmp and _also_ lastlog. So it makes sense to simply disable this functionality completely for Musl.
> > 
> > 
> > So if ChatGPT says it :D :D :D :D
> 
> [@grosjo](https://github.com/grosjo) I verified the claim, I even put the actual link to Alpine, which proves it. I also verified Musl implementation in their Git repository. Musl does not support it, that is the main point of the discussion. Your comment does not add any information and is only offensive.

Not saying your statement is wrong. Just joking on taking ChatGPT as a reference ;)

rockdrilla, 2025-08-11:
[image]