openwrt/openwrt #19585: LTE connectivity issues after switching to 10.2

jjurski, 2025-07-28:
### Describe the bug

LTE connectivity issue exists when running 24.10.0 to 24.10.2 (I tried both official images with the same result) on D-Link DWR-921 C1/C3. Luci shows the following (wwan0 seems to be up and running but wwan_4 which is automatically created is inactive):

[image]

If this is of any help, I noticed the following difference in eeprom load in the 24.10.x logs (no such error shows in 23.5.5 logs):

[image]

Downgrading back to 23.5.5 solved the problem.

### OpenWrt version

https://downloads.openwrt.org/releases/24.10.2/targets/ramips/mt7620/openwrt-24.10.2-ramips-mt7620-dlink_dwr-922-e2-squashfs-sysupgrade.bin

### OpenWrt release

from 24.10.0 to 24.10.2

### OpenWrt target/subtarget

ramips/mt7620

### Device

D-Link DWR-921 C1 (but note that C3 is shown on the label)

### Image kind

Official downloaded image

### Steps to reproduce

23.5.5 to 24.10.0 or to 24.10.2 - LTE connection no longer works; cleaning up the configuration to factory defaults and attempt to manually recreate the configuration does not help

### Actual behaviour

No LTE connection

### Expected behaviour

LTE connectivity expected to work

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

fid0did0, 2025-07-28:
Can you please elaborate this sentence: D-Link DWR-921 C1 (but note that C3 is shown on the label)?

fid0did0, 2025-07-28:
I report all the test I performed on my D-Link DWR-921 C3.
I performed several test in various combination.
I tested:

- Two OWRT version (23.05.05 and 24.10.2)
- Two LTE modules (the original BM806U in DWR-921-C3 and the Quectel EC25-E)
- Two network.wwan.pdptype (ipv4 and ipv6)

here the results
```
+---------------+-----------------------+-------+-------+-----------------------+
+ OWRT	version	+ LTE module		+ ipv	+ ping	+ get-data-status	+
+---------------+-----------------------+-------+-------+-----------------------+
+--  23.05.05	+ BroadMobi BM806U-E1	+ ip4	+ OK	+ connected		+
+--  23.05.05	+ BroadMobi BM806U-E1	+ ip6	+ NOK	+ disconnected		+
+--  23.05.05	+ Quectel EC25-E 	+ ip4	+ OK	+ connected		+
+--  23.05.05	+ Quectel EC25-E	+ ip6	+ NOK	+ disconnected		+
+--  24.10.02	+ BroadMobi BM806U-E1	+ ip4	+ OK	+ connected		+
+--  24.10.02	+ BroadMobi BM806U-E1	+ ip6	+ NOK	+ disconnected		+
+--  24.10.02	+ Quectel EC25-E	+ ip4	+ OK	+ connected		+
+--  24.10.02	+ Quectel EC25-E	+ ip6	+ OK	+ disconnected		+
----------------------------------------+-------+-------+-----------------------+
``` 

Please note that ip4 and ip6 in my case requires different provider configuration (apn, usrname and password).

@jjurski can you please complement the table with your configuration?
In C1 is expected a different LTE module (WNC D18Q1 GSM/UMTS/LTE mPCIe Module) as reported by https://openwrt.org/toh/d-link/dwr-921.

brada4, 2025-07-28:
There is no problem with firmware load messages - no firmware was found in initramfs because there is no initramfs at all and the firmware file was loaded from normal file system's /lib/firmware. 
Only one WAN mode works at a time, so you have to select one you intend to use
https://openwrt.org/docs/guide-user/network/wan/wwan/3gdongle
(some dongles do not support all modes)
Alternatively install rooter goldenorb firmware (OpenWrt derivative) clearing config before and after and snoop on what parameters they select for particular modem with particular network and try to bring that back to OpenWrt.

jjurski, 2025-07-30:
@fid0did0: I only referred to IPv4 when saying the connectivity did not work (I did not pay attention to IPv6 when reporting the problem). I can now report the 23.05.5 status by accessing the router remotely and so here it is:

```
+---------------+-----------------------+-------+-------+-----------------------+
+ OWRT	version	+ LTE module		+ ipv	+ ping	+ get-data-status	+
+---------------+-----------------------+-------+-------+-----------------------+
+ 23.05.5	+ ?			+ IPv4	+ OK	+ connected		+
+---------------+-----------------------+-------+-------+-----------------------+
```
but I no longer can try 24.10.x as I only have remote access to the router now (via LTE) and cannot experiment more. But what I remember is the following:

```
+---------------+-----------------------+-------+-------+-----------------------+
+ 24.10.0	+ ?			+ IPv4	+ NOK	+ uqmi hang*		+
+ 24.10.0	+ ?			+ IPv4	+ NOK	+ uqmi hang*		+
+---------------+-----------------------+-------+-------+-----------------------+
```
(?) - I did not check the module type; how can this be done?

(*) "uqmi hang" means some timeout - in general, with 23.05.5 the `uqmi` command returns the results immediately. With 24.10.x, it was hanging and never really returning any response...

jjurski, 2025-07-30:
@brada4

> There is no problem with firmware load messages 

23.05.5 does not report such error messages in logs:
[image]

> Only one WAN mode works at a time, so you have to select one you intend to use

In both cases (23.05.5 and 24.10.x) the configuration files were exactly the same

> Alternatively install rooter goldenorb firmware (OpenWrt derivative) clearing config before and after and snoop on what parameters they select for particular modem with particular network and try to bring that back to OpenWrt.

I can try next time I have physical access to the router but it will not be anytime soon...

brada4, 2025-07-30:
It s not related to gsm component. If wifi works just pass along, dont worry.

fid0did0, 2025-07-30:
At least for version 23.05.5 the behavior is the same like me.
For 24.10.x (I tested 24..10.2) we have different result, for me IPv4 is working.
To better understand what's is going wrong type:
`logread |grep wwan`
or
`logread |grep netfid`

On this forum post https://forum.openwrt.org/t/lte-in-ipv6-mode-do-not-work/237881 AndrewZ is helping me a lot to fix the IPV6 problem. You can try to follow the same step with IPV6, if do not work you can try to open a ticket as well.

fid0did0, 2025-07-31:
@jjurski 
Could you try to remove the 
'uci set network.wwan.delay='5'
or try bigger delay (if you have this setting).
This has probably an effect.

fid0did0, 2025-08-03:
Ok. Send the log for 24.10.x when you can, will check what's wrong there.

jjurski, 2026-08-16:
I have just tried 25.12.5 and the problem went away. I am closing this issue as I think nobody will want to spend their time to investigate. **Thanks for your willingness to help Giuseppe & Andrew!**

closed 2026-08-16 by jjurski